pop-os/system76-power #2: Add configuration files

jackpot51, 2018-03-15:


ayoungethan, 2019-06-13:
A configuration file is ideal, as it would 
a. help user understand what the power daemon actually modifies (currently it is unclear), and 
b. allow the user to modify or create custom settings for specific use cases

For example, in my case, I need low latency audio settings for my laptop. This requires consistent CPU frequency (minimize scaling) to decrease processing latency and jitter. Both scaling and turbo boost add jitter, whereas scaling adds latency. 

However, "high performance" I believe puts on turbo boost, which adds jitter. Ideally, I would be able to create at least one "low latency" profile that lets me minimize the scaling and turn off turbo boost without losing the other performance enhances that actually help (such as higher core clock frequency).

This extension: https://github.com/konkor/cpufreq provides a front-end which is resource intensive and a bit confusing as it seems to overlap and conflict with a lot of what system76-power does, but the parameters it modifies could be extracted to a text file interface (e.g., in /etc/default/system76-power) which gives syntax and parameters for quickly and easily creating custom power profiles that can appear in the power menu.

The three default settings are fine for most users, but I could offer some language for users seeking low latency, low jitter cpu performance.

Baughn, 2019-10-25:
Another case for me: On a Dell XPS 9380, the 'balanced' power profile makes it extremely hot and leaves the fans running constantly, _if_ the laptop is connected to AC. When it's running on battery it behaves normally, so I end up using 'balanced' on battery and 'powersave' when I don't need to save power.

I'd very much like to figure out which particular setting causes this, but right now there's no way to pick and choose.

mmstick, 2019-10-25:
Turbo mode is likely the culprit. It tends to generate a lot of heat when it's activated.

Baughn, 2019-10-26:
Seems likely enough... but if that's the case, then what do I do about it?

The 'battery' profile limits CPU frequency to 50% of the base, which is 900MHz. I understand that power consumption is usually about the square of the frequency, but that still seems a bit much; there's noticeable lag. I'd like to hit some kind of Goldilocks zone.

(This laptop leans a lot on "turbo". 600MHz minimum, 1.8GHz base, 3 GHz all-core turbo, 3.6GHz peak...)

ayoungethan, 2019-10-26:
It sounds to me like its not the balanced profile so much as some
underlying process(es) running the CPU at too high speed. Balanced should
provide optimal throughput performance, battery life and thermal control in
most circumstances (except where low latency and high throughput are both
needed). But it is a permissive profile and the "race to sleep" logic that
it depends upon can easily be defeated by errant processes that keep the
CPU active.

I would wonder whether you experience this problem in a live boot
situation? And what does CPU utilization look like during this problem? Any
active processes hogging a core or two?

Baughn, 2019-10-26:
I haven't been able to see any difference in active processes; the CPU is idling at ~0% use, top agrees, and `grep MHz /proc/cpuinfo` actually reports it running at minimum frequency, it just gets really hot (and the fans spin up) despite not doing anything special.

That's largely why I'm suspecting some obscure power setting. It's difficult to track down without knowing exactly what the profiles do, but I'd be happy to contribute to a quirks repository if I knew where to look.

Baughn, 2019-10-26:
I'm not sure I found the cause, but I think I at least found a solution.

/sys/module/pcie_aspm/parameters/policy is set to 'default' by default -- surprise. On my system, there appears to be a firmware bug that makes the default _exceedingly_ inefficient, so it has to be explicitly set. Setting it to 'performance' significantly reduced power consumption on AC.

(I've now set it to 'powersupersave', via TLP configuration. There doesn't seem to be a performance loss, but perhaps I've just not run the right application yet.)

ayoungethan, 2019-10-26:
Does it also resolve the heat/fan issues? Given the popularity of that
laptop it looks like a good candidate for a quirk entry. That should
probably be a separate bug report

Baughn, 2019-10-26:
_Seems_ to, but don't quote me on that; it's difficult to be sure, lacking a USB-C power meter. powertop doesn't work with AC connected.

If we're considering quirks entries, here are two more:

- smbios-thermal-ctl should be used to set the firmware to "Cool Bottom" mode, at least by anyone intent on using it on their lap. The default is insanely hot no matter the load, obviously intended for use on a desk.

- mem_sleep_default=deep is a required kernel parameter, otherwise suspend-to-ram doesn't work properly.