pop-os/system76-power #8: Automatically enter High Performance when plugged in

cassidyjames, 2018-04-30:
When connected to power (and perhaps once the battery is charged to a certain percent), we should switch into High Performance mode. I don't see a strong argument for trying to save power when connected to an effectively unlimited source.

Of course we should switch back to whichever profile the user had set if they disconnect power.

DamianRivas, 2018-08-18:
Hi, thank you for the wonderful OS. Is this feature work in progress? I was looking into the power profiles in the system settings and noticed they don't automatically switch when plugging and unplugging the laptop. Google brought me here.

mlncn, 2019-06-12:
Currently i find the option to leave on 'Balanced' (which seems to me to switch between High Performance and Battery Life based on whether it is plugged in) to be very good, and i use the option to override this by explicitly choosing Battery Life when plugged in (to keep it quieter if in a quiet room, was also pretty critical when my fan broke).

So from my perspective this is working exactly as i want, simply use Balanced to have it be automatic, but maybe "Balanced" isn't doing what i perceive it to be doing.

mttkay, 2019-10-20:
> I was looking into the power profiles in the system settings and noticed they don't automatically switch when plugging and unplugging the laptop.

+1

I think it's quite odd that when I unplug the power the laptop does not go into the `Battery` profile. Why is that? It would make so much sense and it's what all laptops did that I have owned over the past 10 years.

DylanC, 2020-04-09:
Also interested in this feature! Its annoying to switch it every time I boot up.

dani763f, 2020-09-28:
+1 Also interested in this feature. 
Plugged in = High performance mode
Disconnect AC = Battery mode
Should be default.

stderr-to-devnull, 2020-11-08:
Well, since this "feature" has been open from 2018, I can objectively deduce that asking nicely won't do the trick...

I'll just take my glib "write a script hurr durr" answer and move on with my life. Who needs usability and common snese in Linux where everyone is willing to search for hours on end on forums / github / reddit for solutions to problems that shouldn't exist in the first place?

Btw your POP shop is notifying me of operating system updates which is unable to install. It is a very sad irony. Well, off to the shell I go.

stefanjaro, 2021-12-06:
This really should be a default feature. 

I've written a Python script that does this and requires a bit of manual setup - 
https://github.com/stefanjaro/popos-power-profile-adjustment

In short, if your battery is being charged but is below 50%, it'll set the profile to "Balanced". If it's above 50% it'll set it to "Performance". But if your battery is not being charged, it'll set the profile to "Battery".

Tom743, 2021-12-15:
I agree. This should be the default behaviour. 
Thanks @stefanjaro for sharing your script! Works great

smdotdev, 2022-01-22:
This should have been the default behavior in the first place, rather there seems to be no development at all from 2018

GlobalAlliance18, 2022-02-14:
> Well, since this "feature" has been open from 2018, I can objectively deduce that asking nicely won't do the trick...
> 
> I'll just take my glib "write a script hurr durr" answer and move on with my life. Who needs usability and common sense in Linux where everyone is willing to search for hours on end on forums / github / reddit for solutions to problems that shouldn't exist in the first place?
> 
> Btw your POP shop is notifying me of operating system updates which is unable to install. It is a very sad irony. Well, off to the shell I go.
> 
> I know I know, it's free yadda yadda. It will, for the forseeable future, remain free and obscure. Bottom of the barrel one could say. That's Linux desktop experience in a nut...shell.

Right. This issue needs a good, firm bump. 

What exactly is going on with this? As far as I can tell, this has been marked as 'may be fixed by #46' , which was deferred until #42 was finished...only that was supposedly scrapped/deferred back before 19.04, for "high priority tasks"? Seriously???

Based on the volume of people asking about this here and on other platforms, for the last  _four years_,  I don't think I'm the only one confused, not just by the absence of any control or function , but also  by  total absence of explanation. It took me a good hour of searching to find this thread and realize I wasn't just missing it in the settings. Even the System76 website tries to palm it off by absurdly suggesting people use tlp, which is ancient, awkward, and actually conflicts with the current manually=selected (and unadjustable!) power management profiles baked into Pop, rendering them ineffective and hamstringing performance in the process. Yes, I checked.

Considering half the work is already done with said manual  power profiles, and there are/were already GUI options for power-state-dependent changes to things like suspend timeouts in Ubuntu, I cannot understand why there can't be a simple window in the Power settings page, to allow people to at least  tweak the percentages in the different rpofiles, and  switch to a profile based on some AC-input acpi triggers or something. It's clearly technically possible based on the number of  scripts and gnome extensions there are awkwardly trying to do the same thing - there just should be something cleaner, reliable and more readily configurable built in. Kindof like my Thinkpad did. On Win98.

Lastly, I'd like to say it's dismaying to see the negative responses to @stderr-to-devnull , who is  bang on the mark about usability and common sense. This sort of  petty gatekeeping against valid expression of dissatisfaction by non-technical users is as big a problem  as hit-and-miss software itself. For an operating system that's trying to be a viable alternative for the mainstream user, and support their own commercial hardware, people have every right to expect that core functions work, and complain when they don't. Attacking them  for not fixing a problem they didn't create is open-source community equivalent of bad tech support  phone-lines, and if you can't see that, feel  free to prove my point with your dislikes. 

Issues like this are why I got my XPS fixed instead of buying a Lemur. They directly undermine my confidence in System76's ability to delivery a functional product and support it.

jacobgkau, 2022-02-14:
@GlobalAlliance18 This is an issue tracker on a development platform. Please keep the discussion on-topic to the feature request; attacking System76 or "the open-source community" is not productive. Yes, Pop!_OS has [many components](https://github.com/orgs/pop-os/repositories) that have been actively developed over the past few years, and the developers' time is split between all of them.

Thumbs-down reactions on a sarcastic, negative comment are not gatekeeping; they are reflective of the "be considerate" rule from the [Pop!_OS Code of Conduct](https://github.com/pop-os/code-of-conduct). Your opinion of the importance of this feature request is noted.

GlobalAlliance18, 2022-02-14:
@jacobgkau For an issue-tracker on a development platform, yours is the first response from devs on this topic in literal _years_. And it took you less than ten minutes from my own remark.

If only the actual problems got addressed this quickly.

mmstick, 2022-02-14:
@GlobalAlliance18 I dropped this because I do not think it's a good idea. I've instead opted for a [different approach to improving performance on AC that actually works](https://github.com/pop-os/system76-scheduler). On modern Intel and AMD CPUs, the Balanced mode gives higher performance than High Performance. Automatically transitioning to High Performance not only slows down how fast a laptop can charge, but it limits the upper boost frequencies that cores can achieve. And regarding TLP, these days the Linux kernel dynamic runtime PM settings are the optimal. System76-Power already applies the most optimal parameters for Balanced mode.

Tom743, 2022-02-14:
May I ask why does the misleading high performance option still exists, or why is not balanced mode renamed? If I understood your comment well, there is no point on using the high performance mode, and I have been using a script that automatically changes the power settings when plugging to AC to realize now that the power options in the UI are not meant to be used. Am I wrong? Or is it that your update to the scheduler hasn't rolled out to the stable versions yet? I don't intend to be rude, I am just trying to understand what is going on here. Thank you

mmstick, 2022-02-14:
@GlobalAlliance18 If you're willing to contribute PRs adding features you want to see, I would not be opposed to helping you get those features implemented.

It also sounds like your laptop's firmware isn't handling pstate parameters well from Linux. Perhaps it is ignoring the parameters assigned by the balanced profile. There should not be any performance issues from using the default settings in Pop. Garbage is not how I'd described how well the balanced mode works with my own open firmware System76 laptop.

Most laptops you'll find are already operating at their highest TDP constraints, with no way of configuring that. They are also entirely limited by thermals, which can only be overcome through undervolting. Even on my own personal laptop, I can boost performance of my CPU by 50% simply by increasing airflow to the CPU with a blower attached to heatsink exhaust. The performance that I get is entirely dependent on the ambient room temperature and how much airflow I can get to its internal heatsinks, rather than the power profile I set.

The battery mode profile is useful because it constrains how often cache is flushed to disks, and limits CPU frequencies while disabling Intel turbo. The balanced mode has recommended defaults for all settings, including allowing the CPU to turbo and choose frequencies however it likes. This is the optimal setting for modern Ryzen and Intel processors because they have chipsets which automatically undervolt/underclock and overvolt/overclock cores dynamically based on CPU load and thermals. You can read about Ryzen Precision Boost Overdrive for more on how that works.

Which means that in practice, the balanced profile attains the highest performance in the widest variety of scenarios. On a highly-threaded workloads, thermal constraints are the limiting factor which throttles CPU frequencies. On lightly-threaded workloads, turbo is still pushing the busy cores as high as it can to reach thermal limits.

In the future, laptop vendors which support use of Linux with their products may support [ACPI Platform Profiles](https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html). System76 Power currently sets this parameter if it exists, but as far as I know only a few Lenovo laptops support it. Firmware which supports it can use power profile data to configure their TDP, so they could offer a reduced TDP for Balanced and a higher TDP for High Performance.

System76 Scheduler is a separate project working to improve desktop responsiveness. It's not really related to System76 Power in any way. It has been released if you have the latest updates installed and haven't disabled the pop-shell extension. it gives CPU priority to actively-focused applications similar to what other OS's have been doing, and increases how often a context switch is performed, which results in the CPU spending more time on busy tasks in the foreground than on background services and applications.

GlobalAlliance18, 2022-02-15:
@mmstick Righto, thanks for the breakdown, very comprehensive, but I think I understand.... most of it. 

In simple terms, it sound like what you're saying is that users should be switching between Balanced for efficient high performance on demand, and Battery mode for power conservation.

Which means we're back where we started - trying to figure out how to switch between two profiles based on whether AC is available or not. That was basically the original point of this request - just nobody knew we should be trying  to "Automatically enter **Balanced mode** when plugged in". So I don't get why this has been dropped. Especially if you're now saying System76 Scheduler is  a 'separate project not related to System76 Power in any way' - which is confusing, because in your last comment, you said you dropped this request because you thought Scheduler was a better approach.

I would love to help, but I'm not a software developer or programmer, and I barely understand what's going on here as is. Judge me for that if you want, but at the end of the day thats what the vast majority of users are, so the best I can think to offer is report problems and make suggestions. I've done that elsewhere with the bugs in your touch gestures, and that was totally ignored too, so rather than open _another_ thread about my personal machines specific power profile behavior, I tried to fix it myself. And here I am.

 I don't think it's my firmware - the same machine has no issues with the normal 'performance' and 'powersave' pstate governors, which as far as I can tell are just baascially alternative profiles to control the same pstates? But maybe your profiles do different things to the pstates? I dont know. But the scope of this request is more about getting everyone's profiles to change automatically on AC, than fixing my specific bugs, so with all the devs apparently already so busy that this was abandoned for years, I'm reluctant to add _more_ requests.

As for how well you say it works on your System76 laptop -  the "trust us, it works if you give us money" approach require more faith that my experiences so far have given me. Sure, maybe Balanced does work great on your laptop. But does it switch between that and the Battery mode profile automatically when you plug and unplug? And if so, then why aren't other laptops? Because I really dont see how that not possible.

mmstick, 2022-02-15:
This issue is about switching to the high performance profile. I decided not to continue pushing for the implementation for that because the balanced profile has the best performance anyway.

Not everyone wants their laptop to automatically switch to battery mode when unplugged. This would come at a major performance cost which wouldn't be necessary in most cases, and then there would be issue reports and complaints about how terrible Pop performs in comparison to {insert other distribution here}.

Android doesn't switch to battery mode until its battery has gotten to the last 10%, and it makes that a dialog request so the user has to consciously decide to do that. This feature would require more design and configuration file support.

System performance in Linux is 100% linked to firmware. You should not need a high performance CPU profile to have good performance. Technically when you query pstate files it will say that system76-power is using the powersave profile, but it tweaks the minimum and maximum frequencies, as well as the turbo mode feature. On balanced it's allowed to choose between 0% and 100% with turbo enabled. It's up to the firmware to decide how to leverage that info.

If your system is having issues, it's because it's either ignoring these values or doing something else entirely. Wouldn't be the first time I've seen reports of laptops [throttling the CPU on Linux](https://news.ycombinator.com/item?id=30333956).

My decision to work on Scheduler is unrelated. I started it in November in my free time outside of my usual work. It doesn't involve power profiles and I didn't feel the need to bump old bug reports to tell people about it when it's better to announce that on Reddit. It's a rewrite of a previous attempt that I made that I couldn't progress on because we were missing the pop-shell piece to make it possible.

GlobalAlliance18, 2022-02-15:
@mmstick The exact words in the original request were " we should switch back to whichever profile the user had set if they disconnect power."  I mean, if you're switching _to_ High Performance on AC, then obviously that implies you are on another profile for battery. Sure, not _everyone_ might want to, but most people DO want  switch back to battery mode when...on Battery. Hence the name. 

Laptops aren't phones. We both know the power, performance and typical usage patterm of some  Android phone with an ARM processor that sits in your pocket, screen off in sleep mode 90% of the time,  vs a core i7 laptop are not a fair comparison. This sort of semantics seems like a really poor grounds for totally abandoning a feature that has been a staple in basically every laptop ever, and has a demonstrably large number of confused users trying to find a way to configure this to their needs. I really don't understand how there's any debate about the need for this functionality.

I understand  what youre saying about System76 power vs the normal performance/powersave profile...but none of that theory changes real world results. At present I'm trying to get some of the older extensions running on 21.10/GNOME41, or at the very least kludge together cpufrequtils some scripts of my own, so I can run benchmarks to see if there's an actual functional difference in performance and power use between System76 Power profiles , and the original performance/powersave governor-based controls. See if 'Balanced" really DOES perform better.

 If I can get it working, I'll both the comparisons and methodology for the sake of  other people, so at least theres some real-world feedbakc and  _some_ way to achieve something maybe usable day to day.

mmstick, 2022-02-15:
If you are using cpufreq, that way of controlling the CPU governor was deprecated when Intel introduce the pstate driver. The pstate powersave profile already performs better than cpufreq perfomance profile while using less energy.

GlobalAlliance18, 2022-02-15:
I know.

If a deprecated method is outperforming your implementation, logically this would suggest your implementation is perhaps not performing as well as you may be assuming.

mmstick, 2022-02-15:
We don't assume anything. Since Pop began, we have evaluated every change ever introduced in a QA lab on multiple systems. I don't have control over the implementation of every laptop vendor's firmware. The only way I could help you is I were to start hardcoding specific board vendor+model IDs and doing alternative workarounds for firmware known to be buggy.

GlobalAlliance18, 2022-02-15:
I'm, not just pulling this out of my hat - I've been trying since 20.04 to get System76 Power to manage power like you claim it will. After years of bashing together bumblebee for the nvidia management, the work youve done there is just amazing. 

And yet my XPS 9550 will not run more than an hour on even 'Powersave', and yet benchmarks like its a netbook.

Turn it all off and go back to cpufreq, OR install any other ubuntu distro, and it's back to 4+ hours, with performance matching what you set. It's just a PITA to set it.

I dont know what else to tell you. It just doesn't work mate.

GlobalAlliance18, 2022-03-14:
So its been ANOTHER month. And this STILL isn't fixed.

I've tested @mmsticks previous assertions about 'balanced performing better than high performance', and SURPRISE SURPRISE - that turned out to be completely incorrect

My testing showed consistently higher benchmarks  in 'high performance' vs 'balanced' on 4 separate machines using phoronix, geekbench and sysbench,  from 11-38% higher, INCLUDING with their own Lemur.  Not a SINGLE test on the Balanced profile performed higher. 

More to the point -  more battery life can be found in the 'battery life'  profile BUT - even MORE life comes out of using  3rd party extensions like Konkors 'cpu-freq' (roughly 8-13% more battery life in equivalent settings).

So - not only is there a clear need for the  system76 profiles to be adjusted based on user nee to achieve best performanced, but ALSO the profiles provided clearly are not optimized properly either. Or in other words - every single excuse given so far for not addressing this is complete bunk.

So - at what point is this going to actually be addressed?

Yes, you better believe I'm going to keep banging this drum - I've paid for your hardware, and donated money before that too, which I've since regretted. This is almost as disappointing as all the  ignored emails, but I've let that slide in favor of taking the technical approach here, only to find nothing but excuses and ambivalence here too, and after 4 years of this not being addressed, it's high time this was taken seriously.

jacobgkau, 2022-03-14:
I have locked the issue from further discussion because the discussion is no longer productive. PRs are always welcome from anyone who wants to implement this or any other potential improvements.

mmstick, 2022-03-14:
@GlobalAlliance18 

> turned out to be completely incorrect

Probably because I just recently made updates to power profiles to give high performance a performance boost over balanced with tweaking the scaling governors. And there's an additional PR about to be merged which tweaks the scaling governor better for AMD and older Intel CPUs.