pop-os/system76-power #15: nvidia card not shutting down

sudopluto, 2018-05-07:
- currently on pop 18.04, nvidia 396
- in intel mode, bbswitch reports that my gpu still on
- I have nouveau blacklisted since it segfaults

is there any logs I can check to see why bbswitch is not turning off the nvidia card?

sudopluto, 2018-05-07:
just checked and realized that #14 basically disabled bbswitch, will leave this bug up for people confused like me.

jackpot51, 2018-05-07:
I am working on fixing an issue with bbswitch, after which it will be reenabled.

jackpot51, 2018-05-07:
If all that is required is adding nvidia_drm.modeset=1, we could do that

sudopluto, 2018-05-12:
hey, I wanted to give this a shot this weekend, but I wanted to figure out some stuff first:

- i was thinking of adding a command like ```system76-power toggle```, which would toggle the card
power state. then the systemd service would just call that on boot. if nvidia module is loaded, calling toggle would just fail silently. this would avoid adding a file somewhere that would keep track of card power state like ubuntu-drivers does

sudopluto, 2018-05-14:
saw the update, the new commands work great as a front-end for bbswitch, though it was kind of a bummer that it didn't fix the boot issue. nvidia session was tear free due to the enablement of prime sync

msjenkins1, 2018-06-11:
I'm loving the updates myself. Everything is just so smooth with no Nvidia driver on my system, however power draw still seems high even when system76-power toggles the discrete card off. Any way to check if the card is indeed being powered down?

mmstick, 2018-06-11:
@msjenkins1 Check if the nvidia card is being reported by `lspci`.

sudopluto, 2018-06-12:
oh, as of right now, system76-power is working perfectly for me
ready to close this bug, since it doesnt really apply anymore

msjenkins1, 2018-06-12:
@mmstick `lspci`doesn't show the discrete card but I still can't force Wayland to start (which is not paramount for me) and the battery life is approximately 3 hours on this Oryx Pro with the GTX 1060, whereas with optimal results like with Ubuntu 16.04 LTS, my battery life is more like 6 hours or with the 4.17 kernel, more like 7 hours. Really impressive for a laptop this size!

Something weird though, I've never been able to see the GPU fans on lm-sensors go to 0, while the CPU fans do go to 0.

sudopluto, 2018-06-12:
do you have tlp installed? tlp fixed my issues with my fans constantly spinning

msjenkins1, 2018-06-13:
@sudopluto I hadn't tried that before; I'll give it a go.

KorvinSzanto, 2018-08-07:
So I'm no expert when it comes to video card drivers and all of that, but after burning a few days and reinstalling pop os several times I've synthesized this down to a few commands that give me reasonable battery life. It literally went from ~25w draw down to 8w which took me from an average of 2 hours battery life to ~6-10 depending on use.

```bash
sudo tlp-stat
sudo system76-power graphics power off
sudo powertop --auto-tune
```

I'm not sure if the auto-tune is required, theoretically tlp-stat should cover all that but I keep seeing things toggle back on in there so I have to keep a hawk eye on my wattage :/

closed 2018-08-08 by brs17

KorvinSzanto, 2018-08-08:
@brs17 Do you mind reopening until there's an official fix for this? I don't think it's reasonable to require tlp-stat and manual power setting on wakeup.

KinshipCourier, 2018-08-08:
@brs17 I concur with @KorvinSzanto. I was in the process of getting those instructions to work with no avail. The GPU continues to have excessive power consumption regardless of my modifications.

mmstick, 2018-08-08:
@KorvinSzanto You should create a new issue specific to your problem.