pop-os/system76-power #19: Brightness updates not reflected in slider

kylecorry31, 2018-05-13:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 0.0.1~1526005757~18.04~1222310
  Candidate: 0.0.1~1526005757~18.04~1222310
  Version table:
 *** 0.0.1~1526005757~18.04~1222310 776
        776 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
When switching power states and the brightness changes, the slider in the menu does not change.

**Steps to reproduce (if you know):**
Start in high power mode. Turn brightness to 100%, toggle to another power mode. The brightness will change but the slider will not change in the user menu.

**Expected behavior:**
The slider should reflect the current brightness level.

**Other Notes:**
gdbus call --session --dest org.gnome.SettingsDaemon.Power --object-path /org/gnome/SettingsDaemon/Power --method org.freedesktop.DBus.Properties.Set org.gnome.SettingsDaemon.Power.Screen Brightness "<int32 XX>"

Sets the brightness, where XX is the percent from 0 to 100.

mmstick, 2018-05-30:
Documentation on the DBUS API is scarce. Here's a dump from `gdbus`.

```
node /org/gnome/SettingsDaemon/Power {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };

  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };

  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };

  interface org.gnome.SettingsDaemon.Power.Screen {
    methods:
      StepUp(out i new_percentage,
             out i output_id);
      StepDown(out i new_percentage,
               out i output_id);
    signals:
    properties:
      readwrite i Brightness = 50;
  };

  interface org.gnome.SettingsDaemon.Power.Keyboard {
    methods:
      StepUp(out i new_percentage);
      StepDown(out i new_percentage);
      Toggle(out i new_percentage);
    signals:
      BrightnessChanged(i brightness,
                        s source);
    properties:
      readwrite i Brightness = 0;
  };
};
```
It seems that in order to control brightness, you have to use the `Brightness` property.

### Getting Brightness

```sh
dbus-send --session --print-reply \
    --dest="org.gnome.SettingsDaemon.Power" \
    /org/gnome/SettingsDaemon/Power \
    org.freedesktop.DBus.Properties.Get \
    string:org.gnome.SettingsDaemon.Power.Screen \
    string:Brightness
```

### Setting Brightness

```sh
dbus-send --session --print-reply \
    --dest="org.gnome.SettingsDaemon.Power" \
    /org/gnome/SettingsDaemon/Power \
    org.freedesktop.DBus.Properties.Set \
    string:org.gnome.SettingsDaemon.Power.Screen \
    string:Brightness \
    variant:int32:${PERCENT}
```

mmstick, 2018-05-30:
Documentation on the DBUS API is scarce. Here's a dump from `gdbus`.

```
node /org/gnome/SettingsDaemon/Power {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };

  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };

  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };

  interface org.gnome.SettingsDaemon.Power.Screen {
    methods:
      StepUp(out i new_percentage,
             out i output_id);
      StepDown(out i new_percentage,
               out i output_id);
    signals:
    properties:
      readwrite i Brightness = 50;
  };

  interface org.gnome.SettingsDaemon.Power.Keyboard {
    methods:
      StepUp(out i new_percentage);
      StepDown(out i new_percentage);
      Toggle(out i new_percentage);
    signals:
      BrightnessChanged(i brightness,
                        s source);
    properties:
      readwrite i Brightness = 0;
  };
};
```
It seems that in order to control brightness, you have to use the `Brightness` property.

### Checking if it exists

```sh
gdbus introspect --session --dest org.gnome.SettingsDaemon.Power \
    --object-path /org/gnome/SettingsDaemon/Power \
        || echo SettingsDaemon.Power does not exist
```

### Getting Brightness

```sh
dbus-send --session --print-reply \
    --dest="org.gnome.SettingsDaemon.Power" \
    /org/gnome/SettingsDaemon/Power \
    org.freedesktop.DBus.Properties.Get \
    string:org.gnome.SettingsDaemon.Power.Screen \
    string:Brightness
```

### Setting Brightness

```sh
dbus-send --session --print-reply \
    --dest="org.gnome.SettingsDaemon.Power" \
    /org/gnome/SettingsDaemon/Power \
    org.freedesktop.DBus.Properties.Set \
    string:org.gnome.SettingsDaemon.Power.Screen \
    string:Brightness \
    variant:int32:${PERCENT}
```