pop-os/system76-power #32: intel mode not switching to intel for Vulkan

nergdron, 2018-06-21:
Note, my laptop is an Oryx Pro.

**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 0.1.2~1527898089~18.04~038f099
  Candidate: 0.1.2~1527898089~18.04~038f099
  Version table:
 *** 0.1.2~1527898089~18.04~038f099 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
     0.1.2~1527898089~18.04~038f099~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main amd64 Packages

**Issue/Bug Description:**
when set to intel mode, opengl apps correctly run using the Intel graphics card, but Vulkan apps fail. the error message given indicates they're still trying to use the NVidia card, which is shut off:

```bash
tessa@viper:~$ system76-power graphics
intel
tessa@viper:~$ vulkaninfo 
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.70

Xlib:  extension "NV-GLX" missing on display ":0".
/build/vulkan-Kbdbga/vulkan-1.1.70+dfsg1/demos/vulkaninfo.c:2700: failed with VK_ERROR_INITIALIZATION_FAILED
```

**Steps to reproduce (if you know):**
set laptop to intel graphics, try to run Vulkan applications.

**Expected behavior:**
Applications run at reduced performance.

**Other Notes:**
It's especially strange because Vulkan should be able to handle video cards from multiple vendors present, so I'm not sure what the graphics switch app is actually doing here to break this.

msjenkins1, 2018-06-26:
I'm experiencing the same issue on an Oryx Pro as well.

KinshipCourier, 2018-09-19:
Well, I've reinstalled a few more times and it looks like the Nvidia card still isn't being powered down. I'm showing a 22 watt minimum baseline on the Intel setting. Is it just me or my system? On Ubuntu with bbswitch, I read an baseline of 8 watts. With that amount of power consumption, which is nearly identical with the Nvidia setting, it doesn't make sense to enable Optimus switching in the UEFI.

jackpot51, 2018-09-25:
Could you provide your `sudo dmidecode`?
That is certainly not expected behavior

KinshipCourier, 2018-09-25:
Sure thing! Luckily, I just reinstalled Pop_OS! with Nvidia again last night, but I hadn't checked the battery power draw yet. I unplugged and the projection is showing 2h:16m until dead on the Intel setting with about 10 minutes of stats so far.

Quite a long file, so I converted to a text file on [Firefox Send](https://send.firefox.com/download/ddbd432255/#LYvP7bJsC6cnjlyiJQbXDg) if that's alright. I do have to say that it's such a slick system and everything except this works really well. Ya'll deserve applause!

jackpot51, 2018-09-25:
We do not support graphics switching on the `oryp3`. There are numerous firmware support items that were added to the `oryp4` and `oryp4-b` that made support possible for those products, as well as other newer Optimus laptops.

The `oryp3` and previous Oryx models were sold with always-on NVIDIA graphics, and NVIDIA has not been forthcoming with information about the Optimus mechanism which was used on older models.

KinshipCourier, 2018-09-25:
Ah! Oh my goodness, that would make sense. Well, at least I can rest easy now. I was afraid I was misconfiguring something.

Thank you for your help.

jackpot51, 2018-09-25:
Sure thing!

crawfxrd, 2021-11-24:
If this is still an issue on the latest system76-power and nvidia-graphics-drivers, please open a new issue against them.

closed 2021-11-24 by crawfxrd