pop-os/system76-power #34: system76-power prevents leow9-w from shutting down when a u.2 drive is connected

brs17, 2018-08-21:
**Issue/Bug Description:**
Having the system76-power package installed, prevents the leow9-w from properly shutting down when a u.2 drive is connected.

**Steps I used to narrow down that system76-power is the package causing issues:**
1. Connect NVME drive as well as U.2 drive.
2. Install Ubuntu 18.04 from the Ubuntu website to the NVME drive.
3. Note that powering the machine off will work just fine.
4. Installed NVIDIA drivers from system76 repo.
5. Successfully power off the machine once again.
6. Add system76-power.
7. See that system does not successfully poweroff on it's own.

jackpot51, 2018-08-23:
@mmstick can you take a look?

mmstick, 2018-08-23:
Does this happen regardless of the power profile that is set when shutting down?

mmstick, 2018-08-23:
My guess is that when the daemon starts, it modifies the pstate values when it sets the balanced profile. When it gets a SIGTERM on shutting down, it retains those values, and for whatever reason the firmware is unable to power off with those values. I was thinking that perhaps we could get the values that were set before the pstate was changed, and then set them back when the daemon receives a SIGTERM signal on shutdown.