pop-os/system76-power #38: Power settings lack performance option

ppsilva, 2018-09-07:
**NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
**

**Issue/Bug Description:**
There is no way to set a performance mode. The only way to set is through the current **System76 Power extension** between:
- High Performance
- Balanced
- Battery Life

Nevertheless, it is currently **impossible** to change this extension's default to high performance when connected to electricity.

**Steps to reproduce (if you know):**
A. Settings
- Access Settings > Power
- Fail to change efficiency mode 

B. System76 Power extension
- Access extension's drop-down menu
- Change to a different value. E.G.: High performance
- Restart computer
- default goes back to Balanced

**Expected behavior:**
A. Be able to set an efficiency mode
B. Be able to set a new default

mmstick, 2018-09-12:
> The only way to set is through the current System76 Power extension

Actually, you can use the CLI client to control the DBus service:

```sh
system76-power profile performance
```

You can set that in a script if you want.

As for remembering settings between boots, that is just part of #2 will solve.

brs17, 2019-11-19:
I am going to close this as the requests here are detailed in #2 as well as other issues.

closed 2019-11-19 by brs17