pop-os/system76-power #41: Unable to switch from Nvidia to Intel

8bitbuddhist, 2018-09-13:
**Distribution (run `cat /etc/os-release`):**
 NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 0.1.2~1536704439~18.04~ad63c63
  Candidate: 0.1.2~1536704439~18.04~ad63c63
  Version table:
 *** 0.1.2~1536704439~18.04~ad63c63 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
Using `system76-power` to switch from the (active) `nvidia` mode to `intel` mode results in nothing happening, and the computer continuing to run in `nvidia` mode. This causes my laptop to heat up faster, run louder, and use more battery. 

**Steps to reproduce (if you know):**
1) Activate "Nvidia Graphics" mode
2) Select "Intel Graphics" from the system menu in Gnome, or run `sudo system76-power graphics nvidia` in a terminal
3) Reboot

**Expected behavior:**
Computer reboots into Intel graphics mode

**Other Notes:**
Test machine: Sony SVF15N18PXB
GPU: NVIDIA GK208M [GeForce GT 735M] (rev a1)

I'm not sure when the original switch from intel to nvidia mode occurred, but it was likely several weeks ago when I was testing a Steam game.

mmstick, 2018-09-13:
Anything mentioned in `journalctl /usr/bin/system76-power`?

jackpot51, 2018-09-14:
Are you running Ubuntu or Pop!_OS?

8bitbuddhist, 2018-09-16:
Pop!_OS, using the 18.04 ISO (I believe it was the Nvidia ISO, but I'm not 100% sure)

Nishan-Vivek, 2018-09-17:
Also experienced this issue when trying Pop!_OS on my MSI GE40 with Optimus GTX 760m and Intel I7-3770k graphics. 

Tried Mint 19 and the exact same thing is happening.  Looks like this is more an issue with the Ubuntu 18.04 base.

To be more specific in my case, I use either distro's built in tool to switch to intel/power saving graphics and restart when prompted. The laptop boots up with everything running fine but with the  nvidia gpu still running. I can tell  as the indicator LED on laptop stays orange (it goes blue on intel graphics) and the laptop continues to pull far more wattage and generate a lot of heat.

solBinger18, 2018-09-17:
I happened to have the same issue after running latest updates from system76. Mine is Ubuntu 18.04, Oryx Pro 2018.

8bitbuddhist, 2018-09-18:
There do seem to be several issues surrounding this in Ubuntu, not Pop:

https://bugs.launchpad.net/ubuntu/+source/nvidia-prime/+bug/1765363
https://bugs.launchpad.net/ubuntu/+source/nvidia-prime/+bug/1757180

My "solution" was to disable the GPU in BIOS and force Pop to use the Intel card, which isn't ideal but works for now. Hopefully there will be an upstream fix soon

solBinger18, 2018-09-20:
Can you please keep this open until there is a solution rather than disable GPU in BIOS?

8bitbuddhist, 2018-09-21:
> Can you please keep this open until there is a solution rather than disable GPU in BIOS?

I agree disabling the GPU isn't a real solution, but from what I've seen in my searching, this is an Ubuntu problem and not a Pop problem

solBinger18, 2018-09-21:
I am able to switch between Intel and Nvidia GPU after turning Turbo Boost back on (or reset BIOS to default). Do it and try:
`$ sudo system76-power daemon`

solBinger18, 2018-09-21:
switched to Intel GPU. nvidia GPU was turned off.

[image]

Nishan-Vivek, 2018-10-31:
I am having exactly the same behaviour (not switching off nvidia gpu when intel is selected) with pop_os! 18.10. Has it been resolved for anyone else? I wonder if it is something particular to my laptop's graphics implementation.

tchar, 2019-03-12:
Hello Guys.

I have the same problem and I also run PopOs 18.04.
Turning to Intel (from Nvidia) does not work neither with gnome-shell-extension nor with the command `sudo system76-power graphics intel`.

My `cat /etc/os-release`
```
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

Here is my `journalctl /usr/bin/system76-power`.

As you can see there it seems to be an error with a service that does not exist:
```
Mar 12 12:25:22 pop-os system76-power[910]: [INFO] DBUS Received GetGraphics method
Mar 12 12:25:23 pop-os system76-power[910]: [INFO] DBUS Received SetGraphics(intel) method
Mar 12 12:25:23 pop-os system76-power[910]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Mar 12 12:25:23 pop-os system76-power[910]: [INFO] Disabling nvidia-fallback.service
Mar 12 12:25:23 pop-os system76-power[910]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
Mar 12 12:25:23 pop-os system76-power[910]: [ERROR] systemctl: failed with exit code: 1
Mar 12 12:25:23 pop-os system76-power[910]: [INFO] Updating initramfs
Mar 12 12:25:23 pop-os system76-power[910]: update-initramfs: Generating /boot/initrd.img-4.18.0-16-generic
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Config    : INFO     Looking for configuration...
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub           : INFO     System information:
Mar 12 12:25:30 pop-os system76-power[910]:     OS:..................Pop!_OS 18.04
Mar 12 12:25:30 pop-os system76-power[910]:     Root partition:....../dev/dm-1
Mar 12 12:25:30 pop-os system76-power[910]:     Root FS UUID:........1ecc7822-ac8d-49ab-b8e3-849d5bbfb181
Mar 12 12:25:30 pop-os system76-power[910]:     ESP Path:............/boot/efi
Mar 12 12:25:30 pop-os system76-power[910]:     ESP Partition:......./dev/nvme0n1p1
Mar 12 12:25:30 pop-os system76-power[910]:     ESP Partition #:.....1
Mar 12 12:25:30 pop-os system76-power[910]:     NVRAM entry #:.......-1
Mar 12 12:25:30 pop-os system76-power[910]:     Boot Variable #:.....0000
Mar 12 12:25:30 pop-os system76-power[910]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
Mar 12 12:25:30 pop-os system76-power[910]:     Kernel Image Path:.../vmlinuz
Mar 12 12:25:30 pop-os system76-power[910]:     Initrd Image Path:.../initrd.img
Mar 12 12:25:30 pop-os system76-power[910]:     Force-overwrite:.....False
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Copying Kernel into ESP
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Backing up old kernel
Mar 12 12:25:30 pop-os system76-power[910]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
```

Seems like the problem is:
`Mar 12 12:25:23 pop-os system76-power[910]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.`

Should you need any other info about my system please tell me

fmarier, 2019-04-29:
I filed https://github.com/pop-os/system76-power/issues/76 for the problem that @tchar and I are seeing.

danger89, 2019-05-05:
I'm also unable to switch between Intel <> Graphics (first time it tried it, and worked actually (although PopOs gave an error it was unable to switch to Intel Graphics, since I saw the led icon change on my laptop.. but from that moment on I'm unable to switch at all).

Log (same log as `journalctl /usr/bin/system76-power`) :
```sh
sudo system76-power daemon
[INFO] Starting daemon
[INFO] Connecting to dbus system bus
[INFO] Rescanning PCI bus
[INFO] 0000:01:00.0: NVIDIA graphics
[INFO] 0000:01:00.0: Function for 0000:01:00.0
[INFO] 0000:00:02.0: Intel graphics
[INFO] 0000:00:02.0: Function for 0000:00:02.0
[INFO] Disabling NMI Watchdog (for kernel debugging only)
[INFO] Setting automatic graphics power
[INFO] Disabling graphics power
[INFO] 0000:01:00.0: Removing
[INFO] Initializing with the balanced profile
[ERROR] Failed to set initial profile: Operation not permitted (os error 1)
[INFO] Registering dbus name com.system76.PowerDaemon
[INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
[INFO] hwmon: acpitz
[INFO] hwmon: coretemp
[ERROR] fan daemon: platform hwmon not found
[INFO] Handling dbus requests
```

Extra distro info:
```
cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.04
DISTRIB_CODENAME=disco
DISTRIB_DESCRIPTION="Pop!_OS 19.04"
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
```

mbeltagy, 2019-06-13:
I have the same problem on my Oryx Pro.  I am now totally locked in NVIDIA.

toefel18, 2019-06-26:
I have the same problem. And the laptop gets really hot and makes a lot of noise

krthkj, 2019-08-08:
Following error keeps appearing and by looking at the code, seems like the fan hwmod for laptop is not configured. I suspect this may be the reason for fan noise and laptop heat.
```
[INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
[ERROR] fan daemon: platform hwmon not found
[INFO] Handling dbus requests
```

mbeltagy, 2019-08-08:
I got the following advice from System76 customer support and it worked quite well for me 
```
Let's reinstall the NVIDIA Driver with these commands:

sudo apt purge nvidia* && sudo apt autoremove
sudo apt update
sudo apt install system76-driver-nvidia
reboot

If that doesn't fix the issue we can reinstall the packages and then test again:

sudo apt install --reinstall gnome-shell-extension-system76-power system76-power
```

toefel18, 2019-08-08:
I also did those commands and that fixes it for a couple of times, but sometimes is breaks again.  My set-up also logs:
```
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] Connecting to dbus system bus
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] Registering dbus name com.system76.PowerDaemon
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] hwmon: coretemp
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] hwmon: pch_cannonlake
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] hwmon: acpitz
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] hwmon: iwlwifi
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] hwmon: system76
Dec 11 17:14:51 pop-os system76-power[943]: [ERROR] fan daemon: platform hwmon not found
Dec 11 17:14:51 pop-os system76-power[943]: [INFO] Handling dbus requests
```
Wondering if that `fan daemon: platform hwmon not found` causes problems

danger89, 2019-08-08:
https://www.youtube.com/watch?v=_36yNWw_07g

mmstick, 2019-08-08:
> Wondering if that `fan daemon: platform hwmon not found` causes problems

This is normal on systems without a platform hwmon

neurolabusc, 2019-11-09:
I also have this issue with my Oryx Pro running PopOS. Our other Oryx Pro burned out its NVidia graphics card, so I would really like to only use the GPU when I need CUDA. Further the NVidia GPU seems to cause the fan to run at a constant loud setting, which is not good for a quiet office nor good battery life. Any news on a fix? I upgraded the ROM but this has not helped.

miguelcarcamov, 2019-11-22:
Same issue, but backwards. I mean I cannot switch to nvidia from intel, when at sometime I switched to nvidia

crawfxrd, 2019-11-22:
Is anyone able to reproduce this issue?

Both nvidia and nouveau are blacklisted in Intel mode via `/etc/modprobe.d/system76-power.conf`, so I'm interested in knowing which set is loaded (and how).

the-hacker-industry, 2020-02-04:
I had the same problem and I solved it. Edit **/etc/default/grub** and replace **quiet splash** with **nouveau.modeset=0** and then run **sudo update-grub** and now reboot the system. You'll have now Intel graphics. To switch Nvidia vice versa this process

sjatkins, 2020-03-17:
All of the above is in my mind workarounds for something obviously quite broken. If I had realized this system had such hybrid grapbics that are known to have problems from when I first run into them when converting windows machines to linux years ago I would never have purchased it.   It is not just pop!OS but it is a bad way to do things.  I can't even run an external monitor on my lovely maxed out Adder without switching to nvidia which of course runs fans loud continuously.   I am very disappointed.

ZacharyRSmith, 2020-04-12:
I also could not change from NVIDIA to Intel.  Then, I ran all my updates.  Then, I was able to change from NVIDIA to Intel.  I'm guessing due to an OS update.

crawfxrd, 2021-11-24:
If this is still an issue on the latest system76-power and nvidia-graphics-drivers, please open a new issue.

closed 2021-11-24 by crawfxrd