pop-os/system76-power #44: Instructions for building a .deb would be nice

terabyte, 2018-09-14:
I am trying to use this code with a custom Debian install, I fetched and successfully built the code but there is no way to use it since it needs various service files and stuff set up due to dbus tomfoolery.  Since it can't work without install scripts, anyone who wants to use it has to build a .deb.  Could we add a make target for that or something?

terabyte, 2018-09-14:
My current attempt to build (after a successful make):
```
dpkg-buildpackage -us -uc
dpkg-buildpackage: info: source package system76-power
dpkg-buildpackage: info: source version 0.1.2
dpkg-buildpackage: info: source distribution bionic
dpkg-buildpackage: info: source changed by Jeremy Soller <jeremy@system76.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build system76-power
 fakeroot debian/rules clean
dh clean --with=systemd
   debian/rules override_dh_auto_clean
make[1]: Entering directory '/home/cmyers/projects/system76-power'
make clean
make[2]: Entering directory '/home/cmyers/projects/system76-power'
cargo clean
make[2]: Leaving directory '/home/cmyers/projects/system76-power'
make vendor
make[2]: Entering directory '/home/cmyers/projects/system76-power'
mkdir -p .cargo
cp vendor_config .cargo/config
cargo vendor
error: no such subcommand: `vendor`
make[2]: *** [Makefile:38: vendor] Error 101
make[2]: Leaving directory '/home/cmyers/projects/system76-power'
make[1]: *** [debian/rules:8: override_dh_auto_clean] Error 2
make[1]: Leaving directory '/home/cmyers/projects/system76-power'
make: *** [debian/rules:4: clean] Error 2
dpkg-buildpackage: error: fakeroot debian/rules clean subprocess returned exit status 2

```

terabyte, 2018-09-14:
also suggest a "make install-deps" or something which apt-get installs all dependencies.

jackpot51, 2018-09-14:
This will add the `cargo vendor` command:

```
cargo install cargo-vendor
```