pop-os/system76-power #48: system76-power daemon unable to start if laptop is on battery mode from start

hongcheng79, 2018-10-05:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.0.0~1537808410~18.04~821cc23
  Candidate: 1.0.0~1537808410~18.04~821cc23
  Version table:
 *** 1.0.0~1537808410~18.04~821cc23 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
     1.0.0~1537808410~18.04~821cc23~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main amd64 Packages

**Issue/Bug Description:**
system76-power daemon unable to start if laptop is on battery from start with the following message
[INFO] Starting daemon
[INFO] Rescanning PCI bus
[INFO] 0000:00:02.0: Intel graphics
[INFO] 0000:01:00.0: NVIDIA graphics
[INFO] Disabling NMI Watchdog (for kernel debugging only)
[INFO] Setting automatic graphics power
[INFO] Enabling graphics power
[INFO] Initializing with the balanced profile
[ERROR] failed to set initial profile: Operation not permitted (os error 1)

This is cause the ability to switch from intel to nvidia or vice versa unable to function as the daemon could not be started.

sudo systemctl status system76-power.service
● system76-power.service - System76 Power Daemon
   Loaded: loaded (/lib/systemd/system/system76-power.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-10-06 00:06:55 +08; 806ms ago
  Process: 7821 ExecStart=/usr/bin/system76-power daemon (code=exited, status=1/FAILURE)
 Main PID: 7821 (code=exited, status=1/FAILURE)

Oct 06 00:06:55 chc-predator systemd[1]: system76-power.service: Service hold-off time over, scheduling restart.
Oct 06 00:06:55 chc-predator systemd[1]: system76-power.service: Scheduled restart job, restart counter is at 5.
Oct 06 00:06:55 chc-predator systemd[1]: Stopped System76 Power Daemon.
Oct 06 00:06:55 chc-predator systemd[1]: system76-power.service: Start request repeated too quickly.
Oct 06 00:06:55 chc-predator systemd[1]: system76-power.service: Failed with result 'exit-code'.
Oct 06 00:06:55 chc-predator systemd[1]: Failed to start System76 Power Daemon.

**Steps to reproduce (if you know):**
Start the laptop on battery

**Expected behavior:**
system76-power daemon should be able to start like previous version and ability to switch intel to nvidia or nvidia to intel fully function

**Other Notes:**

c8r821, 2018-10-28:
> I am also running into this issue but without the condition of starting up on battery. I'm running a Lenovo Yoga 720 15" w/ a GTX 1050 and a clean install of Pop 18.10. On every boot system-76 power fails to start and I am left unable to toggle the GPU on and off.
> 
> ```
> carter@pop-os:~$ system76-power --version
> system76-power 1.0.0
> ```
> ```
> carter@pop-os:~$ sudo systemctl status system76-power
> ● system76-power.service - System76 Power Daemon
>    Loaded: loaded (/lib/systemd/system/system76-power.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Sat 2018-10-27 11:19:38 CDT; 5min ago
>   Process: 1523 ExecStart=/usr/bin/system76-power daemon (code=exited, status=1/FAILURE)
>  Main PID: 1523 (code=exited, status=1/FAILURE)
> 
> Oct 27 11:19:38 pop-os systemd[1]: system76-power.service: Service RestartSec=100ms expired, scheduling restart.
> Oct 27 11:19:38 pop-os systemd[1]: system76-power.service: Scheduled restart job, restart counter is at 5.
> Oct 27 11:19:38 pop-os systemd[1]: Stopped System76 Power Daemon.
> Oct 27 11:19:38 pop-os systemd[1]: system76-power.service: Start request repeated too quickly.
> Oct 27 11:19:38 pop-os systemd[1]: system76-power.service: Failed with result 'exit-code'.
> Oct 27 11:19:38 pop-os systemd[1]: Failed to start System76 Power Daemon.
> ```
> ```
> carter@pop-os:~$ sudo system76-power daemon --verbose
> [INFO] Starting daemon
> [INFO] Rescanning PCI bus
> [INFO] 0000:01:00.0: NVIDIA graphics
> [INFO] 0000:00:02.0: Intel graphics
> [INFO] Disabling NMI Watchdog (for kernel debugging only)
> [DEBUG] Modifying kernel parameter at "/proc/sys/kernel/nmi_watchdog" to 0
> [INFO] Setting automatic graphics power
> [INFO] Enabling graphics power
> [INFO] Initializing with the balanced profile
> [ERROR] failed to set initial profile: Operation not permitted (os error 1)
> ```

Aaaand it's working now. I can't decide if I've done anything that would've fixed it but after a few days and various reboots it's now working properly. If you are interested here are those same commands run now.

```
carter@Andromeda:~$ sudo systemctl status system76-power
● system76-power.service - System76 Power Daemon
   Loaded: loaded (/lib/systemd/system/system76-power.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-10-28 10:34:23 CDT; 6h ago
 Main PID: 1058 (system76-power)
    Tasks: 1 (limit: 4915)
   Memory: 1.9M
   CGroup: /system.slice/system76-power.service
           └─1058 /usr/bin/system76-power daemon

Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Enabling graphics power
Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Initializing with the balanced profile
Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Connecting to dbus system bus
Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Registering dbus name com.system76.PowerDaemon
Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oct 28 10:34:23 andromeda system76-power[1058]: [INFO] Handling dbus requests
Oct 28 10:34:37 andromeda system76-power[1058]: [INFO] DBUS Received GetSwitchable method
Oct 28 10:34:37 andromeda system76-power[1058]: [INFO] DBUS Received GetGraphics method
Oct 28 15:05:12 andromeda system76-power[1058]: [INFO] DBUS Received GetSwitchable method
Oct 28 15:05:12 andromeda system76-power[1058]: [INFO] DBUS Received GetGraphics method
```
```
carter@Andromeda:~$ sudo system76-power daemon --verbose
[INFO] Starting daemon
[INFO] Rescanning PCI bus
[INFO] 0000:01:00.0: NVIDIA graphics
[INFO] 0000:00:02.0: Intel graphics
[INFO] Disabling NMI Watchdog (for kernel debugging only)
[DEBUG] Modifying kernel parameter at "/proc/sys/kernel/nmi_watchdog" to 0
[INFO] Setting automatic graphics power
[INFO] Enabling graphics power
[INFO] Initializing with the balanced profile
[ERROR] failed to set initial profile: Operation not permitted (os error 1)
```

closed 2019-03-25 by mmstick

hongcheng79, 2019-04-17:
Just to let you guys know the latest update solve this issue

PR #67 (2019-03-25): Allow the initial profile on startup to fail
This will fix #48