pop-os/system76-power #50: system76-power automatically disable turbo boost in training

drNoob13, 2018-10-31:
**Distribution (run `cat /etc/os-release`): **
 Ubuntu 18.04.1 LTS
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
$ apt policy $PACKAGE system76-power
system76-power:
  Installed: 1.0.0~1537808410~18.04~821cc23~dev
  Candidate: 1.0.0~1537808410~18.04~821cc23~dev
  Version table:
 *** 1.0.0~1537808410~18.04~821cc23~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**
system76-power automatically disables turbo boost while in training data with DeepLab (TensorFlow)

before turbo boost disabled by system76-power
[image]

after turbo boost disabled by system76-power
[image]

**Steps to reproduce (if you know):**
1. Reduce max_perf_pct to `70%`
```
$ echo 70 | sudo tee /sys/devices/system/cpu/intel_pstate/max_perf_pct
```
```
$ sudo tlp-stat -p | egrep 'min_perf_pct|max_perf_pct|no_turbo|/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor'
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  19 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      =  70 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0`
```

2. Run turbostat to monitor the frequency, temp and power consumption
```
$ watch -n 2 sudo turbostat --quiet --show CPU --show Bzy_MHz --show Busy%  --show PkgWatt --show PkgTmp --show PKG_%   sleep 0
```

3. Run deeplab train.py with more than 3000 steps, i.e. change the `NUM_ITERATIONS` in `local_test.sh` from `10` to `3000` and run it
[Ref: DeepLab Installation](https://github.com/tensorflow/models/blob/master/research/deeplab/g3doc/installation.md
)

**Expected behavior:**
I understand this is the CPU protection mechanism from overheat under high workload. But the temperature and power consumption readings in this case are all good (less than `80C`). In my opinion, system76-power should not automatically disable turbo boost under heavy CPU loads if the value read from `/sys/devices/system/cpu/intel_pstate/max_perf_pct` is under `85%`.

**Other Notes:**

PC spec
```
Oryx Pro 4
OS: Ubuntu 18.04.1 LTS x86_64
Kernel: 4.15.0-38-generic
CPU: Intel i7-8750H (12) @ 4.100GHz
GPU: NVIDIA GeForce GTX 1070 Mobile
Memory: 32 GB

```

jackpot51, 2018-10-31:
The current rules are that performance and balanced mode will allow turbo boost, and battery mode will not.

Are you using battery mode when running these programs? I recommend using balanced mode.

drNoob13, 2018-10-31:
I have used both balanced mode and performance mode with `max_perf_pct` manually set to `70%` to avoid overheating. I know you disable turbo boost in battery mode.

The below video should give a better illustration. Turbo boost was auto disabled at 0:40 when there was a spike of consumed power.

[https://youtu.be/mUtUI6bIDnI](https://youtu.be/mUtUI6bIDnI
)

jackpot51, 2018-10-31:
Thanks for the video. I will check to see what may have caused that

djordan2, 2018-10-31:
You don't happen to be running Tensorflow on the NVIDIA GPU on a 15" Oryx Pro 4, do you?

djordan2, 2018-10-31:
You could try stopping the system76-daemon process and then enabling turbo boost.  That is most likely what is affecting your system.  If you do, make sure to stop your GPU tensorflow job before unplugging from power.

drNoob13, 2018-11-01:
@djordan2 The gtx 1070 was almost fully utilized, and the laptop was plugged in all the time.

In the video at 0:43, it is shown on the left terminal that a tensor device was created on the GTX 1070 MAX-Q allocating 5961 MB of the GPU memory. Also on the top bar, next to the thermostat icon, shows 2 pairs of performance stats (CPU temp., CPU utilization), (GPU temp., GPU utilization) reported by Psensor. At 0:43, 96% of the 1070 GPU was reportedly utilized.

Please feel free to let me know if you need any more info.

drNoob13, 2018-11-01:
> You could try stopping the system76-daemon process and then enabling turbo boost

@djordan2 That's a good idea. However, I haven't been able to stop the system76 daemon unless I uninstall system76-power, which I don't want to since I found system76-power very useful in switching the nvidia/intel graphics.

Perhaps, you could point me in the right way to stop it. What I have tried so far:

```
sudo service system76-power stop
sudo systemctl disable system76-daemon
ps -ef | grep system76-daemon # then sudo kill -9 <s76-PID>
```

jackpot51, 2018-11-01:
```
sudo systemctl disable system76-driver.service
```

drNoob13, 2018-11-03:
Thanks. I find that the combination of the following two commands prevents system76-power from automatically turning turbo boost off under high load:

```
sudo systemctl disable system76-driver.service
sudo service system76-power stop
```

However, this also makes my CPU0 (0-11) fully utilized in iddle. Trying to figure out what's going on in the background, I ran my monitoring toolkit to create the image below.
[image]

This is an issue on its own. I also happens when I tried to uninstall system76-power by `sudo apt remove system76-power`.

djordan2, 2018-11-06:
@drNoob13 Can you figure out what process is causing the high cpu usage?

drNoob13, 2018-11-10:
@djordan2 sorry for the late response. Today I tried to disable system76 driver service again with [above commands](https://github.com/pop-os/system76-power/issues/50#issuecomment-435563624), but I couldn't (i.e. turbo boost got disabled again under high CPU+GPU load). Btw, I have also seen the System76 updates started a day or two ago. Let me see what's going on.

drNoob13, 2018-11-13:
@jackpot51 @djordan2 
New [video](https://youtu.be/51BLioy0YlM): no turbo boost auto disable.

CPU1 usage ~100% in idle: gnome-shell and Xorg seem to be the culprits. Also, the GTX 1070 usage in idle is unexpectedly high (~6%-8%); it was usually 1%-2%.
[image]

drNoob13, 2018-11-14:
@djordan2 : in the screenshot in my last post, the processes that causes CPU1 100%  : 
```bash
/usr/bin/gnome-shell
/usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background
```
Look like, system76-driver being disabled really affects the 1070 GPU display, and the job is transferred to the iGPU (with CPU1?) somehow overloading it.

Question: now after boot in nvidia graphics mode `system76-power graphics` --> `nvidia`, CPU1 is always occupied like that. Is it intended? 

With nvidia GPU completely disable, i.e., Intel graphcis mode, there is no high CPU usage in idle. See image below

[image]

drNoob13, 2018-11-22:
Added debug screenshot when laptop connects to 4K monitor.

Due to ~100% utilization of CPU1, its temperature is 7-10C higher than the other CPU.

[image]

djordan2, 2018-11-26:
Okay, I can reproduce this.  The cpu usage appears to be related to excessive triggering of a gpe interrupt, but went away after a reboot.  Suppressing the interrupt also resolved the CPU usage.