pop-os/system76-power #51: Nvidia GPU not disabled when switching to Intel Graphics (MSI GE40 laptop)

Nishan-Vivek, 2018-10-31:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="18.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.10"
VERSION_ID="18.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.0.0~1537808410~18.10~821cc23
  Candidate: 1.0.0~1537808410~18.10~821cc23
  Version table:
 *** 1.0.0~1537808410~18.10~821cc23 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu cosmic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
NVIDIA GPU is not disabled when switching to Intel Graphics and restarting. Intel driver is being used but LED indicator shows the Nvidia GPU is still enabled and power draw remains high (verified in powertop)

Laptop: MSI GE40
CPU: Intel(R) Core(TM) i7-4702MQ
GPU: Nvidia GTX 760M

**Steps to reproduce (if you know):**
Install nvidia 410 drivers via popshop. 
Switch to intel graphics mode via gnome power settings drop down and restart. 

**Expected behavior:**
Nvidia GPU should be disabled. The laptop LED indicator should be blue and power draw lower. 

**Other Notes:**
GPU appears to be correctly disabled when the default nouveau drivers are used as the LED indicator goes blue and power draw/heat generation is as expected. 

Journalctl /usr/bin/system76-power
```
-- Reboot --
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Starting daemon
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Rescanning PCI bus
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] 0000:01:00.0: NVIDIA graphics
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] 0000:00:02.0: Intel graphics
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Setting automatic graphics power
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Disabling graphics power
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] 0000:01:00.0: Removing
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Initializing with the balanced profile
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Connecting to dbus system bus
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Registering dbus name com.system76.PowerDaemon
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oct 31 14:53:04 pop-os system76-power[1146]: [INFO] Handling dbus requests
Oct 31 14:53:24 pop-os system76-power[1146]: [INFO] DBUS Received GetSwitchable method
Oct 31 14:53:24 pop-os system76-power[1146]: [INFO] DBUS Received GetGraphics method
Oct 31 14:56:52 pop-os system76-power[1146]: [INFO] DBUS Received SetGraphics(nvidia) method
Oct 31 14:56:52 pop-os system76-power[1146]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Oct 31 14:56:52 pop-os system76-power[1146]: [INFO] Enabling nvidia-fallback.service
Oct 31 14:56:52 pop-os system76-power[1146]: Created symlink /etc/systemd/system/multi-user.target.wants/nvidia-fallback.service → /lib/systemd/system/nvidia-fallback.service.
Oct 31 14:56:52 pop-os system76-power[1146]: [INFO] Updating initramfs
Oct 31 14:56:52 pop-os system76-power[1146]: update-initramfs: Generating /boot/initrd.img-4.18.0-10-generic
Oct 31 14:56:57 pop-os system76-power[1146]: cryptsetup: WARNING: Resume target cryptswap uses a key file
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Config    : INFO     Looking for configuration...
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub           : INFO     System information:
Oct 31 14:57:13 pop-os system76-power[1146]:     OS:..................Pop!_OS 18.10
Oct 31 14:57:13 pop-os system76-power[1146]:     Root partition:....../dev/dm-1
Oct 31 14:57:13 pop-os system76-power[1146]:     Root FS UUID:........e86d005e-7d62-4a00-b374-5c83869d5ac8
Oct 31 14:57:13 pop-os system76-power[1146]:     ESP Path:............/boot/efi
Oct 31 14:57:13 pop-os system76-power[1146]:     ESP Partition:......./dev/sdc1
Oct 31 14:57:13 pop-os system76-power[1146]:     ESP Partition #:.....1
Oct 31 14:57:13 pop-os system76-power[1146]:     NVRAM entry #:.......-1
Oct 31 14:57:13 pop-os system76-power[1146]:     Boot Variable #:.....0000
Oct 31 14:57:13 pop-os system76-power[1146]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
Oct 31 14:57:13 pop-os system76-power[1146]:     Kernel Image Path:.../vmlinuz
Oct 31 14:57:13 pop-os system76-power[1146]:     Initrd Image Path:.../initrd.img
Oct 31 14:57:13 pop-os system76-power[1146]:     Force-overwrite:.....False
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Copying Kernel into ESP
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Backing up old kernel
Oct 31 14:57:13 pop-os system76-power[1146]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
-- Reboot --
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Starting daemon
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Rescanning PCI bus
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] 0000:01:00.0: NVIDIA graphics
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] 0000:00:02.0: Intel graphics
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Setting automatic graphics power
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Enabling graphics power
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Initializing with the balanced profile
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Connecting to dbus system bus
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Registering dbus name com.system76.PowerDaemon
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oct 31 14:58:20 pop-os system76-power[1161]: [INFO] Handling dbus requests
Oct 31 14:58:33 pop-os system76-power[1161]: [INFO] DBUS Received GetSwitchable method
Oct 31 14:58:33 pop-os system76-power[1161]: [INFO] DBUS Received GetGraphics method
```

When Intel graphics are selected:
```
inxi -G
Graphics:  Device-1: Intel 4th Gen Core Processor Integrated Graphics driver: i915 v: kernel 
           Display: x11 server: X.Org 1.20.1 driver: modesetting,nvidia unloaded: fbdev,nouveau,vesa tty: N/A 
           OpenGL: renderer: Mesa DRI Intel Haswell Mobile v: 4.5 Mesa 18.2.2 
```

When Nvidia graphics are selected:
```
inxi -G
Graphics:  Device-1: Intel 4th Gen Core Processor Integrated Graphics driver: i915 v: kernel 
           Device-2: NVIDIA GK106M [GeForce GTX 760M] driver: nvidia v: 410.57 
           Display: x11 server: X.Org 1.20.1 driver: modesetting,nvidia unloaded: fbdev,nouveau,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: GeForce GTX 760M/PCIe/SSE2 v: 4.6.0 NVIDIA 410.57 
```

Luke-Nukem, 2019-04-14:
If this is anything like my old MSI GS70 then you will need bbswitch installed and set up too. You're quite likely to also need it to re-enable the GPU before standby or hibernation or neither will work.

You should follow the bbswitch guide [here](https://wiki.archlinux.org/index.php/Bumblebee#Power_management).

Note: It seems on newer machines once system76-power has managed to unbind and remove the nvidia bits the card turns off on its own (tested on MSI GS65)

Luke-Nukem, 2019-04-14:
@Nishan-Vivek additionally to the above, you may have issues with standby and shutdown when using system76-power *if* the card is required to be powered on before those actions - `bbswitch` will be unable to power the card on due to the interface being unbound.

crawfxrd, 2021-11-24:
If this is still an issue on the latest system76-power and nvidia-graphics-drivers, please open a new issue.

closed 2021-11-24 by crawfxrd