pop-os/system76-power #56: Switching power profiles will change display backlight

hjchai, 2018-12-10:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="18.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.10"
VERSION_ID="18.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.0.0~1544027737~18.10~413bafe
  Candidate: 1.0.0~1544027737~18.10~413bafe
  Version table:
 *** 1.0.0~1544027737~18.10~413bafe 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu cosmic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
According to [Wiki](https://github.com/pop-os/system76-power/wiki), switching power profiles between balanced and performance should not change display backlight (The wiki even provides a section discussing why this should be the case).  I totally agree with this setting.

However, on my laptop, system76-power doesn't behave as it is supposed to do. 

- When switch from performance to balanced, the display backlight is dimmed to a medium-low level from what ever I was using when I was in performance mode. 
- When switch from balanced up to performance, the display backlight is not changed. This case is working properly just like the wiki says.
- When switch to battery life mode from either performance or balanced, the light is automatically dimmed to low level to save battery life. This case is also working properly.

**Steps to reproduce (if you know):**

**Expected behavior:**
Switch power profiles between performance and balanced should not modify screen backlight like what is suggested in the wiki.

**Other Notes:**

brs17, 2019-01-02:
I can confirm this issue.

WatchMkr, 2019-01-04:
This is actually working as intended. The wiki was not updated with design decision changes and reasoning. We're working on the system76-power roadmap and will update documentation during the process.