pop-os/system76-power #57: Switch GPU without reboot

JeffLabonte, 2018-12-20:
**Distribution (run `cat /etc/os-release`):**

NAME="Pop!_OS"
VERSION="18.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.10"
VERSION_ID="18.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power:
  Installed: 1.0.0~1545254257~18.10~d861dfd
  Candidate: 1.0.0~1545254257~18.10~d861dfd
  Version table:
 *** 1.0.0~1545254257~18.10~d861dfd 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu cosmic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**

It would be great to be able to switch GPU without needing to reboot the PC.

There is an implementation that has been made by [MathieuGras](https://github.com/matthieugras/Prime-Ubuntu-18.04)

fliepeltje, 2019-04-13:
Alternatively, if this is not possible, it would also be great to have separate boot options for nvidia/intel in the bootmanager that you can select when powering on.

JeffLabonte, 2019-04-20:
It is possible. It is a feature that ubuntu has, also I have already used other alternatives like nvidia-xrun.

JeffLabonte, 2019-05-21:
@mmstick you should have a look on [this implementation](https://github.com/Askannz/optimus-manager).

I am currently running Arch on my laptop and I assure you that it works like a charm ( I am running a XPS 15 9560, a relatively new laptop).

mlncn, 2019-06-12:
Yes, please!  (Likewise to at least as a boot option.)

crawfxrd, 2019-10-25:
**Intel -> NVIDIA|Hybrid**:
```
system76-power graphics <hybrid|nvidia>
system76-power graphics power on
systemctl restart gpu-manager
# Logout and login
```

**NVIDIA <-> Hybrid**:
```
system76-power graphics <hybrid|nvidia>
systemctl restart gpu-manager
# Log out and log in
```

**NVIDIA|Hybrid -> Intel**
Not friendly, but works.
```
system76-power graphics intel
systemctl restart display-manager
# Log in
```

The dGPU is still present on the bus, but *should* be in a suspended state.
```
$ cat /sys/bus/pci/devices/0000\:01\:00.0/power/runtime_status
suspended
```

As an extra measure, it can be removed
```
system76-power graphics power off
systemctl restart gpu-manager
```

JeffLabonte, 2019-11-06:
@crawfxrd, those steps works very well.. the system76-power software is really well done.

crawfxrd, 2019-11-26:
I'm seeing a refcount issue that requires running gpu-manager and logging out twice after switching to Intel in order to unload the nvidia driver.

goldcoders, 2019-12-13:
@JeffLabonte  Hi may i know how you implement this on your end?
Im using also archlinux , but i wanna use this also...

I havent installed any proprietary software like nvdia etc
basically my graphics as of now is off,
can you give brief instructions like install nvdia , do this and do that...
thanks

JeffLabonte, 2020-02-15:
@goldcoders Sorry for the late response, I used optimus-manager from the AUR. I usually set the switching to none so it is properly managed by Linux.

kikislater, 2020-03-20:
> @goldcoders Sorry for the late response, I used optimus-manager from the AUR. I usually set the switching to none so it is properly managed by Linux.

@JeffLabonte But it doesn't fully disable dGPU when you are on intel like they do in system76.  So it will still consume a lot of power. On an MSI PS63, I need to to it manually on arch and reach power drop : system power move from 14-15W to 7W ...

chunga2020, 2020-04-28:
> ### Intel -> NVIDIA|Hybrid
> ```
> system76-power graphics <hybrid|nvidia>
> system76-power graphics power on
> systemctl restart display-manager
> # Login
> ```
> 
> **Note**: If you switch to NVIDIA first then Hybrid later, without rebooting, the run-time power management rule will not be applied to the loaded driver. I'm unsure of the soundness of it, but adding it to the NVIDIA modprobe file may be used as a workaround so it's always applied when the drivers are loaded (even NVIDIA-only mode).
> 
> ```
> echo "options nvidia NVreg_DynamicPowerManagement=0x02" >> /lib/modprobe.d/nvidia-graphics-drivers.conf
> ```
> 
> ### NVIDIA <-> Hybrid
> ```
> system76-power graphics <hybrid|nvidia>
> systemctl restart gpu-manager
> # Log out and log in
> ```
> 
> ### NVIDIA|Hybrid -> Intel
> ```
> system76-power graphics intel
> systemctl restart display-manager
> # Log in
> ```
> 
> The dGPU is still present on the bus. If the runtime power management workarounds are applied, it will be suspended.
> 
> ```
> $ cat /sys/bus/pci/devices/0000\:01\:00.0/power/runtime_status
> suspended
> ```
> 
> As an extra measure, it can be removed.
> 
> ```
> # Drivers *must* be unloaded!
> system76-power graphics power off
> ```

These commands work well, but is there a way to switch GPUs without having to restart my session? It's not necessarily a deal-breaker, but it _would_ be nice.

crawfxrd, 2020-04-28:
That's the use case for hybrid graphics mode. Besides that, no, you have to restart the session.

chunga2020, 2020-04-30:
> That's the use case for hybrid graphics mode. Besides that, no, you have to restart the session.

Ah gotcha. Never really set it to hybrid mode before.

crawfxrd, 2020-09-28:
GDM session on vt1 is no longer kept alive on 3.38. `display-manager` should no longer need to be killed.

Still issues with switching. In particular, switching from NVIDIA/hybrid back to compute/integrated.

JeffLabonte, 2021-01-21:
The solution you brought up @crawfxrd is not working anymore on Pop_OS! 20.10. I will probably need to remove or add the `nvidia` modules manually while having `display-manager` killed.

JeffLabonte, 2021-01-22:
I tested to switch from `integrated` to `hybrid` or `Nvidia`.

Results:
* integrated -> Hybrid
  * The interface becomes slow as if the intel is not rendering correctly.
* integrated -> Nvidia
  * It works well so far.
  
Steps:
* Switch to a different TTY
* `sudo systemctl stop display-manager`
* Power on the discrete GPU -> `sudo system76-power graphics power on`
* Load kernel modules:
   * `sudo modprobe nvidia`
   * `sudo modprobe nvidia_drm`
   * `sudo modprobe nvidia_modeset`
   * `sudo modprobe nvidia_uvm`
* `sudo systemctl start display-manager`
* Log in
* `nvidia-smi`

[image]

I will test the other way around later today.

tailg83r, 2021-03-01:
Does the system76-power work with 2 discrete graphic cards in a desktop? I have one AMD and one Nvidia and I would like to let the Nvidia only to compute (ML), and be able to disable the GPU when not needed. Thanks

crawfxrd, 2021-12-10:
Switching between hybrid and nvidia modes can be done without requiring a reboot. It would still require logging out and logging back in though, as it requires changing the X configuration (whether `PrimaryGPU` is set).

I'm not sure about compute though, because nvidia-drm is still in use.