pop-os/system76-power #68: Graphics Switching Option Does Not Appear In Pop 19.04 on Oryx 4

isantop, 2019-03-18:
**Distribution (run `cat /etc/os-release`):**

```
NAME="Pop!_OS"
VERSION="19.04"
```
 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power 1.0.0~1551755982~19.04~f2d01db

**Issue/Bug Description:**

Graphics Switcher does not show up on Oryx 4 Laptop.

**Steps to reproduce (if you know):**
Install 19.04.
Reboot
Check power menu

**Expected behavior:**
Power Menu shows options to switch between Nvidia and Intel Graphics

**Other Notes:**
Switching manually using `system76-power graphics` works fine. Additionally, `system76-power graphics switchable` reports that the graphics are indeed switchable. Output of `lspci | grep -i vga`:

```
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile)
01:00.0 VGA compatible controller: NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] (rev a1)
```

GNOME Shell logs are attached.
[gnome-shell_2018-03-18-17-23.log](https://github.com/pop-os/system76-power/files/2980902/gnome-shell_2018-03-18-17-23.log)

OgulcanCelik, 2019-04-08:
This issue also present on
`NAME="Pop!_OS"
VERSION="18.10"
`
system76-power graphics switchable reports that the graphics are indeed switchable. 

The issue is about the new gnome-shell and gnome related packages I believe. Before the update today, It was working fine.

brs17, 2019-04-10:
@OgulcanCelik , the issue in 18.10 should have been fixed with an update the following day.

OgulcanCelik, 2019-04-10:
yeah, it seems like it's fixed with the new update.

closed 2019-04-10 by jackpot51

isantop, 2019-04-15:
#71 is a duplicate of this bug. Since that report has all of the technical information, we should leave this one closed and proceed on that one.