pop-os/system76-power #71: 19.x release has long delay in unbinding the Nvidia GPU

Luke-Nukem, 2019-04-15:
**Distribution (run `cat /etc/os-release`):**
 
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
system76-power:
  Installed: 1.0.0~1553639437~19.04~ce5c6e4
  Candidate: 1.0.0~1553639437~19.04~ce5c6e4
  Version table:
 *** 1.0.0~1553639437~19.04~ce5c6e4 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu disco/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**

Using Ubuntu base 19.04, unbinding the Nvidia GPU now takes approx 2-3 minutes vs 18.10 being nearly instant.

```
-- Reboot --
Apr 15 15:58:32 pop-os systemd[1]: Started System76 Power Daemon.
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Starting daemon
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Connecting to dbus system bus
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Rescanning PCI bus
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:01:00.0: NVIDIA graphics
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:01:00.2: Function for 0000:01:00.0
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:01:00.3: Function for 0000:01:00.0
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:01:00.1: Function for 0000:01:00.0
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:00:02.0: Intel graphics
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] 0000:00:02.0: Function for 0000:00:02.0
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Setting automatic graphics power
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] Disabling graphics power
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] xhci_hcd: Unbinding 0000:01:00.2
Apr 15 15:58:32 pop-os system76-power[945]: [INFO] nvidia-gpu: Unbinding 0000:01:00.3
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] snd_hda_intel: Unbinding 0000:01:00.1
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] 0000:01:00.2: Removing
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] 0000:01:00.0: Removing
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] 0000:01:00.3: Removing
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] 0000:01:00.1: Removing
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] Initializing with the balanced profile
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] Registering dbus name com.system76.PowerDaemon
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] hwmon: coretemp
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] hwmon: acpitz
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] hwmon: iwlwifi
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] hwmon: pch_cannonlake
Apr 15 16:02:07 pop-os system76-power[945]: [ERROR] fan daemon: platform hwmon not found
Apr 15 16:02:07 pop-os system76-power[945]: [INFO] Handling dbus requests
```

**Expected behaviour:**

This should be the same behaviour as 18.x distro release where the unbinding happens almost instantly.

**Other Notes:**

I will be trying builds from master, and perhaps insert some logging to see if I can identify where the delay is actually happening.

Luke-Nukem, 2019-04-16:
I have tried to see if I can isolate a commit where the issue appears by checking out and building historical commits. This seems to be a dead-end. I'll now try debugging and see what I can do here.

I suspect the issue may be with Ubuntu 19.04. Can anyone confirm latest build from master works fine on 18.x?

mmstick, 2019-04-16:
It works fine in 18.x, yes. The latest build is actually what you get in Pop!_OS 18.04 and 18.10 stable. Our GitHub repo is monitored by a packaging CI which automatically packages the master branches of each repo with a debian directory to Pop!_OS Proposed, which soon gets swept up into stable and released into a new ISO. Code doesn't get to master if it hasn't already had a code review and been tested in the lab by QA.

Luke-Nukem, 2019-04-16:
Ah okay, good to know :smiley_cat: . Have there been internal tests for 19.04? I'm curious to know if others are having the same issue. It certainly seems related to 19.x now, but I'm unsure quite where to start in debugging.

mmstick, 2019-04-16:
Yes, see https://github.com/orgs/pop-os/projects/8

This is causing switchable graphics issues in 19.04.

Luke-Nukem, 2019-04-16:
Thanks for the link! I'll see if I can help out here. The delay also means that shutdown/standby/reboot etc are delayed until the unbind is complete, so there may be other things affected?

Luke-Nukem, 2019-04-16:
Issue might be in `sysfs-class` around the file_write. Something seems to be making this take a long time to complete, unsure if `sysfs-class` is the actual issue, or if something in 19.04 is causing the write to be delayed - but this is the area I'll be looking in for now.

mmstick, 2019-04-16:
This would be a kernel issue, but we could perhaps send all I/O tasks to a background thread to prevent the daemon from stalling.

Luke-Nukem, 2019-04-16:
Confirming it's a kernel issue. I tried:

- 4.18.0, no issue, unbinds quick
- 5.0.0, unbind takes 3 minutes
- 5.0.7, unbind takes 3 minutes

I haven't tried 5.1 RC kernels yet. Probably using a threaded model would be best in the long run, or even futures - but I can't help with this as I have limited time (and need to earn a living) sorry.

jackpot51, 2019-04-16:
I found that the i2c-nvidia-gpu module is the cause, it was recently added to Linux. I am blacklisting it by default on the oryp5, where this problem occurs.

closed 2019-04-16 by jackpot51

Luke-Nukem, 2019-04-17:
@jackpot51 excellent! Thank you for the information and fix!

yochananmarqos, 2020-12-19:
Maybe add the following patch instead of blacklisting it?

https://bugzilla.kernel.org/show_bug.cgi?id=206653