pop-os/system76-power #73: [REQUEST] Add option for AMD - NVIDIA switching for AMD APUs (Ryzen)

unsuitable001, 2019-04-16:
I don't know why no one is talking about this. But, this is a problem I'm currently stuck at.
I have an laptop with AMD APU (Ryzen 5 - 2500U) & nvidia gtx 1050. This laptop is Muxless. This laptop's display is attached to the iGPU (Vega 8).

I want to use Vega for normal tasks. & Nvidia for process hungry tasks (mainly CUDA & ML training stuffs).

There are many (glitchy) options for switching graphics (e.g Prime, bumblebee) . But, everyone states about Intel-nvidia. If you search more, you can find some about intel-amd switching.

But, nothing for AMD-nvidia switching.

APUs are things. Can anyone tell me how I can do this?

Note : I can't disable any graphics on bios. My laptop doesn't allow it.

Laptop : Asus F570D

What I have tried : I tried installing ubuntu multiple times. It (almost) disappointed me everytime. The last time I did, it was ALMOST working correctly (by setting flags like - `noveau.modeset=0` ). It just sometimes freeze during boot forever.

stdevel, 2019-08-01:
Hello,
agreed - I'd also like to have an option for AMD switching.

Best wishes,
Christina.

skybreakerrd, 2019-09-01:
Agreed, having the same problem, can't switch to amd graphics with my amd/nvidia laptop.

crawfxrd, 2019-11-18:
Output of `sudo gpu-manager` would be useful for this.

AFAICT, using `ubuntu-drivers-common` to enable offloading only works with Intel [\[1\]](https://git.launchpad.net/ubuntu/+source/ubuntu-drivers-common/tree/share/hybrid/gpu-manager.c?h=applied/ubuntu/eoan&id=e91ecc5af6f25d1f7cc38dd43eba3c001a991d6e#n1133).

Xeddius, 2019-12-05:
Also would love to have this as I've switched to an ASUS laptop with a Ryzen5 (Vega8) CPU, and Nvidia 1650 GPU and I've been stuck using the iGPU. I guess these computers are too new for the community to have created proper support and patches. But this is inevitable as gaming laptops become more widespread and Linux users continue to increase. I hope someone can come up with a solution soon because otherwise I'm stuck with Windows if I want to play even one slightly demanding game.

crawfxrd, 2019-12-27:
tseliot/ubuntu-drivers-common#40 if you would like to test it.

Xeddius, 2020-01-23:
@crawfxrd Thank you for the patches, is there any chance you could build a .deb or share a link to build it? (I'm currently on Windows because of issues caused by lack of AMD/Prime support in Linux.)

crawfxrd, 2020-01-27:
I've pushed a branch to our fork so Jenkins will build a package (19.10 only).

It can be installed from:
```
add-apt-repository "deb [arch=amd64] http://apt.pop-os.org/staging/amd-nvidia eoan main"
```

Xeddius, 2020-02-22:
Despite the fact it's set up for ubuntu 19.10 I can confirm that it will install successfully on Linux Mint 19.3 Tricia using the packages from the repo. After installing the packages, running gpu-manager, and rebooting the Nvidia configuration panel shows up and recognizes the dedicated Nvidia GPU! it's nice to finally have something that works!!!

Syzygianinfern0, 2020-04-24:
> I've [pushed a branch](https://github.com/pop-os/ubuntu-drivers-common/commits/amd-nvidia_eoan) to our fork of `ubuntu-drivers-common` so Jenkins will build a package (19.10 only).
> 
> It can be installed from:
> 
> ```
> add-apt-repository "deb [arch=amd64] http://apt.pop-os.org/staging/amd-nvidia eoan main"
> ```

How can I get this running on 20.04 and a 18.04 version of PopOs?

crawfxrd, 2020-04-24:
Manually add that line, replacing `eoan` with `focal`, or use the [`apt` script](https://github.com/pop-os/pop/blob/master/scripts/apt)

```
./scripts/apt add amd-nvidia
sudo apt install system76-power ubuntu-drivers-common
```

It won't be available on 18.04.

Syzygianinfern0, 2020-04-24:
> Manually add that line, replacing `eoan` with `focal`, or use the [`apt` script](https://github.com/pop-os/pop/blob/master/scripts/apt)
> 
> ```
> ./scripts/apt add amd-nvidia
> sudo apt install system76-power ubuntu-drivers-common
> ```
> 
> It won't be available on 18.04.

TYSM :heart_eyes: 
https://i.imgur.com/zB1tX6D.png
Although, tiny bug of `Intel` showing up rather than `AMD Integrated`

crawfxrd, 2020-04-24:
That's part of the [shell extension](https://github.com/pop-os/gnome-shell-extension-system76-power), which I haven't updated.

Syzygianinfern0, 2020-04-24:
> That's part of the [shell extension](https://github.com/pop-os/gnome-shell-extension-system76-power), which I haven't updated.

Hope that would be done soon in the best spirits. I am extremely grateful to the dev community of PopOs for creating such a wonderful experience. It's you people who make Linux Adoption bloom :smile:

Syzygianinfern0, 2020-04-26:
@crawfxrd The flags in [hybrid](https://github.com/pop-os/system76-power#hybrid) don't seem to be affecting anything. Are those expected? Currently to force applications to use the Nvidia GPU, I boot into the "Nvidia" profile and it works fine. How do I trigger it from the "Hybrid" profile??

crawfxrd, 2020-04-26:
First, `/var/log/gpu-manager.log` will tell you if it worked. It should have logged "NVIDIA PRIME detected" when in hybrid mode.

In 20.04 you can right click on the application icon and select "Launch using Dedicated Graphics Card".

But support depends on the application: what are you trying to run?

e.g., no GNOME applications that I've tried work, Firefox needs `webgl.force-enabled` set.

Syzygianinfern0, 2020-04-26:
> First, `/var/log/gpu-manager.log` will tell you if it worked. It should have logged "NVIDIA PRIME detected" when in hybrid mode.
<details>
<summary>My Log</summary>
 
```
log_file: /var/log/gpu-manager.log
last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/5.3.0-7648-generic/updates/dkms
Found nvidia module: nvidia.ko
Looking for amdgpu modules in /lib/modules/5.3.0-7648-generic/updates/dkms
Is nvidia loaded? yes
Was nvidia unloaded? no
Is nvidia blacklisted? no
Is intel loaded? no
Is radeon loaded? no
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? no
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? yes
Is nvidia kernel module available? yes
Is amdgpu kernel module available? no
Vendor/Device Id: 10de:1f91
BusID "PCI:1@0:0:0"
Is boot vga? no
Vendor/Device Id: 1002:15d8
BusID "PCI:5@0:0:0"
Is boot vga? yes
Skipping "/dev/dri/card1", driven by "nvidia-drm"
Found "/dev/dri/card0", driven by "amdgpu"
output 0:
	card0-eDP-1
Number of connected outputs for /dev/dri/card0: 1
Skipping "/dev/dri/card1", driven by "nvidia-drm"
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card1", driven by "nvidia-drm"
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card1", driven by "nvidia-drm"
Skipping "/dev/dri/card0", driven by "amdgpu"
Does it require offloading? yes
last cards number = 2
Has amd? yes
Has intel? no
Has nvidia? yes
How many cards? 2
Has the system changed? No
NVIDIA PRIME detected # <<<<<<<< It's there 🎉
Creating /usr/share/X11/xorg.conf.d/11-nvidia-offload.conf
Removing /usr/share/X11/xorg.conf.d/11-nvidia-prime.conf
Setting power control to "auto" in /sys/bus/pci/devices/0000:01:00.0/power/control
```
</details>

> But support depends on the application: what are you trying to run?

I'm trying to run Minecraft from a .jar file using `java -jar Minecraft.jar`

crawfxrd, 2020-04-27:
If that's the launcher it would probably have to set in each installation's executable path.

antienemy, 2020-05-01:
- I've Freshly Installed 20.04
- I've updated using Updates available in the bundled software.
- I've installed mesa-utils
- I ran 
`add-apt-repository "deb [arch=amd64] http://apt.pop-os.org/staging/amd-nvidia focal main"`
- This fixed The AMD APU issues,, I think.
- Still can't see anything related to NVIDIA. nvidia-smi recognizes the GPU but says "no process running"
- Yesterday, this somehow worked on 19.1 But after a clean install of 20.04, it seems I don't remember what to do exactly.
- Help Please

Syzygianinfern0, 2020-05-01:
> * Still can't see anything related to NVIDIA. nvidia-smi recognizes the GPU but says "no process running"

Try to run anything with the NVidia GPU then. Or you can boot into the NVidia only mode from the system-76 power options [like this](https://i.imgur.com/zB1tX6D.png). If you cant see that you must [follow this](https://github.com/pop-os/system76-power/issues/73#issuecomment-619240681) to install it.

antienemy, 2020-05-01:
@Syzygianinfern0 
- I Don't see any of those Graphics options.
- I honestly have a problem following it,
is running `add-apt-repository "deb [arch=amd64] http://apt.pop-os.org/staging/amd-nvidia focal main"` enough ? Cause I've done that much already and can't seem to get anything to work yet

Syzygianinfern0, 2020-05-01:
Do it this way.
``` 
$ git clone https://github.com/pop-os/pop.git
$ cd pop
$ ./scripts/apt add amd-nvidia
$ sudo apt install system76-power ubuntu-drivers-common
```
Then reboot. You will see the menu in under the power options status indicators in the top bar. 

You can select NVidia only and reboot after that to have all rendering done on NVidia if that is what you want. I just keep it on hybrid to save power while also have CUDA compute capabilities as I use it on a reguar basis. Same goes for gaming

antienemy, 2020-05-01:
**Thank you @Syzygianinfern0 this fixed everything.** 
- On first reboot, screen was flickering. It's due to being set to _NVIDIA Graphics - only_ Mode. Switched it to  Hybrid & Rebooted. All good. Awesome.
- Also, are these warnings something serious? If yes, how do I fix them?
- **could not find registry key file**
https://imgur.com/a/eNcRAEl
`fire@pop-os:~$ nvidia-settings
Gtk-Message: 13:24:01.344: Failed to load module "appmenu-gtk-module"

(nvidia-settings:3570): GLib-GObject-CRITICAL **: 13:24:01.461: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

ERROR: nvidia-settings could not find the registry key file. This file should
       have been installed along with this driver at
       /usr/share/nvidia/nvidia-application-profiles-key-documentation. The
       application profiles will continue to work, but values cannot be
       prepopulated or validated, and will not be listed in the help text.
       Please see the README for possible values and descriptions.

** (nvidia-settings:3570): WARNING **: 13:24:01.515: PRIME: Failed to execute child process “/usr/bin/prime-supported” (No such file or directory)
** Message: 13:24:01.515: PRIME: is it supported? no
`
- **displayed "Possible Missing firmware for amd" while following instructions**
https://imgur.com/a/ASlMgQh

Syzygianinfern0, 2020-05-01:
> * screen was flickering

You can currently fix that by suspending and logging in again. (Fn-F11 as I think we have the same machine :stuck_out_tongue: )
I'm not really sure what is wrong but you can contact me over [Telegram](https://t.me/Syzygianinfern0) as I have a few questions. If you aren't available on TG just leave any social handle you are comfy with.

antienemy, 2020-05-01:
> I'm not really sure what is wrong but you can contact me over [Telegram](https://t.me/Syzygianinfern0) as I have a few questions. If you aren't available on TG just leave any social handle you are comfy with.
I don't have a Telegram handle unfortunately, we can communicate `[here](https://tlk.io/xxxsxxx)`

swagglepuf, 2020-05-03:
> Do it this way.
>
> ```
> $ git clone https://github.com/pop-os/pop.git
> $ cd pop
> $ ./scripts/apt add amd-nvidia
> $ sudo apt install system76-power ubuntu-drivers-common
> ```
>
> Then reboot. You will see the [...]

I suspect that this is the only way to achieve this properly. I tried adding the repo and changing eoan to focal. Then the install command would say everything is up to date and installed. On reboot there was still no option for switching.

Thank you for these instructions and getting this to work on amd/nvidia.

Syzygianinfern0, 2020-05-05:
Do any of you have issues of the mouse being not slightly laggy (IDK what to call it.. maybe moving at a lower FPS) in the NVidia only mode?

crawfxrd, 2020-05-12:
Keeping this open until ubuntu-drivers-common work is done.

crawfxrd, 2020-05-14:
Our ubuntu-drivers-common package has been patched. All system76-power components have been updated to support AMD+NVIDIA.

closed 2020-05-14 by crawfxrd

Syzygianinfern0, 2020-05-22:
> Our ubuntu-drivers-common package has been patched. All system76-power components have been updated to support AMD+NVIDIA.

Does this mean that it will work even on 18.04 version of PopOS?

crawfxrd, 2020-05-22:
No, this feature is not backported to 18.04.

swagglepuf, 2020-05-29:
http://apt.pop-os.org/staging/amd-nvidia focal main
No longer has a release file. Is this still the location of updates for the amd+nvidia drivers, or is there another ppa that has been established?

brs17, 2020-05-29:
> http://apt.pop-os.org/staging/amd-nvidia focal main
> No longer has a release file. Is this still the location of updates for the amd+nvidia drivers, or is there another ppa that has been established?

That is because the code was released two weeks ago (https://github.com/pop-os/system76-power/pull/151) You do not need to use any staging branches for this feature anymore.

keshavvinayak01, 2020-09-15:
Hey, I tried to run 
```
sudo add-apt-repository "deb [arch=amd64] http://apt.pop-os.org/staging/amd-nvidia focal main"
```

But I get hit with this error while updateing : 
```
W: The repository 'http://apt.pop-os.org/staging/amd-nvidia focal Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://apt.pop-os.org/staging/amd-nvidia/dists/focal/main/binary-amd64/Packages  404  Not Found [IP: 52.38.47.2 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Help?

antienemy, 2020-09-15:
Check reddit reply

PR #151 (2020-05-12): Allow switching on AMD+NVIDIA systems
Allow switching on systems with AMD+NVIDIA graphics. Rename the "Intel" graphics mode to "Integrated" to reflect this change.

Until gpu-manager is updated, it will not actually have any effect on AMD+NVIDIA systems if run.

Requires: tseliot/ubuntu-drivers-common#40 (or pop-os/ubuntu-drivers-common#3)
Resolves: #73