pop-os/system76-power #74: Discrete graphics turn on after suspend

xnaveira, 2019-04-18:
**Distribution (run `cat /etc/os-release`):**
 
```
NAME="Pop!_OS"
VERSION="18.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.10"
VERSION_ID="18.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
system76-power:
  Installed: 1.0.0~1555432002~18.10~1c347b8
  Candidate: 1.0.0~1555432002~18.10~1c347b8
  Version table:
 *** 1.0.0~1555432002~18.10~1c347b8 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu cosmic/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**

I am on a Lenovo T480 with intel and nvidia (mx150), when running on intel, after returning from suspend, the nvidia card is silently powered on.

**Steps to reproduce (if you know):**

Boot the machine in intel mode, check the power with `system76-power graphics power`, suspend and then power on and check again.

**Expected behavior:**

If nvidia is powered off at any point in time, I would expect to be kept off even if I suspend the machine for a while.

I dropped the following script into `/lib/systemd/system-sleep` as a workaround:

```
#!/usr/bin/env bash
case $1 in
  post)
    if [ $(/usr/bin/system76-power graphics) == "intel" ]; then 
     /usr/bin/system76-power graphics power off;
    fi
    ;;
esac
```

**Other Notes:**

Also noticed while testing that issuing `system76-power graphics power off` while in nvidia mode leaves the `system76-power` command unresponsive and the machine hangs trying to suspend, that's why the script has an `if`, I would expect `system76-power` just to give an error saying it is not possible to power off the card being in use.

crawfxrd, 2019-10-30:
Is this still reproducible? If so, do you have logs for this?

anoulis, 2019-11-07:
For some reason in Pop os 18.04 doesn't work automatically after suspend. Only if i run it manually works.
Any clue?

MrSquaare, 2019-11-27:
Hello,

I would like to complete this issue.

I do have the same problem.

Distribution (run `cat /etc/os-release`):
```
NAME="Pop!_OS"
VERSION="19.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.10"
VERSION_ID="19.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=eoan
UBUNTU_CODENAME=eoan
LOGO=distributor-logo-pop-os
```

Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):
```
system76-power:
  Installed: 1.0.1~1573832218~19.10~1a6e703
  Candidate: 1.0.1~1573832218~19.10~1a6e703
  Version table:
 *** 1.0.1~1573832218~19.10~1a6e703 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu eoan/main amd64 Packages
        100 /var/lib/dpkg/status
```

crawfxrd, 2019-11-27:
To confirm:
- `system76-power graphics power` reports `off` before suspend?
- `system76-power graphics power` reports `on` after suspend?

What computer model do you have? And is this reproducible every suspend?

MrSquaare, 2019-11-27:
> To confirm:
> 
>     * `system76-power graphics power` reports `off` before suspend?
> 
>     * `system76-power graphics power` reports `on` after suspend?
> 
> 
> What computer model do you have? And is this reproducible every suspend?

This is indeed what happens, status `off` before suspension, status `on` after suspension and it's reproducible at each suspension

My computer model is a Lenovo Y540, the graphics card is a NVidia RTX 2060

crawfxrd, 2019-11-27:
I would assume some script is rescanning the PCI bus on resume. I know we have one script ([`system76-thunderbolt-reload`](https://github.com/pop-os/system76-driver/blob/d3d1aeff1bef52013c02cc196e2ed46cc7d657ee/system76-thunderbolt-reload)) that rescans the PCI bus on resume, but this is limited to our darp6 and galp4 models which don't have dGPUs.

```
grep -rw rescan /lib/systemd/system-sleep
```

MrSquaare, 2019-11-28:
> I would assume some script is rescanning the PCI bus on resume. I know we have one script [...]

I don't know if I was supposed to execute the command or not, but I didn't get an output

SuryashankarDas, 2020-05-21:
I can confirm this issue on pop os 20.04.
On system start it remains power off.
On resume from sleep it shows power on.

dwilliams-1, 2021-06-15:
It looks like this is still an issue in pop os 20.10. Are there any plans to fix this?

chunga2020, 2022-09-13:
Arch Linux user chiming in here that this happens on my gaze14.  I keep my laptop in `integrated` mode and keep my dGPU off, because I take it to class, where it runs on battery.  When I shut down before, the dGPU is off and the graphics mode is `integrated`.  When I power back up, it's still in `integrated` mode, but for some reason, the dGPU is on.

OS: Arch
Kernel: 5.19.7
`system76-power` version: 1.1.23, installed from AUR

closed 2025-09-17 by crawfxrd