pop-os/system76-power #76: system76-power incompatible with nvidia-driver-418

fmarier, 2019-04-29:
**Distribution (run `cat /etc/os-release`):**
 
 ```
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

On an `oryp4` laptop running nvidia-driver-418.

**Issue/Bug Description:**

I cannot switch to Intel graphics using `system76-power graphics intel`.

The command itself doesn't print out any errors, but here's what's in the logs:

```
Apr 29 16:01:18  system76-power[1423]: [INFO] DBUS Received SetGraphics(intel) method
Apr 29 16:01:18  system76-power[1423]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Apr 29 16:01:18  system76-power[1423]: [INFO] Disabling nvidia-fallback.service
Apr 29 16:01:18  system76-power[1423]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
Apr 29 16:01:18  system76-power[1423]: [ERROR] systemctl: failed with exit code: 1
Apr 29 16:01:18  system76-power[1423]: [INFO] Updating initramfs
Apr 29 16:01:18  system76-power[1423]: update-initramfs: Generating /boot/initrd.img-4.18.0-18-generic
Apr 29 16:01:34  system76-power[1423]: kernelstub.Config    : INFO     Looking for configuration...
Apr 29 16:01:34  system76-power[1423]: kernelstub           : INFO     System information:
Apr 29 16:01:34  system76-power[1423]:     OS:..................Pop!_OS 18.04
Apr 29 16:01:34  system76-power[1423]:     Root partition:....../dev/dm-1
Apr 29 16:01:34  system76-power[1423]:     Root FS UUID:........21898a4f-6af8-4792-86aa-e3ba4bdc6302
Apr 29 16:01:34  system76-power[1423]:     ESP Path:............/boot/efi
Apr 29 16:01:34  system76-power[1423]:     ESP Partition:......./dev/nvme0n1p1
Apr 29 16:01:34  system76-power[1423]:     ESP Partition #:.....1
Apr 29 16:01:34  system76-power[1423]:     NVRAM entry #:.......-1
Apr 29 16:01:34  system76-power[1423]:     Boot Variable #:.....0000
Apr 29 16:01:34  system76-power[1423]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
Apr 29 16:01:34  system76-power[1423]:     Kernel Image Path:.../vmlinuz
Apr 29 16:01:34  system76-power[1423]:     Initrd Image Path:.../initrd.img
Apr 29 16:01:34  system76-power[1423]:     Force-overwrite:.....False
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Copying Kernel into ESP
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Backing up old kernel
Apr 29 16:01:34  system76-power[1423]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
```

Looking at this part:

```
Apr 29 16:01:18  system76-power[1423]: [INFO] Disabling nvidia-fallback.service
Apr 29 16:01:18  system76-power[1423]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
Apr 29 16:01:18  system76-power[1423]: [ERROR] systemctl: failed with exit code: 1
```

I found that the file existed in the `nvidia-kernel-common-390` package:

```
$ apt-file search nvidia-fallback.service
nvidia-kernel-common-390: /lib/systemd/system/nvidia-fallback.service
```

but not `nvidia-kernel-common-418`:

```
$ dpkg -L nvidia-kernel-common-418
/.
/lib
/lib/modprobe.d
/lib/modprobe.d/nvidia-graphics-drivers.conf
/lib/udev
/lib/udev/rules.d
/lib/udev/rules.d/71-nvidia.rules
/sbin
/sbin/create-uvm-dev-node
/usr
/usr/share
/usr/share/doc
/usr/share/doc/nvidia-kernel-common-418
/usr/share/doc/nvidia-kernel-common-418/changelog.Debian.gz
/usr/share/doc/nvidia-kernel-common-418/copyright
/usr/share/initramfs-tools
/usr/share/initramfs-tools/hooks
/usr/share/initramfs-tools/hooks/framebuffer-nvidia
```

mmstick, 2019-04-29:
It is compatible with 418, as that's what all of our ISOs and hardware ships with today. We test switchable graphics before releasing new ISOs or NVIDIA drivers.

```
Unit file nvidia-fallback.service does not exist
```

This is normal and expected behavior.

fmarier, 2019-04-30:
> This is normal and expected behavior.

While this message isn't the cause of the problem, graphics switching still is broken with 418 on my oryp4 on Pop!_OS. It was working fine a few weeks ago.

thermochromic, 2019-05-09:
Can confirm this. I also have a oryp4 with Ubuntu that system76-power no longer switches to Intel graphics. 
`
~$ uname -a
Linux oryxpro 4.18.0-15-generic #16pop0~18.04.1-Ubuntu SMP Thu Feb 21 16:15:23 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
`
`
~$ nvidia-smi
Thu May  9 14:11:23 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.56       Driver Version: 418.56       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1060    Off  | 00000000:01:00.0  On |                  N/A |
| N/A   41C    P3    18W /  N/A |    329MiB /  6078MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1276      G   /usr/lib/xorg/Xorg                           226MiB |
|    0      1948      G   ...quest-channel-token=8353807858628215354   100MiB |
+-----------------------------------------------------------------------------+
`

abdo1819, 2019-05-10:
i confirm this too 
### Distribution (run `cat /etc/os-release`):
```
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```
###  `NVIDIA-SMI`
```
Fri May 10 14:25:45 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.56       Driver Version: 418.56       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 105...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   56C    P0    N/A /  N/A |    327MiB /  4040MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1906      G   /usr/lib/xorg/Xorg                           144MiB |
|    0      2016      G   /usr/bin/gnome-shell                          94MiB |
|    0      2601      G   ...quest-channel-token=4110863545584547843    86MiB |
+-----------------------------------------------------------------------------+
```
the log after running `system76-power graphics intel`
```
-- Reboot --
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Starting daemon
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Connecting to dbus system bus
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Rescanning PCI bus
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] 0000:01:00.0: NVIDIA graphics
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] 0000:00:02.0: Intel graphics
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] 0000:00:02.0: Function for 0000:00:02.0
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Setting automatic graphics power
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Enabling graphics power
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Initializing with the balanced profile
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Registering dbus name com.system76.PowerDaemon
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] hwmon: coretemp
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] hwmon: acpitz
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] hwmon: iwlwifi
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] hwmon: pch_cannonlake
May 10 13:56:55 abdo-pc system76-power[1078]: [ERROR] fan daemon: platform hwmon not found
May 10 13:56:55 abdo-pc system76-power[1078]: [INFO] Handling dbus requests
May 10 13:57:09 abdo-pc system76-power[1078]: [INFO] DBUS Received GetSwitchable method
May 10 13:57:09 abdo-pc system76-power[1078]: [INFO] DBUS Received GetGraphics method
May 10 13:57:09 abdo-pc system76-power[1078]: [INFO] DBUS Received GetProfile method
May 10 14:05:28 abdo-pc system76-power[1078]: [INFO] DBUS Received SetGraphics(intel) method
May 10 14:05:28 abdo-pc system76-power[1078]: [INFO] Creating /etc/modprobe.d/system76-power.conf
May 10 14:05:28 abdo-pc system76-power[1078]: [INFO] Disabling nvidia-fallback.service
May 10 14:05:28 abdo-pc system76-power[1078]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
May 10 14:05:28 abdo-pc system76-power[1078]: [ERROR] systemctl: failed with exit code: 1
May 10 14:05:28 abdo-pc system76-power[1078]: [INFO] Updating initramfs
May 10 14:05:28 abdo-pc system76-power[1078]: update-initramfs: Generating /boot/initrd.img-4.18.0-18-generic
[... 30 lines cut ...]
May 10 14:20:07 abdo-pc system76-power[1078]:     OS:..................Ubuntu 18.04
May 10 14:20:07 abdo-pc system76-power[1078]:     Root partition:....../dev/sdb3
May 10 14:20:07 abdo-pc system76-power[1078]:     Root FS UUID:........865861be-a6ff-4c94-9626-f0e2b804c25a
May 10 14:20:07 abdo-pc system76-power[1078]:     ESP Path:............/boot/efi
May 10 14:20:07 abdo-pc system76-power[1078]:     ESP Partition:......./dev/sdb2
May 10 14:20:07 abdo-pc system76-power[1078]:     ESP Partition #:.....2
May 10 14:20:07 abdo-pc system76-power[1078]:     NVRAM entry #:.......-1
May 10 14:20:07 abdo-pc system76-power[1078]:     Boot Variable #:.....0000
May 10 14:20:07 abdo-pc system76-power[1078]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
May 10 14:20:07 abdo-pc system76-power[1078]:     Kernel Image Path:.../vmlinuz
[... cut ...]
May 10 14:20:07 abdo-pc system76-power[1078]: kernelstub.Installer : INFO     Backing up old kernel
May 10 14:20:07 abdo-pc system76-power[1078]: kernelstub.Installer : INFO     Making entry file for Ubuntu
``` 
has same error 
```
May 10 14:19:55 abdo-pc system76-power[1078]: [INFO] Disabling nvidia-fallback.service
May 10 14:19:55 abdo-pc system76-power[1078]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
May 10 14:19:55 abdo-pc system76-power[1078]: [ERROR] systemctl: failed with exit code: 1
```

pperle, 2019-05-11:
I am having the same error, but I am stuck on the intel gpu unable to switch to nvidia, and I am running 19.04.

## Distribution (run cat /etc/os-release):
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
```

The log after running `system76-power graphics nvidia` has same error as mensurn before.

```
[INFO] DBUS Received SetGraphics(nvidia) method
[INFO] Creating /etc/modprobe.d/system76-power.conf
[INFO] Enabling nvidia-fallback.service
Failed to enable unit: Unit file nvidia-fallback.service does not exist.
[ERROR] systemctl: failed with exit code: 1
[INFO] Updating initramfs
update-initramfs: Generating /boot/initrd.img-5.0.0-13-generic
cryptsetup: WARNING: The initramfs image may not contain cryptsetup binaries
    nor crypto modules. If that's on purpose, you may want to uninstall the
    'cryptsetup-initramfs' package in order to disable the cryptsetup initramfs
    integration and avoid this warning.
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information:
    OS:..................Pop!_OS 19.04
    Root partition:....../dev/nvme0n1p6
    Root FS UUID:........f1afb827-6b6d-463f-a627-588e3074d955
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p5
    ESP Partition £:.....5
    NVRAM entry £:.......-1
    Boot Variable £:.....0000
    Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
    Kernel Image Path:.../vmlinuz
    Initrd Image Path:.../initrd.img
    Force-overwrite:.....False
kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     Making entry file for Pop!_OS
```

zpcv, 2019-05-11:
I'm having the same problem as  @pperle, cant switch from the Intel GPU to the Nvidia one.

abdo1819, 2019-05-13:
it seem the problem is not of  `nvidia-fallback.service` as it is service for switching from nvidia to Nouveau driver [found it here](https://github.com/negativo17/nvidia-driver/issues/73)
```
● nvidia-fallback.service - Fallback to nouveau if nvidia did not load
   Loaded: loaded (/usr/lib/systemd/system/nvidia-fallback.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2019-02-10 10:57:02 CET; 8h ago
  Process: 829 ExecStart=/sbin/modprobe nouveau (code=exited, status=0/SUCCESS)
 Main PID: 829 (code=exited, status=0/SUCCESS)
```
and in code suggest to ignore this erro here [src/graphics.rs](https://github.com/pop-os/system76-power/blob/1c347b83f5fc186862f79fde70c5096b7873f124/src/graphics.rs#L202)
```rust
info!("Disabling nvidia-fallback.service");
                let status = process::Command::new("systemctl").arg("disable").arg("nvidia-fallback.service").status()?;
                if ! status.success() {
                    // Error is ignored in case this service is removed
                    error!("systemctl: failed with {}", status);
                }
```

but still i couldn't know what is the problem

mmstick, 2019-05-13:
I'm currently working on reworking the error handling in the daemon so that the error messages will be very descriptive about the nature of each error, rather than giving a generic I/O error code message.

fmarier, 2019-05-23:
I was able to switch to Intel and then back to NVIDIA on Pop 18.04 using:

- `system76-power` version `1.0.0~1557760114~18.04~99e80af`
- `nvidia-driver-418` version `418.56-0ubuntu1pop1~1558036981`

michaeltintiuc, 2019-07-15:
Had the same problem with a clean install, switched to Intel successfully, yet couldn't switch back to Nvidia.

michaeltintiuc, 2019-07-15:
I just tried a quick OS refresh through the live usb and it appears I can switch GPUs fine, although brightness controls are now broken :man_facepalming:

brs17, 2020-04-30:
440.82 is now the latest version of the NVIDIA driver in Pop!_OS. Please open a new issue if similar issues exist on the latest driver.

closed 2020-04-30 by brs17

kevr, 2020-07-16:
Problem exists on my system with nvidia after upgrading to Pop OS 20.04 from 19.10.

kevr, 2020-07-16:
Problem resolved after i reinstalled nvidia: `apt-get install --reinstall nvidia-driver-440 nvidia-dkms-440` for my system.