pop-os/system76-power #82: Brightness at startup is fixed and has to be changed manually

IntronV, 2019-05-16:
**Hardware:**

Lenovo Thinkpad T430

**Distribution (run `cat /etc/os-release`):**
 
 Pop!_OS 18.10
 

**Issue/Bug Description:**

Screen brightness after booting is fixed to a certain level (about 50%), no matter at what value it was set in the last session. You have to adjust the brightness manually everytime. Usually the last value should be set again.

**Other Notes:**

Others also have the problem, see: https://pop-planet.info/forums/threads/brightness-at-startup.80/

ahnafalrafi, 2019-05-16:
Are there any plans to implement this in a future update/release?

mmstick, 2019-05-16:
Yes, this has been part of the configuration file feature.

IntronV, 2019-05-17:
Great, I'm looking forward to it!

PR #94 (2019-07-29): Persistable brightness
Fixes #82.
closed 2019-07-29 by mmstick