pop-os/system76-power #89: Possible system76-power Hard Drive power management problem

AAnzel, 2019-06-03:
**Distribution (run `cat /etc/os-release`):**

NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power:
  Installed: 1.0.0 1558479403 19.04 33fe432
  Candidate: 1.0.0 1558479403 19.04 33fe432
  Version table:
 *** 1.0.0 1558479403 19.04 33fe432 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu disco/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**

Non-OS Hard Drive keeps falling in sleep mode and waking up in intervals of a few seconds. After going into Gnome Disks aplication, then drive settings and changing APM to 128 disk behaves as before (goes to sleep after normal time of inactivity).

**Steps to reproduce (if you know):**

N/A

**Expected behavior:**

Hard Drive goes to sleep afte few minutes of inactivity.

**Other Notes:**

Hard Drive model name:
ST1000LM024 HN-M101MBB (2BA30002), 1.0TB

mmstick, 2019-06-04:
The default HDD APM in the balanced profile is currently `127`, with an autosuspend delay of `-1` (auto). I guess that means we should define an autosuspend delay that is not automatically-configured by the kernel. The HDD should only wake up when there are I/O requests for it, though.

AAnzel, 2019-06-04:
I guess that APM value is not a problem, but as You noted it might be autosuspend delay. I guess that it should be set on certain values for each power profile accordingly, just in order to avoid possible differences between kernel-configured delays on different machines.
Also, it seems that changing power profile overwrites APM values set in Gnome Disk and HDD goes back to same, bad behaviour.

mmstick, 2019-06-04:
Yes, power profiles are currently hard-coded to set the parameters that they have defined. So if you change power profiles in the daemon, then APM and autosuspend parameters are also set according to that profile.

If you select the battery profile, there's a defined value of 15000 (15 seconds). For balanced and performance it is `-1` (auto).

I am also currently working on adding support for configuration files, which would also enable custom profiles.

AAnzel, 2019-06-04:
Ok, thanks! I was just confused how HDD started to misbehave all of a sudden a few days ago. I guess I will wait for configuration functionality to be added and use Gnome Disk utility until then to set the HDD behaviour to my liking.
Also, after 19.04 upgrade there is loud poping noise coming from speakers but only on machine restart. There is no poping sound during powering on/off. Is this problem PopOS or Ubuntu related?

I would also like to thank You and Your team for this great OS. My laptop is finally breathing with full lungs and using its maximum. Keep doing a great job!

MartinPesek, 2019-06-04:
Hello,

I see this bug too.

**Distribution (run cat /etc/os-release):**
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
```

**Related Application and/or Package Version (run apt policy $PACKAGE NAME):**
```
system76-power:
  Installed: 1.0.0~1559655543~19.04~1be5c67
  Candidate: 1.0.0~1559655543~19.04~1be5c67
  Version table:
 *** 1.0.0~1559655543~19.04~1be5c67 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu disco/main amd64 Packages
        100 /var/lib/dpkg/status
```

However my issue is a bit more severe, my HDD is spinning down in mere seconds. It spins down even before I see the login screen only to be spinned up and then spinned down right away before Gnome is even fully loaded.

If I set APM value to 254 using `gnome-disks` the HDD will stop spinning down. However this is not preserved between reboots, or more specifically, in `gnome-disks` it looks like everything is set correctly but it just doesn't work. I have to completely change the settings and then change it back to make it stick until next reboot. I presume that's because of what's described in [previous comment](https://github.com/pop-os/system76-power/issues/89#issuecomment-498823216).

I have one another internal HDD and one external HDD and these are not affected by this bug. System disk is a NVME disk.

Before this bug, all my HDDs were not spinning down at all, which is exactly what I wanted and how it worked out of the box for me.

mmstick, 2019-06-04:
I'll push an update out that sets a reasonable value for the autosuspend delay.

AAnzel, 2019-06-05:
I checked both for High Performance and Balanced profile and it seems that it works great on High Performance but on Balanced it still behaves bad. That is especially noticeable when trying to do small but periodic reads (or writes) from HDD (like listening to music). It spins up for a read, then spins down and after a few seconds cycle repeats. I guess that it just needs to be set on higher value.

mmstick, 2019-06-05:
If you're willing to, you could try out the configuration file branch. It lets you manually alter these values in a TOML config file.

```sh
git clone https://github.com/pop-os/pop
./pop/scripts/apt add s76powerconf
sudo apt upgrade
```

Then you can create a file at `/etc/system76-power/config.toml` with the following contents:

```toml
[profiles.balanced]
disk.apm_level = 127
disk.autosuspend_delay = 60
```

You can restart the daemon after modifying the file to have its parameters take affect.

```sh
sudo system76-power restart system76-power
```

Note that the system-maintained reference config is located at `/usr/lib/system76-power/config.toml`, but note that this file is not meant to be edited.

AAnzel, 2019-06-05:
Ok, now I'm confused. It appears that now it works even for Balanced profile. I completely rebooted the system after installing the new version of system-76-power and it didn't work. Then I accidentally logged out and then back in and magic happened. Now it works for both High Performance and Balanced profile and still behaves the same for Battery Life profile, but that should be like that so I guess it's fixed.
I don't know why reboot didn't restart the daemon but it doesn't matter now.
Thanks for a quick response! Greetings from Serbia!

mmstick, 2019-06-05:
You may want to try a few reboots, and see if it continues to work each time.

AAnzel, 2019-06-05:
I did several reboots and noticed a few things. After reboot problem is not resolved, Maximum Performance yields good HDD behaviour, Balanced profile doesn't. After logging out and then back in HDD works as it should. But after changing profile to Maximum Performance (or Battery Life) and then back to Balanced, it returns to bad behaviour. I don't know what could possibly be the cause of this.

mmstick, 2019-06-05:
So first login = bad
Second login = good
Profile change = bad

The power daemon only sets parameters on system startup and during a profile change, so something is overriding the disk behavior on that second login. I'd recommend trying out the configuration file branch and tweaking parameters in the config to see if they make a difference.

AAnzel, 2019-06-05:
Exactly.
I'll be short on time for a few days. As soon as I find some spare time I'll try it out and report my results.

AAnzel, 2019-06-06:
I have a few questions considering **profiles.rs**. I noticed that You are setting laptop mode in every profile but passed arguments for Battery Life and Balanced are the same **b"2"**. Is this desired or maybe an error? I'm not very familiar with what those arguments mean but I suspect that it's short for _battery lvl. 2_ or similar. Nevertheless, I inspected **laptop mode** source code [link](https://www.kernel.org/doc/Documentation/laptops/laptop-mode.txt) and noticed that there is also **hdparm** calls which could overwrite calls from **profiles.rs**, so that might cause undefined behaviour on logins.

mmstick, 2019-06-11:
@AAnzel The laptop mode does configure a number of parameters internally with some overlap, although we do call hdparm ourselves after calling laptop mode, so it overrides those calls in particular.