pop-os/system76-power #90: Documentation request

drNoob13, 2019-06-04:
**Hardware: **
Few System76 Oryx pro 4 laptops with GTX 1070 Max-Q that run on Ubuntu 16.04 LTS.

**Distribution (run `cat /etc/os-release`):**
```bash
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.6 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.6 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
``` 

 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
Not compatible on Ubuntu 16.04.

**Issue/Bug Description:**
There is no information for System76 laptop owners to build the drivers.

**Steps to reproduce (if you know):**
There is no documentation for System76 customers to build the driver packages for their System76 laptops. System76 doesn't provide ppa for driver installation in Ubuntu 16.04.

**Expected behavior:**
Could you provide build instruction/documentation so that System76 laptop owner can build the driver by themselves? Thank you.

**Other Notes:**

mmstick, 2019-06-04:
Every repository that builds packages contains a `debian` directory. The `debian/control` file defines all build and runtime dependency requirements. Debian packages can be built with `dpkg-buildpackage -b`, `debuild`, `sbuild`, or any other debian package builder of your choice.

As long as you have a recent-enough version of Rust (defined in `rust-toolchain`), Rust repos will build with a mere `cargo build --release`, though you'll want to use the included Makefiles (`make && sudo make install prefix=/usr`) if you are packaging our software for another distribution, or installing without creating a debian package. If required to vendor dependencies for packaging elsewhere, you can install cargo vendor with `cargo install cargo-vendor` and then proceed to use the included `make vendor` rule to generate the `vendor.tar.xz` file. Example in `debian/rules`.

closed 2019-06-04 by mmstick

drNoob13, 2019-06-13:
Thanks for the reply.

If my understanding is correct, this should go together with the system76-driver, shouldn't it? Could you have look at this issue as well:

https://github.com/pop-os/system76-driver/issues/110