pop-os/system76-power #103: Failed to switch to Nvidia graphics

sefeng211, 2019-07-19:
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
```

I was playing on my Nvidia graphic, and suddenly everything became unresponsive, so I did a force-reboot. After the reboot, I am unable to switch to my Nvidia. 

This is the recent reboot logs 
```
-- Reboot --
Mar 29 23:59:56 pop-os system76-power[1074]: [INFO] Starting daemon
Mar 29 23:59:56 pop-os system76-power[1074]: [INFO] Connecting to dbus system bus
Mar 29 23:59:56 pop-os system76-power[1074]: [INFO] Rescanning PCI bus
Mar 29 23:59:56 pop-os system76-power[1074]: [INFO] 0000:01:00.0: NVIDIA graphics
...skipping...
Jul 19 19:52:55 saber-home system76-power[1189]:     nor crypto modules. If that's on purpose, you may want to uninstall the
Jul 19 19:52:55 saber-home system76-power[1189]:     'cryptsetup-initramfs' package in order to disable the cryptsetup initramfs
Jul 19 19:52:55 saber-home system76-power[1189]:     integration and avoid this warning.
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Config    : INFO     Looking for configuration...
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub           : INFO     System information:
Jul 19 19:53:04 saber-home system76-power[1189]:     OS:..................Pop!_OS 19.04
Jul 19 19:53:04 saber-home system76-power[1189]:     Root partition:....../dev/nvme0n1p4
Jul 19 19:53:04 saber-home system76-power[1189]:     Root FS UUID:........6867d90f-d33e-4e19-be8e-1c661ade81d9
Jul 19 19:53:04 saber-home system76-power[1189]:     ESP Path:............/boot/efi
Jul 19 19:53:04 saber-home system76-power[1189]:     ESP Partition:......./dev/nvme0n1p2
Jul 19 19:53:04 saber-home system76-power[1189]:     ESP Partition #:.....2
Jul 19 19:53:04 saber-home system76-power[1189]:     NVRAM entry #:.......-1
Jul 19 19:53:04 saber-home system76-power[1189]:     Boot Variable #:.....0000
Jul 19 19:53:04 saber-home system76-power[1189]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
Jul 19 19:53:04 saber-home system76-power[1189]:     Kernel Image Path:.../vmlinuz
Jul 19 19:53:04 saber-home system76-power[1189]:     Initrd Image Path:.../initrd.img
Jul 19 19:53:04 saber-home system76-power[1189]:     Force-overwrite:.....False
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Copying Kernel into ESP
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Backing up old kernel
Jul 19 19:53:04 saber-home system76-power[1189]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Jul 19 19:53:08 saber-home system76-power[1189]: [INFO] caught signal: 15
Jul 19 19:53:09 saber-home system76-power[1189]: [INFO] daemon exited from loop
-- Reboot --
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Starting daemon
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Connecting to dbus system bus
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Rescanning PCI bus
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] 0000:01:00.0: NVIDIA graphics
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] 0000:01:00.1: Function for 0000:01:00.0
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] 0000:00:02.0: Intel graphics
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] 0000:00:02.0: Function for 0000:00:02.0
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Setting automatic graphics power
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] Disabling graphics power
Jul 19 19:53:34 saber-home system76-power[1154]: [INFO] snd_hda_intel: Unbinding 0000:01:00.1
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] 0000:01:00.0: Removing
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] 0000:01:00.1: Removing
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] Initializing with the balanced profile
Jul 19 19:53:35 saber-home system76-power[1154]: [WARN] Failed to set initial profile: Errors found when setting profile:
Jul 19 19:53:35 saber-home system76-power[1154]:     - failed to set scsi host profiles: failed to set link time power management policy med_power_with_dipm on host0: No such file or director
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] Registering dbus name com.system76.PowerDaemon
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Jul 19 19:53:35 saber-home system76-power[1154]: [ERROR] fan daemon: platform hwmon not found
Jul 19 19:53:35 saber-home system76-power[1154]: [INFO] Handling dbus requests
Jul 19 19:54:03 saber-home system76-power[1154]: [INFO] DBUS Received GetSwitchable method
Jul 19 19:54:03 saber-home system76-power[1154]: [INFO] DBUS Received GetGraphics method
Jul 19 19:54:03 saber-home system76-power[1154]: [INFO] DBUS Received GetProfile method

```

sefeng211, 2019-07-20:
I somehow can't set to performance profile
```
sefeng@saber-home:~$ system76-power profile performance
setting power profile to performance
daemon returned an error message: "Errors found when setting profile:\n    - failed to set scsi host profiles: failed to set link time power management policy med_power_with_dipm on host0: No such file or directory (os error 2)"

```
seems relevant?

mmstick, 2019-07-20:
Errors are permitted when setting a profile. If a certain parameter fails to set, it records the error and continues on, then prints all errors that occurred while setting the profile.

As for failing to set NVIDIA graphics, the daemon should be receiving and acting on a `SetGraphics` method call. I'm not seeing that method getting called at all. You could try setting NVIDIA graphics with `system76-power` directly.

sefeng211, 2019-07-20:
Thanks for the reply! Use `system76-power` to switch directly didn't work, it didn't print out anything and I was still at Intel after a reboot. 

```
sefeng@saber-home:~$ system76-power graphics nvidia
setting graphics to nvidia
```
I think I also had some other issues now. Yesterday I was trying to reinstall the nvidia driver because I read something about 430 doesn't work well with Linux kernel 5.0.0-21. So I uninstalled the nvidia-driver-430 and tried to reinstall. However, I couldn't do so because I kept getting error messages like this 
```
ERROR (dkms apport): kernel package linux-headers-5.0.0-21-generic is not supported
Error! Bad return status for module build on kernel: 5.0.0-21-generic (x86_64)
```

Full Error Message 
```
sefeng@saber-home:~$ sudo apt-get upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up nvidia-dkms-430 (430.34-1pop1~1563200531~19.04~2a70632) ...
update-initramfs: deferring update (trigger activated)
INFO:Enable nvidia
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp2-ess
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3-ess
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/dell_latitude
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-serw10
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3-b
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp2
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/put_your_quirks_here
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/lenovo_thinkpad
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-bonw11
Removing old nvidia-430.34 DKMS files...

------------------------------
Deleting module version: 430.34
completely from the DKMS tree.
------------------------------
Done.
Loading new nvidia-430.34 DKMS files...
Building for 5.0.0-21-generic
Building for architecture x86_64
Building initial module for 5.0.0-21-generic
ERROR (dkms apport): kernel package linux-headers-5.0.0-21-generic is not supported
Error! Bad return status for module build on kernel: 5.0.0-21-generic (x86_64)
Consult /var/lib/dkms/nvidia/430.34/build/make.log for more information.
dpkg: error processing package nvidia-dkms-430 (--configure):
 installed nvidia-dkms-430 package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of nvidia-driver-430:
 nvidia-driver-430 depends on nvidia-dkms-430 (= 430.34-1pop1~1563200531~19.04~2a70632); however:
  Package nvidia-dkms-430 is not configured yet.

dpkg: error processing package nvidia-driver-430 (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                          No apport report written because the error message indicates its a followup error from a previous failure.
                      dpkg: dependency problems prevent configuration of nvidia-driver-418:
 nvidia-driver-418 depends on nvidia-driver-430; however:
  Package nvidia-driver-430 is not configured yet.

dpkg: error processing package nvidia-driver-418 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of system76-driver-nvidia:
 system76-driver-nvidia depends on nvidia-driver-418; however:
  Package nvidia-driver-418 is not configured yet.

dpkg: error processing package system76-driver-nvidia (--configure):
 dependency problems - leaving unconfigured
No apport report written because MaxReports is reached already
                                                              Processing triggers for initramfs-tools (0.131ubuntu19.1pop1) ...
update-initramfs: Generating /boot/initrd.img-5.0.0-21-generic
cryptsetup: WARNING: The initramfs image may not contain cryptsetup binaries 
    nor crypto modules. If that's on purpose, you may want to uninstall the 
    'cryptsetup-initramfs' package in order to disable the cryptsetup initramfs 
    integration and avoid this warning.
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information: 

    OS:..................Pop!_OS 19.04
    Root partition:....../dev/nvme0n1p4
    Root FS UUID:........6867d90f-d33e-4e19-be8e-1c661ade81d9
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p2
    ESP Partition #:.....2
    NVRAM entry #:.......-1
    Boot Variable #:.....0000
    Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
    Kernel Image Path:.../vmlinuz
    Initrd Image Path:.../initrd.img
    Force-overwrite:.....False

kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     Making entry file for Pop!_OS
Errors were encountered while processing:
 nvidia-dkms-430
 nvidia-driver-430
 nvidia-driver-418
 system76-driver-nvidia
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

mmstick, 2019-07-20:
The 430 drivers are the current drivers in Pop right now, so they should be working fine. I would recommend purging all NVIDIA packages from the system, and then reinstalling `system76-driver-nvidia`.

Removing nvidia drivers cleanly can be tricky. Sometimes `sudo apt purge *nvidia*` does not work on its own, and you need to `sudo dpkg -P package1 package2 package3` all of the NVIDIA packages that give dependency errors before you can do it.

sefeng211, 2019-07-20:
hmm, still seeing the same error when installing the driver.

statX, 2019-08-05:
Not sure if this is the exact problem here, but I had something similar go wrong with "nvidia-dkms-430 not configured". Does $(which gcc) return version 8.x? If not set gcc to link to gcc-8 and try again.

crawfxrd, 2019-11-22:
Is this still reproducible?

sefeng211, 2019-11-22:
Sorry, I ended up just reinstall the OS, closing it.

closed 2019-11-22 by sefeng211

williantenfen, 2019-12-12:
> The 430 drivers are the current drivers in Pop right now, so they should be working fine. I would recommend purging all NVIDIA packages from the system, and then reinstalling [...]

Thanks! This saved my life! :slightly_smiling_face: