pop-os/system76-power #107: Authentication error!

LinArcX, 2019-08-07:
**Distribution (run `cat /etc/os-release`):**
 Now i'm in void-linux.

**Issue/Bug Description:**
I tried to remove some applications(Because i don't need some of them at all and i wanted to replace some others with better ones in IMHO). For example evince, gnome-user-doc, seahorse, nautilus.
When i removed nautilus for example, the font of terminal/system messed up. After reboot, i can't see the normal login screen of popos. it replaced by default gnome login screen and when i want to write my password, a hint appear in the below of inputbox says: "Authentication error". and after 2 second, the screen become black and i had to restart it. Again and again i see this issue, until i go back to void-linux.
Guys, why we can't delete default apps?

**Steps to reproduce (if you know):**

**Expected behavior:**

**Other Notes:**

mmstick, 2019-08-07:
This is expected when you begin to remove desktop components which are critical to the functioning of the GNOME desktop. Certain GNOME desktop components are interconnected to other GNOME desktop components. These dependencies are not optional in the `pop-desktop` packaging.

closed 2019-08-07 by mmstick

LinArcX, 2019-08-07:
Got it. Thanks for clarification. But I think there should be a strong way that prevent users to do such these things. (or at least warn them about consequences) I don't know... Maybe using apparmor?

mmstick, 2019-08-07:
There should be, but `apt` generally assumes that you know what you're doing. The only time it gives a warning is when you try to remove the init system.