pop-os/system76-power #108: Turn off animations on battery mode

7FFY, 2019-08-24:
The idea is to turn off gnome animations on battery mode as:

Battery Life --> animations OFF
Balanced --> animations ON
High Performance --> animations ON

Main reason is that animations on battery mode can be slow probably caused by the cpu clock reduction, resulting in a bad UX + I think it make sense to disable animations as it's gonna consumes less resources (more battery life).

cassidyjames, 2019-08-25:
As a note, Android used to do this. However, it degraded the user experience enough that they reverted it in later versions and now focus on other areas of less user-perceptible ways of reducing power draw, like more strict background app restrictions.

7FFY, 2019-08-26:
Got it, but mobile and desktop devices still a different beast.
And as said it's not about the battery power (mine is nice), it's more about the cpu clock reduction that might (on some devices) be too much, resulting choppy animations :/

ayoungethan, 2020-05-28:
Android, to my knowledge, still has a "low power" mode that you can configure to turn off animations. 
I appreciate having it when I need to squeeze a few extra hours out of a long day on the phone.

It should be optional. Or configurable to automatically kick in past a threshold point. As it is in Android. 

But, as someone who LOVES long battery life (I have a Darter Pro that can give me 10-14 hrs of work time), I actually recommend getting rid of the "battery life" mode altogether. It doesn't actually save battery life, and in my (and I suspect, most) use cases it actually causes slightly worse battery life. It violates the "race to sleep" philosophy of integrated throughput performance and power saving. The only times I can think of where you really need a different performance parameter than "Balanced" are three scenarios:
1. For low latency applications (where you want a consistent frequency)
2. For low latency throughput (where you want a consistent HIGH frequency)
3. For situations where something in the computer has gone haywire and hijacked the CPU and you need to limit the max frequency to buy yourself some time before maxed-out CPU runs down your battery. That, IMO, is not a good reason to have something in the GUI, and if there was something there, it should be labeled "panic button" and ideally also help the user pause a runaway process so that they can evaluate how to proceed with a more level head (e.g., save work, kill the process via system monitor and/or restart the computer).

So, really, there are two main modes that mostly matter (ha!):
1. Balanced, for most things that most people do and
2. A "Performance" mode that lets you set a stable CPU frequency of your choice. This is really important for low latency audio work, for example, to minimize jitter and maximize throughput if you are doing a lot of DSP. A slider would let you adjust the target frequency. I say "target" because no guarantees with thermals, especially in a laptop. But it would let the user who needs low latency to choose an appropriate frequency sufficient for their DSP workload and find some balance with battery life if necessary.

This, in my opinion (2 settings with a target frequency slider in performance mode), would be much more simple and effective for a broader range of use cases!

"Balanced" mode should be optimized for the race to sleep philosophy, and that means keeping as many cores in as deep a sleep state as possible for as long as possible during low-intensity use, while keeping one or two more responsive so that the user experience doesn't lag. And perhaps alternating between cores so that they all get a chance to sleep and cool off (not sure how much thermal burden they share vs take on independently). I find that UI lag occurs (e.g., with EPP set to 'balance energy' or 'priority energy' -- I forget the exact terms that HWP uses) when the computer is keen to put cores to sleep but reluctant to awaken ANY of them. It seems to me like it could be further optimized to do the workload with the currently-awakened core(s) while also awakening the other cores to take over seamlessly so that the initial cores can sleep. There is some overhead to this as a sort of context switching, but it's in "balanced" mode, not "performance" mode so I think the little bit of additional overhead may be justified if it results in a cooler system, longer battery life and still no noticeable UI lagginess.