pop-os/system76-power #113: Switch between nvidia/intel shows an error, but actually works

dannyk81, 2019-10-09:
**Distribution (run `cat /etc/os-release`):**
 
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
LOGO=distributor-logo-pop-os
``` 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
N/A
```

**Issue/Bug Description:**

Switching nvidia -> intel and intel -> nvidia with the UI shows the below error message, however when restarting the laptop it does switch and works just fine.

[image]

Logs:
```
$ journalctl /usr/bin/system76-power -f
Oct 08 20:48:02 x1-extreme system76-power[1336]: [INFO] DBUS Received SetGraphics(intel) method
Oct 08 20:48:02 x1-extreme system76-power[1336]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Oct 08 20:48:02 x1-extreme system76-power[1336]: [INFO] Disabling nvidia-fallback.service
Oct 08 20:48:02 x1-extreme system76-power[1336]: Removed /etc/systemd/system/graphical.target.wants/nvidia-fallback.service.
Oct 08 20:48:03 x1-extreme system76-power[1336]: [INFO] Updating initramfs
Oct 08 20:48:03 x1-extreme system76-power[1336]: update-initramfs: Generating /boot/initrd.img-5.0.0-31-generic
Oct 08 20:48:12 x1-extreme system76-power[1336]: cryptsetup: WARNING: The initramfs image may not contain cryptsetup binaries
Oct 08 20:48:12 x1-extreme system76-power[1336]:     nor crypto modules. If that's on purpose, you may want to uninstall the
Oct 08 20:48:12 x1-extreme system76-power[1336]:     'cryptsetup-initramfs' package in order to disable the cryptsetup initramfs
Oct 08 20:48:12 x1-extreme system76-power[1336]:     integration and avoid this warning.
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Config    : INFO     Looking for configuration...
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub           : INFO     System information:
Oct 08 20:48:31 x1-extreme system76-power[1336]:     OS:..................Pop!_OS 19.04
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Root partition:....../dev/nvme0n1p5
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Root FS UUID:........3c5a4310-5f0f-4c18-a33b-3c78ca0ca5cc
Oct 08 20:48:31 x1-extreme system76-power[1336]:     ESP Path:............/boot/efi
Oct 08 20:48:31 x1-extreme system76-power[1336]:     ESP Partition:......./dev/nvme0n1p1
Oct 08 20:48:31 x1-extreme system76-power[1336]:     ESP Partition #:.....1
Oct 08 20:48:31 x1-extreme system76-power[1336]:     NVRAM entry #:.......-1
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Boot Variable #:.....0000
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Kernel Image Path:.../vmlinuz
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Initrd Image Path:.../initrd.img
Oct 08 20:48:31 x1-extreme system76-power[1336]:     Force-overwrite:.....False
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Copying Kernel into ESP
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Backing up old kernel
Oct 08 20:48:31 x1-extreme system76-power[1336]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
```

switching using cli command (`system76-power graphics nvidia|intel`) also works fine.

**Steps to reproduce (if you know):**

described above

**Expected behavior:**

successful message displayed with an option to restart/restart later

**Other Notes:**

Runing on Lenovo X1 Extreme Gen1

brs17, 2019-10-09:
I have seen this as well, although I have not seen it be reproducible again and again. @dannyk81 are you able to reproduce the issue consistently?

dannyk81, 2019-10-09:
@brs17 indeed, this happens every single time I switch between GPUs.

I had very few cases where the error did not appear, however I could never figure out what was different in these cases (the logs looked identical), but the "happy path" happened maybe 5 times out of a 100 :sweat_smile:

Since it's easy to reproduce this on my setup I'd be happy to provide any additional data/info :+1:

dannyk81, 2019-10-23:
@brs17 I upgraded to Pop!_OS 19.10 several days ago and the problem disappeared...

brs17, 2019-10-23:
> @brs17 I upgraded to Pop!_OS 19.10 several days ago and the problem disappeared...

In 19.10, we extended the timeout which I think will solve most issues that were being faced (related to rebuilding the initramfs).

closed 2019-10-23 by brs17