pop-os/system76-power #114: "High Performance" mode keeps scaling_governor on powersave

pperle, 2019-10-09:
**Distribution (run `cat /etc/os-release`):**
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
LOGO=distributor-logo-pop-os
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

**Issue/Bug Description:**
`scaling_governor` is kept on `powersave` when switching to "High Performance" mode.  

**Steps to reproduce (if you know):**
1. run `cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor | uniq` when in "Balanced" mode, this should return `powersave`
2. switch to "High Performance" via UI or by running `system76-power profile performance`
3. run `cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor | uniq` which returns `powersave`

**Expected behavior:**
`scaling_governor` switches to `performance` when switching to "High Performance" mode and back to `powersave` when leaving this mode. 

**Other Notes:**
This behavior has been documented by https://kofler.info/pop_os/ (german) and in this recent comment https://github.com/erpalma/throttled/issues/144#issuecomment-540091113, which caught my attention. This is the same when I am on battery or plugged in.

mmstick, 2019-10-09:
This is expected. We aren't using predefined profiles, because we are modifying the pstate values directly, so the string in that file doesn't matter.

closed 2019-10-09 by mmstick

pperle, 2019-10-09:
@mmstick thank you for the quick response can you please explain to me why the "current CPU frequency" (from `cpupower frequency-info`) stays constant when switching between "Balanced" and "High Performance" but changes when I run `sudo cpupower frequency-set -g performance` manually?

mmstick, 2019-10-09:
If you run `system76-power profile`, you can see what the frequency range is for a given profile. The main difference between CPU frequencies for our balanced and performance profile is that performance raises the minimum frequency to 50%.

pperle, 2019-10-09:
Maybe I misunderstand the "High Performance" mode. In https://github.com/pop-os/system76-power/wiki is says that High Performance mode "should turn the processor tuning to the high performance mode" which I understand as 'give me the best performance'.

On "Balanced" I get a single-core score of [585](https://browser.geekbench.com/v5/cpu/352071). With "High Performance" mode activated, I get a single-core score of [668](https://browser.geekbench.com/v5/cpu/351960) once I set `scaling_governor` to `performance` I can achieve a single-core score of [1266](https://browser.geekbench.com/v5/cpu/351985).