pop-os/system76-power #115: Thinkpad X1E Gen1 overheating in suspend

ArtificialAmateur, 2019-10-15:
**Distribution (run `cat /etc/os-release`):**
```
NAME="Pop!_OS"
VERSION="19.04"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.04"
VERSION_ID="19.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
LOGO=distributor-logo-pop-os
```

 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
system76-power:
  Installed: 1.0.1~1570998296~19.04~d0013c0
  Candidate: 1.0.1~1570998296~19.04~d0013c0
  Version table:
 *** 1.0.1~1570998296~19.04~d0013c0 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu disco/main amd64 Packages
        100 /var/lib/dpkg/status

```

**Issue/Bug Description:**
Suspending the laptop doesn't seem to be actually suspending the device.

**Steps to reproduce (if you know):**
Click suspend, close laptop, or run `systemctl suspend`.

**Expected behavior:**
Resource utilization drops drastically, producing less heat and consuming less battery life.

**Other Notes:**
I've run `stress` before and tried suspending it and it still is hot to the touch after 5 or so minutes.

**More info:**

dmesg errors:

```
[  341.289344] mce: CPU8: Core temperature above threshold, cpu clock throttled (total events = 787)
[  341.289345] mce: CPU2: Core temperature above threshold, cpu clock throttled (total events = 787)
[  341.289375] mce: CPU2: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289378] mce: CPU8: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289381] mce: CPU1: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289382] mce: CPU7: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289383] mce: CPU3: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289384] mce: CPU0: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289385] mce: CPU9: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289386] mce: CPU6: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289387] mce: CPU5: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289388] mce: CPU4: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289389] mce: CPU11: Package temperature above threshold, cpu clock throttled (total events = 2090)
[  341.289390] mce: CPU10: Package temperature above threshold, cpu clock throttled (total events = 2090)
```

Thermald status:
```

Oct 06 12:27:15 geri thermald[1798]: 22 CPUID levels; family:model:stepping 0x6:9e:a (6:158:10)
Oct 06 12:27:15 geri thermald[1798]: Polling mode is enabled: 4
Oct 06 12:27:15 geri thermald[1798]: sensor id 23 : No temp sysfs for reading raw temp
Oct 06 12:27:15 geri thermald[1798]: I/O warning : failed to load external entity "/etc/thermald/thermal-conf.xml"
Oct 06 12:27:15 geri thermald[1798]: error: could not parse file /etc/thermald/thermal-conf.xml
Oct 06 12:27:15 geri thermald[1798]: sysfs open failed
```

System76-power boot status:
```
Oct 15 13:32:14 geri system76-power[1526]: [WARN] Failed to set automatic graphics power: does not have switchable graphics
Oct 15 13:32:14 geri system76-power[1526]: [INFO] Initializing with the balanced profile
Oct 15 13:32:14 geri system76-power[1526]: [WARN] Failed to set initial profile: Errors found when setting profile:
Oct 15 13:32:14 geri system76-power[1526]:     - failed to set scsi host profiles: failed to set link time power management policy med_power_with_dipm on hos
Oct 15 13:32:14 geri system76-power[1526]: [INFO] Registering dbus name com.system76.PowerDaemon
Oct 15 13:32:14 geri system76-power[1526]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Oct 15 13:32:14 geri system76-power[1526]: [ERROR] fan daemon: platform hwmon not found
Oct 15 13:32:14 geri system76-power[1526]: [INFO] Handling dbus requests
Oct 15 13:32:26 geri system76-power[1526]: [INFO] DBUS Received GetSwitchable method
Oct 15 13:32:26 geri system76-power[1526]: [INFO] DBUS Received GetProfile method
```

crawfxrd, 2019-10-30:
If you use `systemctl suspend`, does it appear to stay suspended (display off, blinking power LED if it has one), or does it come back on (display turns on, solid power LED)?

mmstick, 2019-10-30:
Possibly related to https://github.com/pop-os/default-settings/pull/63

ArtificialAmateur, 2019-11-02:
Problem no longer persistent in latest release.

closed 2019-11-02 by ArtificialAmateur