pop-os/system76-power #117: Add indicator for Prime Offloading

odysseywestra, 2019-10-24:
Not sure if this is the correct place to ask, but now that we have Hybrid mode in for Nvidia graphics, it would be nice to have an indicator to let us know when the GPU is in use.

btw, I'm using Hybrid mode now and it work great so far. Keep up the awesome work!

mmstick, 2019-10-24:
Watching the output of `nvidia-smi pmon -d $DELAY` would make that possible.

crawfxrd, 2019-10-24:
If you just want to know if the card is on or off, and not _what_ is using it, it can be obtained from sysfs:

```
$ cat /sys/bus/pci/devices/<DEVICE>/power/runtime_status
active
```

If you don't have a Turing card then it's always "in use" (it will never go into a low power state).
If you have a Turing card, then you need some udev rules to allow it to go into a low power state [\[1\]](https://download.nvidia.com/XFree86/Linux-x86_64/435.21/README/dynamicpowermanagement.html).

From what I've seen, `nvidia-smi` (without options) will cause the card to wake up. I don't know if the `pmon` subcommand has the same effect.

closed 2025-09-17 by crawfxrd