pop-os/system76-power #122: Turing gpu power management

haydenhughes, 2019-10-30:
I'm not sure if this is an issue that is isolated to my laptop as it boots using the Nvidia GPU but as far as I'm aware, Pop_OS! doesn't implement the [RTD3 power management](http://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/dynamicpowermanagement.html) to allow for Turing chips to be powered down whilst not in use by PRIME offloading. It would be great if this could be enabled when the graphics mode is set to hybrid.

haydenhughes, 2019-10-30:
Following Nvidia's automated setup appears to freeze the system76-power daemon when it reaches `[INFO] Rescanning PCI bus`.

crawfxrd, 2019-10-30:
Does `echo 1 > /sys/bus/pci/rescan` have the same effect, and are there any relevant kernel log messages generated with either?

haydenhughes, 2019-10-30:
`echo 1 > /sys/bus/pci/rescan` also just hangs. Through some trial and error, it seems like the removal of the NVIDIA USB Type-C UCSI device is causing the issue.

crawfxrd, 2019-12-17:
@hugglesfox do you still experience the freeze with the latest 440.44 drivers?

hugglesfox, 2019-12-17:
Unfortunately yes

flukejones, 2020-04-11:
I think the section that does the scanning should be moved to a thread so that main part of the daemon can still receive commands and not prevent the PC from shutting down or sleeping.

flukejones, 2020-04-12:
Fix [here](https://github.com/pop-os/system76-power/pull/123#issuecomment-612676836)

Issue is characterized by PCI scan taking a while, and GPU not switching off until after rescan. Then there is a crash in `i2c_nvidia_gpu`:

 ```
BUG: unable to handle page fault for address: ffffb15300649000
#PF: supervisor read access in kernel mode
#PF: error_code(0x0000) - not-present page
PGD 85a555067 P4D 85a555067 PUD 85a558067 PMD 857941067 PTE 0
Oops: 0000 [#1] SMP PTI
CPU: 5 PID: 182 Comm: kworker/5:1 Tainted: P           OE     5.4.0-7618-generic #22~1584134448~20.04~82a44c0-Ubuntu
Hardware name: ASUSTeK COMPUTER INC. Zephyrus S GX502GW_GX502GW/GX502GW, BIOS GX502GW.308 02/15/2020
Workqueue: events ccg_pm_workaround_work [ucsi_ccg]
RIP: 0010:gpu_i2c_check_status.isra.0+0x53/0xa0 [i2c_nvidia_gpu]
Code: 25 00 00 00 60 3d 00 00 00 60 75 26 be 58 02 00 00 bf f4 01 00 00 e8 ac 8d 6a fd 48 8b 05 d5 cc 1d fe 48 39 d8 79 0b 49 8b 06 <44> 8b 20 45 85 e4 78 cb 48 8b 05 be cc 1d fe 48 39 c3 0f 88 c1 02
RSP: 0018:ffffb1530037bc78 EFLAGS: 00010293
RAX: ffffb15300649000 RBX: 0000000100003bdd RCX: 0000000000000000
RDX: ffff8d965db5dfc0 RSI: 0000000000000005 RDI: ffffb1530037bbc0
RBP: ffffb1530037bc98 R08: 0000000000000000 R09: 0000000000000009
R10: 0000000000000015 R11: ffff8d965db699e0 R12: 00000000e0000000
R13: ffff8d9654b49018 R14: ffff8d9654b49020 R15: ffff8d9654b49018
FS:  0000000000000000(0000) GS:ffff8d965db40000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: ffffb15300649000 CR3: 000000084d3e0006 CR4: 00000000003606e0
Call Trace:
 gpu_i2c_master_xfer+0xf5/0x23b [i2c_nvidia_gpu]
 __i2c_transfer.part.0+0xae/0x290
 __i2c_transfer+0xfc/0x2c0
 i2c_transfer+0x8b/0x110
 ccg_read+0x124/0x170 [ucsi_ccg]
 ? __switch_to_asm+0x40/0x70
 ucsi_ccg_sync+0x56/0xb0 [ucsi_ccg]
 ucsi_notify+0x26/0x120 [typec_ucsi]
 ccg_pm_workaround_work+0x15/0x20 [ucsi_ccg]
 process_one_work+0x1eb/0x3b0
 worker_thread+0x4d/0x400
 kthread+0x104/0x140
 ? process_one_work+0x3b0/0x3b0
 ? kthread_park+0x90/0x90
 ret_from_fork+0x35/0x40
Modules linked in: rfcomm ccm cmac algif_hash algif_skcipher af_alg bnep mei_hdcp snd_sof_pci intel_rapl_msr snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda snd_sof_intel_byt snd_hda_codec_hdmi snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_hda_ext_core snd_hda_codec_realtek snd_soc_acpi_intel_match snd_hda_codec_generic snd_soc_acpi ledtrig_audio snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_nhlt snd_hda_codec nvidia_drm(POE) snd_hda_core x86_pkg_temp_thermal intel_powerclamp snd_hwdep asus_nb_wmi coretemp snd_seq_midi snd_seq_midi_event kvm_intel nvidia_modeset(POE) snd_rawmidi asus_wmi kvm iwlmvm serio_raw intel_cstate sparse_keymap snd_pcm wmi_bmof btusb intel_rapl_perf snd_seq mac80211 btrtl btbcm btintel bluetooth libarc4 snd_seq_device nvidia(POE) ecdh_generic joydev nls_iso8859_1 input_leds snd_timer ecc iwlwifi ucsi_ccg mei_me snd mei hid_multitouch cfg80211 ipmi_devintf soundcore ipmi_msghandler processor_thermal_device
 ucsi_acpi intel_rapl_common typec_ucsi intel_pch_thermal intel_soc_dts_iosf typec int3403_thermal int340x_thermal_zone mac_hid acpi_pad asus_wireless int3400_thermal acpi_thermal_rel acpi_tad sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) system76_acpi(OE) hid_logitech_hidpp hid_logitech_dj usbhid hid_generic crct10dif_pclmul crc32_pclmul ghash_clmulni_intel mxm_wmi aesni_intel i915 crypto_simd cryptd glue_helper i2c_algo_bit drm_kms_helper i2c_i801 nvme r8169 syscopyarea sysfillrect realtek sysimgblt fb_sys_fops intel_lpss_pci intel_lpss nvme_core drm idma64 virt_dma i2c_nvidia_gpu i2c_hid hid wmi video pinctrl_cannonlake pinctrl_intel
CR2: ffffb15300649000
---[ end trace 340c6b8517ccad9e ]---
RIP: 0010:gpu_i2c_check_status.isra.0+0x53/0xa0 [i2c_nvidia_gpu]
Code: 25 00 00 00 60 3d 00 00 00 60 75 26 be 58 02 00 00 bf f4 01 00 00 e8 ac 8d 6a fd 48 8b 05 d5 cc 1d fe 48 39 d8 79 0b 49 8b 06 <44> 8b 20 45 85 e4 78 cb 48 8b 05 be cc 1d fe 48 39 c3 0f 88 c1 02
RSP: 0018:ffffb1530037bc78 EFLAGS: 00010293
RAX: ffffb15300649000 RBX: 0000000100003bdd RCX: 0000000000000000
RDX: ffff8d965db5dfc0 RSI: 0000000000000005 RDI: ffffb1530037bbc0
RBP: ffffb1530037bc98 R08: 0000000000000000 R09: 0000000000000009
R10: 0000000000000015 R11: ffff8d965db699e0 R12: 00000000e0000000
R13: ffff8d9654b49018 R14: ffff8d9654b49020 R15: ffff8d9654b49018
FS:  0000000000000000(0000) GS:ffff8d965db40000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: ffffb15300649000 CR3: 000000084d3e0006 CR4: 00000000003606e0
Worker [514] terminated by signal 9 (KILL)
0000:01:00.3: Worker [514] failed
```

I'll send this off to Nvidia too.

crawfxrd, 2020-04-23:
The udev workarounds are no longer required on the kernels used Pop!_OS or Ubuntu 20.04.

closed 2020-05-20 by crawfxrd