pop-os/system76-power #126: Add graphics preference in desktop file for specific supported applications

brs17, 2019-11-15:
Patch Steam and other Pop maintained and supported applications with a graphics preference description in the desktop file.

cassidyjames, 2019-11-15:
I believe GNOME has something like this upstream for (I assume a different method of) launching apps on certain GPUs; it might be worth checking how they do it and/or implementing this in a GNOME Shell extension, as modifying the .desktop file seems fragile.

crawfxrd, 2019-11-15:
There is currently no persistent preference for running an application on the dGPU.

This work covers the following point of [GNOME's dual GPU support](https://wiki.gnome.org/Design/OS/DualGPU):

> Allow applications telling the OS that they would prefer using the discreet GPU

The upstream tracking tickets for this is [GNOME/gnome-shell#1804](https://gitlab.gnome.org/GNOME/gnome-shell/issues/1804), including an [XDG spec](https://gitlab.freedesktop.org/xdg/xdg-specs/merge_requests/13) for the feature.

cassidyjames, 2019-11-15:
@crawfxrd awesome! That makes much more sense. Carry on. :)

crawfxrd, 2020-05-05:
`PrefersNonDefaultGPU` has been added to xdg-spec.