pop-os/system76-power #128: Integrate system76-power settings into gnome-control-center

brs17, 2019-11-19:
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power
gnome-shell-extension-system76-power

**Issue/Bug Description:**
The settings currently only available via gnome-shell-extension-system76-power should also be available in gnome-control-center. This is the first step in supporting configurable power profiles.

mmstick, 2019-11-19:
Functioning prototype:

[image]

ayoungethan, 2020-05-28:
I have some further optimization suggestions for the performance power profiles, to both simplify and streamline them and also help them cover a wider range of usage scenarios, including low latency priority and low latency throughput priority.

Should I put those suggestions here or in a separate bug report?

ayoungethan, 2020-07-02:
https://pop-planet.info/forums/threads/power-management-design-interface.739

also, see https://github.com/martin31821/cpupower/issues/14 as they are interested in expanding the functionality of their already-excellent extension. There is some overlap that might translate to collaboration.

yochananmarqos, 2020-12-14:
This is already being worked on upstream, see: https://www.phoronix.com/scan.php?page=news_item&px=GNOME-power-profiles-daemon

ayoungethan, 2020-12-14:
> This is already being worked on upstream, see: https://www.phoronix.com/scan.php?page=news_item&px=GNOME-power-profiles-daemon

thanks for sharing. it looks like they are replicating the same mistaken belief that the 3rd power profile, "power saving" actually saves power. it does not, in most cases, because it violates race to sleep, which is why "balanced" is appropriate for the vast majority of use cases. the remaining use cases will benefit from having a tuned "performance" setting. "power saving" really shouldn't exist except for a few niche cases, and should be renamed "low heat" or "low noise" because ultimately that is what it accomplishes.

the idea that throttling a cpu saves power or battery life really needs to die. it just keeps a cpu in a less efficient active state for longer, which is only good for niche cases that need to manage the thermal envelope or noise under load at the expense of overall efficiency. that is why Mac OS has better battery life - it embraces race to sleep design principles.

yochananmarqos, 2020-12-15:
@ayoungethan I've been researching this a little today and I agree. I created a udev rule to set the Balanced profile on battery and Performance on AC for now.

ayoungethan, 2020-12-15:
@yochananmarqos it seems like retreading over the original debate between the "ondemand" and "powersave" ACPI software performance governors before that functionality was moved over to hardware. Limiting max CPU frequency in the age of frequency scaling is really niche (and I am speaking from the perspective of someone who has use for that niche -- when doing low latency audio recording, I don't need a lot of CPU performance, but I need stable low-latency, low-jitter performance and ideally minimal fan noise and thus heat generation, which means locking the CPU at a lower-ish stable frequency on top of performance tuning to give the audio subsystem high priority, eliminate the overhead of symmetric multi-threading, etc).

Perhaps what is most missing, I think, is an integrated power management API that requires that an (or at least allows developers to easily ensure that their) application conforms to a "race to sleep" standard set by the OS developers. To my very limited knowledge, Mac OS really pioneered this. I haven't been keeping up to date on things, though, so maybe someone is working on this already...!

EJFJCorp, 2021-02-03:
Very good, but state of defectuous battery is not taken in account : if the battery is at less than 1% when the programm starts, it should not do any test and consider there is no battery for now... Because it crashes the computer on system start... In command line, with Ctrl+Alt+F5 no problem ,the computer works fine, it worked correctly 2 weeks ago, but now it crashes just after showing the background, and sometime one notification... I tried all interactions, with mouse, and touch screen, to try interesting the computer to what I am doing, so that it can, as it did sometimes 2 or 3 weeks ago, continue working after the little alimentation break during defectuous battery test! Thank you for correcting this, this is just one if instruction, but it will change a lot to users who have not yet replaced their battery... But a notice should be given to hellp them get the best battery now, as it should automatically be done by Mail Direct International PSC and all EJFJ humano-normed Reforms... And the MT-System Card would do the battery management, without letting the OS caring of it! The OS need to really care of the alive beings needs, by identifying, with hypomagnetic microchips, what they carry... As the MT System Card, extension of the main board, should, as the main board, and any other electronical material, identify (reading the hypo-magnetic micro-chips indicating the type of material, the producer, material habilities, to determine which is the best to accomplish which job, in order to satisfy all alive beings... And technical constraints need to be accepted but solved prevently, without letting any accident happen (explosion or fire, or other accident, coming always from few well know problems). Only the best is legal, so thank you, your solution should replace all others, but should be easier to use, more effective, but lighter for the system...

EJFJCorp, 2021-02-03:
> https://pop-planet.info/forums/threads/power-management-design-interface.739
> 
> also, see [martin31821/cpupower#14](https://github.com/martin31821/cpupower/issues/14) as they are interested in expanding the functionality of their already-excellent extension. There is some overlap that might translate to collaboration.

Yes, collaboration need to be automatised, without need of human beings to understand, nor even know each other... SDC++ should do this, but blocked by Microsoft Corporation, even in Linux world...

EJFJCorp, 2021-02-03:
> Functioning prototype:
> 
> [image]

This is to be used by the user, or specificities need to be positively contextual and changes between these 3 or more profiles need to be fully automatized depending on the real activities of the user... If the user starts an activity, or get a phone call, things need to be adapted consequently! EJFJ Contextual Computing depends on your Public Services Companies (still parodied and blocked by the SFM:id2004 and their numerous partners), mostly MediSecure International PSC and Bill !nternational Technologies PSC for detection and understandings of habilities of objects used by human beings, who are not servitors of such objects, but who should only benefit, without any risk nor constraint, of these objects and all others existing around... Human beings are artists, never perfect, but always surprising... Computers should be, with EJFJ Positive Contextualisation, effective and static, progressive but never partial nor surprising! The design and the illustrations human beings did with them can be, this is far enough!

EJFJCorp, 2021-02-03:
> **Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
> system76-power
> gnome-shell-extension-system76-power
>
> **Issue/Bug Description:**
> The settings currently only available [...]

Of course, according to EJFJ Utility Principle and EJFJ Unicitiy Principle, this should be globally done... Even all other Operating Systems should benefit of any good thing done on Linux, with correct retribution of all those who invented, proposed, improved, realized, adapted, corrected, or applied... All alive beings should only benefit, no risk should exist, and no constraint! Which is totally not the case for now, because of SFM:id2004, and their most important/exploited partners in non EJFJ Contextual Computing : Microsoft Corporation, IBM, Intel, Oracle, ...

hadess, 2021-08-06:
>  it looks like they are replicating the same mistaken belief that the 3rd power profile, "power saving" actually saves power

FWIW, I replied to this in:
https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/issues/21#note_851091
In short, "race-to-idle" solves a different problem from "I want my laptop to still have battery in an hour".

You can find here my attempt at breaking down what system76-power does and seeing what can be done in the vanilla kernel, or would still require tweaking:
https://gitlab.freedesktop.org/hadess/power-profiles-daemon/-/issues/26#note_1021525

If you want to keep your own profiles daemon (it's written in Rust, that's already better than what I'm using), maybe having it implement the power-profiles-daemon D-Bus API would be the best course of action, though mimicking its behaviour (like remembering the profile across reboots) might be a bit harder.

Ideally though, all the tweaks you have would be available in the vanilla kernel, or exported in your machines' firmwares through to the kernel through the `platform_profile` API.

Don't hesitate to reach out if you have any questions or want to discuss this in a higher bandwith setting.