pop-os/system76-power #129: Switched nvidia to intel, but now I cannot switch from nvidia to intel

miguelcarcamov, 2019-11-22:
NAME="Pop!_OS"
VERSION="18.04 LTS"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.04 LTS"
VERSION_ID="18.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

 
system76-cuda:
  Installed: 0pop2
  Candidate: 0pop2
  Version table:
 *** 0pop2 500
        500 http://apt.pop-os.org/proprietary bionic/main amd64 Packages
        100 /var/lib/dpkg/status
miguel@pop-os:~$ apt policy system76-power
system76-power:
  Installed: 1.0.1~1574207757~18.04~4f6a818
  Candidate: 1.0.1~1574207757~18.04~4f6a818
  Version table:
 *** 1.0.1~1574207757~18.04~4f6a818 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

Switched to intel graphics, but know I can't switch back to nvidia. I have done the process of switching via terminal and via GUI and after rebooting there is no change at all.

The steps to reproduce the error is basically having a hybrid GPU nvidia/intel. Installing pop os. Then, switch to intel. Try to switch back to nvidia, and the error will happen.

The expected behavior is that you could switch to nvidia successfully, however the OS remains using just the intel GPU.
Other notes: Logs from system76-power

`-- Reboot --
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Starting daemon
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Connecting to dbus system bus
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Rescanning PCI bus
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] 0000:01:00.0: NVIDIA graphics
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] 0000:00:02.0: Intel graphics
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] 0000:00:02.0: Function for 0000:00:02.0
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Setting automatic graphics power
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Disabling graphics power
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] 0000:01:00.0: Removing
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Initializing with the balanced profile
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Registering dbus name com.system76.PowerDaemon
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Nov 22 12:38:22 pop-os system76-power[951]: [ERROR] fan daemon: platform hwmon not found
Nov 22 12:38:22 pop-os system76-power[951]: [INFO] Handling dbus requests
Nov 22 12:39:01 pop-os system76-power[951]: [INFO] DBUS Received GetSwitchable method
Nov 22 12:39:01 pop-os system76-power[951]: [INFO] DBUS Received GetGraphics method
Nov 22 12:39:01 pop-os system76-power[951]: [INFO] DBUS Received GetProfile method
Nov 22 12:40:08 pop-os system76-power[951]: [INFO] DBUS Received Performance method
Nov 22 12:40:13 pop-os system76-power[951]: [INFO] DBUS Received Battery method
Nov 22 12:41:01 pop-os system76-power[951]: [INFO] DBUS Received GetGraphics method
Nov 22 12:41:12 pop-os system76-power[951]: [INFO] DBUS Received SetGraphics(nvidia) method
Nov 22 12:41:12 pop-os system76-power[951]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Nov 22 12:41:12 pop-os system76-power[951]: [INFO] Enabling nvidia-fallback.service
Nov 22 12:41:12 pop-os system76-power[951]: Failed to enable unit: Unit file nvidia-fallback.service does not exist.
Nov 22 12:41:12 pop-os system76-power[951]: [WARN] systemctl: failed with exit code: 1 (not an error if service does not exist!)
Nov 22 12:41:12 pop-os system76-power[951]: [INFO] Updating initramfs
Nov 22 12:41:13 pop-os system76-power[951]: update-initramfs: Generating /boot/initrd.img-5.3.0-22-generic
Nov 22 12:41:24 pop-os system76-power[951]: cryptsetup: WARNING: target cryptswap has a random key, skipped
Nov 22 12:41:51 pop-os system76-power[951]: kernelstub.Config    : INFO     Looking for configuration...
Nov 22 12:41:51 pop-os system76-power[951]: kernelstub           : INFO     System information:
[... 263 lines cut ...]
Nov 22 12:51:01 pop-os system76-power[899]: kernelstub.Installer : INFO     Copying Kernel into ESP
Nov 22 12:51:01 pop-os system76-power[899]: kernelstub.Installer : INFO     Copying initrd.img into ESP
Nov 22 12:51:01 pop-os system76-power[899]: kernelstub.Installer : INFO     Setting up loader.conf configuration
Nov 22 12:51:01 pop-os system76-power[899]: kernelstub.Installer : INFO     Making entry file for Pop!_OS
Nov 22 12:51:01 pop-os system76-power[899]: kernelstub.Installer : INFO     Backing up old kernel
Nov 22 12:52:00 pop-os system76-power[899]: [INFO] DBUS Received GetGraphics method
Nov 22 12:52:03 pop-os system76-power[899]: [INFO] DBUS Received SetGraphics(nvidia) method
Nov 22 12:52:03 pop-os system76-power[899]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Nov 22 12:52:03 pop-os system76-power[899]: [INFO] Enabling nvidia-fallback.service
Nov 22 12:52:03 pop-os system76-power[899]: Failed to enable unit: Unit file nvidia-fallback.service does not exist.
Nov 22 12:52:03 pop-os system76-power[899]: [WARN] systemctl: failed with exit code: 1 (not an error if service does not exist!)
Nov 22 12:52:03 pop-os system76-power[899]: [INFO] Updating initramfs
Nov 22 12:52:03 pop-os system76-power[899]: update-initramfs: Generating /boot/initrd.img-5.3.0-22-generic
Nov 22 12:52:06 pop-os system76-power[899]: cryptsetup: WARNING: target cryptswap has a random key, skipped
Nov 22 12:52:16 pop-os system76-power[899]: kernelstub.Config    : INFO     Looking for configuration...
Nov 22 12:52:16 pop-os system76-power[899]: kernelstub           : INFO     System information:
Nov 22 12:52:16 pop-os system76-power[899]:     OS:..................Pop!_OS 18.04
Nov 22 12:52:16 pop-os system76-power[899]:     Root partition:....../dev/sda6
Nov 22 12:52:16 pop-os system76-power[899]:     Root FS UUID:........c7ea262a-7c36-40a8-a384-765a71446281
Nov 22 12:52:16 pop-os system76-power[899]:     ESP Path:............/boot/efi
Nov 22 12:52:16 pop-os system76-power[899]:     ESP Partition:......./dev/sda5
Nov 22 12:52:16 pop-os system76-power[899]:     ESP Partition #:.....5
Nov 22 12:52:16 pop-os system76-power[899]:     NVRAM entry #:.......-1
Nov 22 12:52:16 pop-os system76-power[899]:     Boot Variable #:.....0000
`

crawfxrd, 2019-11-22:
> Try to switch back to nvidia, and the error will happen

It looks like it worked to me. The only two errors I see are the benign missing `nvidia-fallback` and missing hwmon.

> the OS remains using just the intel GPU

How are you determining this?

What is the output of `system76-power graphics`?
What is the output of `system76-power graphics power`?

crawfxrd, 2019-11-22:
Oh, I see. We're logging signals as INFO.
```
[INFO] caught signal: 15
```

miguelcarcamov, 2019-11-22:
> It looks like it worked to me. The only two errors I see are the benign missing `nvidia-fallback` and missing hwmon.
>
>
> How are you determining this?
>
> What is the output of `system76-power graphics`? [...]

I determine this because:

1. The output of nvidia-smi after restart is: NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.
2. Because the GUI "dot" is still marking Intel Graphics.
3. Through terminal when I do `system76-power graphics` the output is `intel`.

> What is the output of system76-power graphics power?

`off (discrete)`

What does that mean?

miguelcarcamov, 2019-11-22:
> Are there any relevant logs not associated with system76-power?

I haven't found anything else... However, what logs would be worth to post here for example?

crawfxrd, 2019-11-22:
- Ensure `/etc/modprobe.d/system76-power.conf` is empty (should just have the autogenerated line)
- Ensure `cat /etc/prime-discrete` is `on`

(These should already have be done by `system76-power graphics nvidia`.)

Then see if `sudo update-initramfs -u` works.

If it does, the NVIDIA modules should then be in the initramfs.

```
lsinitramfs /boot/initrd.img | grep 'nvidia.*\.ko'
```

miguelcarcamov, 2019-11-22:
> Ensure /etc/modprobe.d/system76-power.conf is empty (should just have the autogenerated line)

Done. It's empty and there is only the autogenerated line.

> Ensure cat /etc/prime-discrete is on

It's ON.

> If it does, the NVIDIA modules should then be in the initramfs.
`
update-initramfs: Generating /boot/initrd.img-5.3.0-22-generic
cryptsetup: WARNING: target cryptswap has a random key, skipped
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information: 

    OS:..................Pop!_OS 18.04
    Root partition:....../dev/sda6
    Root FS UUID:........c7ea262a-7c36-40a8-a384-765a71446281
    ESP Path:............/boot/efi
    ESP Partition:......./dev/sda5
    ESP Partition #:.....5
    NVRAM entry #:.......-1
    Boot Variable #:.....0000
    Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
    Kernel Image Path:.../boot/vmlinuz-5.3.0-22-generic
    Initrd Image Path:.../boot/initrd.img-5.3.0-22-generic
    Force-overwrite:.....False

kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
`

`lsinitramfs /boot/initrd.img-5.3.0-22-generic | grep 'nvidia.*\.ko'
lib/modules/5.3.0-22-generic/kernel/drivers/net/ethernet/nvidia/forcedeth.ko
lib/modules/5.3.0-22-generic/kernel/drivers/i2c/busses/i2c-nvidia-gpu.ko
lib/modules/5.3.0-22-generic/updates/dkms/nvidia.ko
lib/modules/5.3.0-22-generic/updates/dkms/nvidia-uvm.ko
`

miguelcarcamov, 2019-11-22:
I restarted after doing that and repeated the steps to see if I the laptop is still using intel and yes, the laptop is still using intel graphics :(.

crawfxrd, 2019-11-22:
Some differences I see from my dev machine are
- using full path for initrd/vmlinuz and not the undecorated symlinks
- missing `nvidia-drm.ko` and `nvidia-modeset.ko` from the initramfs
- missing a 'Making entry...' line after 'Backing up old kernel'

Do those drivers exist on your machine in `/var/lib/dkms/nvidia/4440.31/5.3.0-22-generic/x86_64/module`? (Modesetting should be set in `/usr/lib/modprobe.d/nvidia-graphics-drivers.conf`)

What is the output of `sudo kernelstub -vv`?

miguelcarcamov, 2019-11-22:
> Do those drivers exist on your machine in /var/lib/dkms/nvidia/4440.31/5.3.0-22-generic/x86_64/module? 

```
ls /var/lib/dkms/nvidia/440.31/5.3.0-22-generic/x86_64/module/
nvidia-drm.ko  nvidia.ko  nvidia-modeset.ko  nvidia-uvm.ko
```

> Modesetting should be set in /usr/lib/modprobe.d/nvidia-graphics-drivers.conf

```

ls /usr/lib/modprobe.d/nvidia-graphics-drivers.conf
ls: cannot access '/usr/lib/modprobe.d/nvidia-graphics-drivers.conf': No such file or directory
```

> What is the output of sudo kernelstub -vv?

```
kernelstub           : DEBUG    Got command line options: Namespace(add_options=None, dry_run=False, esp_path=None, force_update=False, initrd_path=None, install_stub=False, k_options=None, kernel_path=None, log_file=None, manage_mode=False, off_loader=False, preserve_live=False, print_config=False, remove_options=None, root_path=None, setup_loader=False, verbosity=2)
kernelstub.Config    : DEBUG    loaded kernelstub.Config
kernelstub.Config    : INFO     Looking for configuration...
kernelstub.Config    : DEBUG    Checking /etc/kernelstub/configuration
kernelstub.Config    : DEBUG    Configuration found!
kernelstub.Config    : DEBUG    {'kernel_options': ['quiet', 'loglevel=0', 'systemd.show_status=false', 'splash'], 'esp_path': '/boot/efi', 'setup_loader': True, 'manage_mode': True, 'force_update': False, 'live_mode': False, 'config_rev': 3}
kernelstub.Config    : DEBUG    Configuration version: 3
kernelstub.Config    : DEBUG    Configuration up to date
kernelstub           : DEBUG    {
  "default": {
    "kernel_options": [
      "quiet",
      "splash"
    ],
    "esp_path": "/boot/efi",
    "setup_loader": false,
    "manage_mode": false,
    "force_update": false,
    "live_mode": false,
    "config_rev": 3
  },
  "user": {
    "kernel_options": [
      "quiet",
      "loglevel=0",
      "systemd.show_status=false",
      "splash"
    ],
    "esp_path": "/boot/efi",
    "setup_loader": true,
    "manage_mode": true,
    "force_update": false,
    "live_mode": false,
    "config_rev": 3
  }
}
kernelstub           : DEBUG    Checking configuration integrity...
kernelstub           : DEBUG    Structing objects
kernelstub.Drive     : DEBUG    loaded kernelstub.Drive
kernelstub.Drive     : DEBUG    root path = /
kernelstub.Drive     : DEBUG    esp_path = /boot/efi
kernelstub.Drive     : DEBUG    Getting a list of drives
kernelstub.Drive     : DEBUG    ['sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0\n', 'proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0\n', 'udev /dev devtmpfs rw,nosuid,relatime,size=8094208k,nr_inodes=2023552,mode=755 0 0\n', 'devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0\n', 'tmpfs /run tmpfs rw,nosuid,noexec,relatime,size=1627460k,mode=755 0 0\n', '/dev/sda6 / ext4 rw,noatime,errors=remount-ro 0 0\n', 'securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0\n', 'tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0\n', 'tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0\n', 'tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0\n', 'cgroup /sys/fs/cgroup/unified cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate 0 0\n', 'cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,name=systemd 0 0\n', 'pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0\n', 'efivarfs /sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0\n', 'cgroup /sys/fs/cgroup/net_cls,net_prio cgroup rw,nosuid,nodev,noexec,relatime,net_cls,net_prio 0 0\n', 'cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpu,cpuacct 0 0\n', 'cgroup /sys/fs/cgroup/hugetlb cgroup rw,nosuid,nodev,noexec,relatime,hugetlb 0 0\n', 'cgroup /sys/fs/cgroup/rdma cgroup rw,nosuid,nodev,noexec,relatime,rdma 0 0\n', 'cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0\n', 'cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0\n', 'cgroup /sys/fs/cgroup/perf_event cgroup rw,nosuid,nodev,noexec,relatime,perf_event 0 0\n', 'cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0\n', 'cgroup /sys/fs/cgroup/pids cgroup rw,nosuid,nodev,noexec,relatime,pids 0 0\n', 'cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0\n', 'cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0\n', 'systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=27,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=16587 0 0\n', 'debugfs /sys/kernel/debug debugfs rw,relatime 0 0\n', 'hugetlbfs /dev/hugepages hugetlbfs rw,relatime,pagesize=2M 0 0\n', 'mqueue /dev/mqueue mqueue rw,relatime 0 0\n', 'tracefs /sys/kernel/debug/tracing tracefs rw,relatime 0 0\n', 'fusectl /sys/fs/fuse/connections fusectl rw,relatime 0 0\n', 'configfs /sys/kernel/config configfs rw,relatime 0 0\n', '/dev/loop0 /snap/core/8039 squashfs ro,nodev,relatime 0 0\n', '/dev/sda5 /boot/efi vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro 0 0\n', 'tmpfs /run/user/118 tmpfs rw,nosuid,nodev,relatime,size=1627456k,mode=700,uid=118,gid=124 0 0\n', 'tmpfs /run/user/1000 tmpfs rw,nosuid,nodev,relatime,size=1627456k,mode=700,uid=1000,gid=1000 0 0\n', 'gvfsd-fuse /run/user/1000/gvfs fuse.gvfsd-fuse rw,nosuid,nodev,relatime,user_id=1000,group_id=1000 0 0\n']
kernelstub.Drive     : DEBUG    Getting the block device file for /
[... cut ...]
kernelstub.Installer : DEBUG    [Errno 2] No such file or directory: '/boot/initrd.img-5.3.0-22-generic.old'
kernelstub.Installer : DEBUG    Couldn't back up old initrd.img. There's probably only one kernel installed.
kernelstub.Installer : DEBUG    Copying: /proc/cmdline => /boot/efi/EFI/Pop_OS-c7ea262a-7c36-40a8-a384-765a71446281
kernelstub           : DEBUG    Saving configuration to file
kernelstub.Config    : DEBUG    Saving configuration to /etc/kernelstub/configuration
kernelstub.Config    : DEBUG    Configuration saved!
kernelstub           : DEBUG    Setup complete!
```

crawfxrd, 2019-11-22:
`/lib/modprobe.d/nvidia-graphics-drivers.conf` is installed by `nvidia-kernel-common-440`. If it doesn't exist then something is at least wrong with the NVIDIA driver installation. It may be worth trying to reinstall the NVIDIA packages.

Are you able to manually load the NVIDIA drivers (`sudo modprobe nvidia; lsmod | grep nvidia`)?

miguelcarcamov, 2019-11-22:
> /lib/modprobe.d/nvidia-graphics-drivers.conf is installed by nvidia-kernel-common-440. If it doesn't exist then something is at least wrong with the NVIDIA driver installation. It may be worth trying to reinstall the NVIDIA packages.

Yeah, this path is different from the other in fact, the file exists and contains:

```
blacklist nouveau
blacklist lbm-nouveau
alias nouveau off
alias lbm-nouveau off

options nvidia-drm modeset=1
```

> Are you able to manually load the NVIDIA drivers (sudo modprobe nvidia; lsmod | grep nvidia)?

```
sudo modprobe nvidia; lsmod | grep nvidia
modprobe: ERROR: ../libkmod/libkmod-module.c:832 kmod_module_insert_module() could not find module by name='off'
modprobe: ERROR: could not insert 'off': Unknown symbol in module, or unknown parameter (see dmesg)
```

crawfxrd, 2019-11-22:
Yeah, I didn't consider that 18.04 was pre-UsrMerge.

> ERROR: could not insert 'off': Unknown symbol in module, or unknown parameter (see dmesg)

Is `/etc/modprobe.d/system76-power.conf` still empty? If so, something else is blacklisting the NVIDIA modules; try `grep -r 'blacklist nvidia' /{etc,lib}/modprobe.d/`.

miguelcarcamov, 2019-11-22:
> ls /etc/modprobe.d/system76-power.conf still empty?
Yes, still empty

> try grep -r 'blacklist nvidia' /{etc,lib}/modprobe.d/.

```
grep -r 'blacklist nvidia' /{etc,lib}/modprobe.d/.
/etc/modprobe.d/./blacklist-framebuffer.conf:blacklist nvidiafb
/lib/modprobe.d/./blacklist-nvidia.conf:blacklist nvidia
/lib/modprobe.d/./blacklist-nvidia.conf:blacklist nvidia-drm
/lib/modprobe.d/./blacklist-nvidia.conf:blacklist nvidia-modeset
```

crawfxrd, 2019-11-22:
Alright, there's the answer. Remove `/lib/modprobe.d/blacklist-nvidia.conf` and reboot, and you should be good to go.

(Is that file associated with a package? `dpkg -S /lib/modprobe.d/blacklist-nvidia.conf`)

miguelcarcamov, 2019-11-22:
> Is that file associated with a package? dpkg -S /lib/modprobe.d/blacklist-nvidia.conf

Doesn't seem so.
```
dpkg -S /lib/modprobe.d/blacklist-nvidia.conf
dpkg-query: no path found matching pattern /lib/modprobe.d/blacklist-nvidia.conf
```

miguelcarcamov, 2019-11-22:
Yes! That worked! Thank you!

Hope this answer could help anyone with the same issue.

closed 2019-11-22 by crawfxrd