pop-os/system76-power #131: System76-Power options no longer visable.

Philbohackinz, 2019-12-04:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="19.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.10"
VERSION_ID="19.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=eoan
UBUNTU_CODENAME=eoan
LOGO=distributor-logo-pop-os
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
System76-Power

**Issue/Bug Description:**
I no longer have power menu options to change video cards and power profiles. 
[image]

**Steps to reproduce (if you know):**

My guess is the recent update  for the system76-power or a recent update has changed things. 

**Expected behavior:** 
Have an option to change my power options and graphics card in the menu again. 

**Other Notes:**
I am still able to change my graphics cards and power profiles in the terminal.

crawfxrd, 2019-12-04:
Is the extension disabled? You can check with Tweaks (`gnome-tweak-tool`).

Philbohackinz, 2019-12-05:
That did the trick. Somehow all extensions were disabled.

closed 2019-12-05 by Philbohackinz