pop-os/system76-power #143: System boots into black screen after switching to Intel graphics

digital-idiot, 2020-01-11:
**Hardware / Model**

- System76 Adder WS

**Distribution**
 ```
NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
PRETTY_NAME="Manjaro Linux"
ANSI_COLOR="1;32"
HOME_URL="https://www.manjaro.org/"
SUPPORT_URL="https://www.manjaro.org/"
BUG_REPORT_URL="https://bugs.manjaro.org/"
LOGO=manjarolinux
```

**Related Applications and Package Versions**

- system76-driver-git 19.04.20.r2.g5445255-1
- system76-dkms-git 109-1
- system76-acpi-dkms 1.0.1-1
- system76-oled 0.1.3-2
- system76-io-dkms-git r26.ea5f61a-2- system76-power-git r233.44945a5-1
- video-nvidia-440xx

**Issue/Bug Description:**
System freezes in to black screen on boot after switching into `intel` gpu using `system76-power` utility. When freezed / booted into black screen, can't even switch to other `tty` or `console`

**Steps to reproduce (if you know):**

1. Switch to `intel` GPU using `system76-power graphics intel`
2. Reboot 

**Expected behavior:**
- Succesfully boot into WM on `intel` GPU

**Other Notes:**
- I will be happy to provide any `log` as required.
- [A similar issue](https://github.com/pop-os/pop/issues/194) reported to Pop OS.

brs17, 2020-01-13:
I would recommend creating a System76 support ticket so that the team can really dig in and help you get this running. This does currently work in Pop!_OS, so the question will be what differences do you have in your manjaro install, as that is probably the causing issue. I would also recommend linking this issue so that this issue can be updated as more information is discovered.

digital-idiot, 2020-01-14:
> I would recommend creating a System76 support ticket so that the team can really dig in and help you get this running. This does currently work in Pop!_OS, so the question will be what differences [...]

@brs17 I opened a support ticket and as you said 'it works on Pop OS' they replied the same thing. However it would be great, if this can be addressed somehow. 

If `system76-*` are not causing this, as everything works fine in Pop OS, I suspect `xorg.conf` is causing this. I don't have a direct `/etc/X11/xorg.conf` but this is my `/etc/X11/xorg.conf.d/optimus.conf`:
```
Section "Module"
    Load "modesetting"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    BusID "PCI:1:0:0"
    Option "AllowEmptyInitialConfiguration"
EndSection
```

Do I need to append:
```
Section "Device"
    Identifier "Intel Graphics"
    Driver "intel"
    BusID "PCI:0:2:0" 
EndSection
```
to it?

crawfxrd, 2020-01-14:
On Ubuntu, `gpu-manager` will add/remove Xorg configs depending on the mode set in `/etc/prime-discrete`. In the case of `off` (Intel mode), it will remove the NVIDIA Xorg configs before starting the display manager.

digital-idiot, 2020-02-13:
It seems, the issue has been solved with new `prime` enabled driver and `Linux 5.4.18-2`. I have the following graphics drivers currently installed:

```
> Installed PCI configs:
--------------------------------------------------------------------------------------------------
                  NAME                             VERSION              FREEDRIVER           TYPE
--------------------------------------------------------------------------------------------------
           video-linux                           2018.05.04                true              PCI
video-hybrid-intel-nvidia-440xx-prime            2019.10.25               false              PCI

```

The graphics switching works. But simple logging out and logging back in does not make the switching to take effect, a system restart is required.

closed 2020-02-13 by digital-idiot