pop-os/system76-power #145: hybrid mode - screen tearing / corruption after resume

jucajuca, 2020-01-17:
**Distribution (run `cat /etc/os-release`):**
 
```
NAME="Pop!_OS"
VERSION="19.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 19.10"
VERSION_ID="19.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=eoan
UBUNTU_CODENAME=eoan
LOGO=distributor-logo-pop-os
```
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

```
nvidia-driver-440:
  Installed: 440.44-1pop1~1576524804~19.10~fb5b8b3
  Candidate: 440.44-1pop1~1576524804~19.10~fb5b8b3
  Version table:
     440.48.02-0ubuntu0~~19.10.1 500
        500 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu eoan/main amd64 Packages
 *** 440.44-1pop1~1576524804~19.10~fb5b8b3 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu eoan/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**

In **hybrid mode** the screen works normally after the power up. However after sleeping, resuming there is a horrible amount of screen tearing / corruption that makes the laptop unusable. Apparently PRIME is not activated.

I tried this without any improvement. http://ubuntuhandbook.org/index.php/2018/07/fix-screen-tearing-ubuntu-18-04-optimus-laptops/

**Steps to reproduce (if you know):**

power up, sleep, resume

**Expected behavior:**

Screen should work normally.

**Other Notes:**

xrandr --properties 
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 194mm
	EDID: 
		00ffffffffffff0030e4c00500000000
		001b0104a54f0078eae085a3544e9b26
		0e505400000001010101010101010101
		0101010101015e8780a070384d403020
		350058c21000001b683880a070384d40
		3020350058c21000001b000000000000
		00000000000000000000000000000002
		000c30f20f3c962e12309e00000000a8
	scaling mode: Full aspect 
		supported: Full, Center, Full aspect
	max bpc: 12 
		range: (6, 12)
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 86 
		supported: 86
	non-desktop: 0 
		range: (0, 1)
   1920x1080    144.00*+  60.01    59.97    59.96    60.00    59.93  
   1680x1050     84.94    74.89    69.88    59.95    59.88  
....
```

sudo lshw -C video
```
  *-display                 
       description: VGA compatible controller
       product: GP106M [GeForce GTX 1060 Mobile]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:133 memory:a5000000-a5ffffff memory:90000000-9fffffff memory:a0000000-a1ffffff ioport:6000(size=128) memory:a6000000-a607ffff
  *-display
       description: VGA compatible controller
       product: UHD Graphics 630 (Mobile)
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:132 memory:a4000000-a4ffffff memory:80000000-8fffffff ioport:7000(size=64) memory:c0000-dffff

```

nvidia-settings -q NvidiaDriverVersion

```
Gtk-Message: 14:50:12.413: Failed to load module "appmenu-gtk-module"

  Attribute 'NvidiaDriverVersion' (pop-os:1[gpu:0]): 440.44

```

jucajuca, 2020-01-17:
update: it also happens in **intel graphics**.

in **nvidia mode** the flickering and tearing is **always** there.

paulocosta-gmu, 2020-09-21:
I have exactly the same issue. All works superbly after booting, but after waking from suspend both Firefox and Chrome UI get corrupted.
However, it is not a hybrid laptop. It is a nvidia-only computer (AMD Threadripper 1950 CPU with Nvidia 1080 GPU. It only happens in PopOS. I have Fedora and Manjaro installed and running stable for months and never faced this issue. PopOS is my daily driver in my laptop for years but I simply cannot use in the main computer. I did three installations already but the problem persists.
I'm using nivida 440.100 driver.

> ⟪pcdesktop-pop⟫ in ~ via C base with /usr/bin/zsh 
🕙[ 22:18:32 ] ➜ xrandr --properties
Screen 0: minimum 8 x 8, current 6400 x 1600, maximum 32767 x 32767
DVI-D-0 disconnected (normal left inverted right x axis y axis)
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: DVI-D 
	ConnectorNumber: 0 
	_ConnectorLocation: 0 
	non-desktop: 0 
		supported: 0, 1
HDMI-0 disconnected (normal left inverted right x axis y axis)
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: HDMI 
	ConnectorNumber: 3 
	_ConnectorLocation: 3 
	non-desktop: 0 
		supported: 0, 1
DP-0 disconnected (normal left inverted right x axis y axis)
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: DisplayPort 
		supported: DisplayPort
	ConnectorType: DisplayPort 
	ConnectorNumber: 4 
	_ConnectorLocation: 4 
	non-desktop: 0 
		supported: 0, 1
DP-1 disconnected (normal left inverted right x axis y axis)
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: DisplayPort 
	ConnectorNumber: 4 
	_ConnectorLocation: 4 
	non-desktop: 0 
		supported: 0, 1
DP-2 connected 2560x1600+0+0 (normal left inverted right x axis y axis) 641mm x 400mm
	_MUTTER_PRESENTATION_OUTPUT: 0 
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	EDID: 
		00ffffffffffff00046257a000000000
		1e170104803c2278288e05ad4f33b026
		0d50542108008bc0b30081c08100a940
		81809500d100b06800a0a0402e603020
		360081902100001e0000000000000000
		00000000000000000000000000fd0032
		3d1d711c0000000000000000000000fc
[... 97 lines cut ...]
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: DisplayPort 
	ConnectorNumber: 1 
	_ConnectorLocation: 1 
	non-desktop: 0 
		supported: 0, 1

> ⟪pcdesktop-pop⟫ in ~ via C base with /usr/bin/zsh 
🕙[ 22:24:27 ] ➜ sudo lshw -C video
  *-display                 
       description: VGA compatible controller
       product: GP104 [GeForce GTX 1080]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:0a:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:189 memory:d8000000-d8ffffff memory:c0000000-cfffffff memory:d0000000-d1ffffff ioport:1000(size=128) memory:c0000-dffff

shiftlessbadger, 2020-09-30:
I have the same problem -- AMD Ryzen 7 2700X + NVIDIA GT GTX 1030 running pop os 20.04, severe graphics corruption after resume from suspend. I'm using driver 450.66, which is the newest driver I can find, and I can't find anything about this specific bug online. Anyone having any luck?

> brock@pop-os:~$ xrandr --properties
Screen 0: minimum 8 x 8, current 1920 x 1080, maximum 32767 x 32767
DVI-D-0 disconnected (normal left inverted right x axis y axis)
	CTM: 0 1 0 0 0 0 0 0 0 1 0 0 0 0 0 0 
		0 1 
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: DVI-D 
	ConnectorNumber: 0 
	_ConnectorLocation: 0 
	non-desktop: 0 
		supported: 0, 1
HDMI-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 509mm x 286mm
	_MUTTER_PRESENTATION_OUTPUT: 0 
	CTM: 0 1 0 0 0 0 0 0 0 1 0 0 0 0 0 0 
		0 1 
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	EDID: 
		00ffffffffffff000469c42375310000
		1419010380331d782ae595a656529d27
		105054bfef00d1c0b300950081808140
		81c0714f0101023a801871382d40582c
		4500fd1e1100001e000000ff0046354c
		4d544a3031323636310a000000fd0032
		4b185311000a202020202020000000fc
		00415355532056433233390a2020015b
		020322714f0102031112130414050e0f
		1d1e1f90230917078301000065030c00
		10008c0ad08a20e02d10103e9600fd1e
		11000018011d007251d01e206e285500
		fd1e1100001e011d00bc52d01e20b828
		5540fd1e1100001e8c0ad09020403120
		0c405500fd1e11000018000000000000
		00000000000000000000000000000069
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: HDMI 
	ConnectorNumber: 1 
	_ConnectorLocation: 1 
	non-desktop: 0 
		supported: 0, 1
   1920x1080     60.00*+  59.94    50.00    60.00    50.04  
   1680x1050     59.95  
   1440x900      59.89  
   1440x576      50.00  
   1440x480      59.94  
   1280x1024     75.02    60.02  
   1280x960      60.00  
   1280x720      60.00    59.94    50.00  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   800x600       75.00    72.19    60.32    56.25  
   720x576       50.00  
   720x480       59.94  
   640x480       75.00    72.81    59.93    59.94 

> sudo lshw -C video
  *-display                 
       description: VGA compatible controller
       product: GP108 [GeForce GT 1030]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:26:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: irq:84 memory:f6000000-f6ffffff memory:e0000000-efffffff memory:f0000000-f1ffffff ioport:e000(size=128) memory:c0000-dffff

gibbsoft, 2020-11-26:
I have the same problem when connected to an external monitor over usb-c on a System76 Lemur9 running pop-os 20.10. 

```
$ xrandr --properties
Screen 0: minimum 320 x 200, current 3840 x 1080, maximum 16384 x 16384
eDP-1 connected (normal left inverted right x axis y axis)
	_MUTTER_PRESENTATION_OUTPUT: 0 
	EDID: 
		00ffffffffffff000daed71400000000
		2c1a0104951f117802ee95a3544c9926
		0f505400000001010101010101010101
		010101010101b43b804a713834403020
		350035ad1000001ab43b804a71386a42
		3020350035ad1000001a000000000000
		00000000000000000000000000000002
		000c2dff113caa150c23aa0000000069
	scaling mode: Full aspect 
		supported: Full, Center, Full aspect
	Colorspace: Default 
		supported: Default, RGB_Wide_Gamut_Fixed_Point, RGB_Wide_Gamut_Floating_Point, opRGB, DCI-P3_RGB_D65, BT2020_RGB, BT601_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, BT2020_CYCC, BT2020_YCC
	max bpc: 12 
		range: (6, 12)
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	panel orientation: Normal 
		supported: Normal, Upside Down, Left Side Up, Right Side Up
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 95 
		supported: 95
	non-desktop: 0 
		range: (0, 1)
   1920x1080     60.01 +  60.01    59.97    59.96    40.01    59.93  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
[... 133 lines cut ...]
	non-desktop: 0 
		range: (0, 1)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)
	PRIME Synchronization: 1 
		supported: 0, 1
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 34 
		supported: 34
	non-desktop: 0 
		range: (0, 1)
$ sudo lshw -C video
  *-display                 
       description: VGA compatible controller
       product: UHD Graphics
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 02
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:145 memory:b0000000-b0ffffff memory:a0000000-afffffff ioport:1000(size=64) memory:c0000-dffff
```

zot, 2021-01-12:
Looks like I'm getting the same problem. Tearing occurs only on my laptop screen. I just restarted, suspended, and resumed and I'm not getting tearing yet. I think for me it may not be related to suspending.

Here's what it looks like (had to capture this with my phone -- Kazam doesn't record the tearing so it must not be a corruption in video memory). I hope this looping isn't too annoying.

[image]

btallin, 2021-01-13:
I have the same issue on 20.04 when running hybrid or integrated graphics mode and it appears on my laptop display and external monitors. The issue disappears if I disable the built-in display when connected to connected to external monitors, run in nvidia graphics mode or use Wayland with integrated graphics.
```
$ xrandr --properties
Screen 0: minimum 320 x 200, current 3072 x 1920, maximum 16384 x 16384
eDP-1 connected primary 3072x1920+0+0 (normal left inverted right x axis y axis) 336mm x 210mm
	_MUTTER_PRESENTATION_OUTPUT: 0 
	EDID: 
		00ffffffffffff004d10d11400000000
		041e0104a52215780ede50a3544c9926
		0f505400000001010101010101010101
		010101010101283c80a070b023403020
		360050d210000018203080a070b02340
		3020360050d210000018000000fe0046
		4b52314b804c513135364e3100000000
		0002410332001200000a010a2020003a
	scaling mode: Full aspect 
		supported: Full, Center, Full aspect
	Colorspace: Default 
		supported: Default, RGB_Wide_Gamut_Fixed_Point, RGB_Wide_Gamut_Floating_Point, opRGB, DCI-P3_RGB_D65, BT2020_RGB, BT601_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, BT2020_CYCC, BT2020_YCC
	max bpc: 12 
		range: (6, 12)
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	panel orientation: Normal 
		supported: Normal, Upside Down, Left Side Up, Right Side Up
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 95 
		supported: 95
	non-desktop: 0 
		range: (0, 1)
   1920x1200     59.95 +  59.88*   47.96  
   1920x1080     60.01    59.97    59.96    59.93  
   1600x1200     60.00  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
[... 70 lines cut ...]
       description: 3D controller
       product: NVIDIA Corporation
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list
       configuration: driver=nvidia latency=0
       resources: irq:180 memory:b3000000-b3ffffff memory:70000000-7fffffff memory:80000000-81ffffff ioport:3000(size=128)
  *-display
       description: VGA compatible controller
       product: UHD Graphics
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       logical name: /dev/fb0
       version: 05
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom fb
       configuration: depth=32 driver=i915 latency=0 mode=1920x1200 visual=truecolor xres=1920 yres=1200
       resources: iomemory:600-5ff iomemory:400-3ff irq:161 memory:604a000000-604affffff memory:4000000000-400fffffff ioport:4000(size=64) memory:c0000-dffff
```

Dustoe, 2021-01-18:
I don't really know what to add to the discussion but the tearing is less major for me on nvidia graphics mode but is still present on certain games. Do you have ideas on what could be the problem for me?