pop-os/system76-power #146: top bar power controls controls GUI changed - no option to change graphics - nvidia/inter/hybrid

joewav, 2020-01-24:
pop-os 19.10 
 
 
I only get "power setting" where before I had the option of switching between nvidia and intel (or hybrid).

I've seen a similar issue reported as a bug before, but issue closed last year around April: https://github.com/pop-os/gnome-shell-extension-system76-power/issues/20

as a related side note, all the GUI for the top right menu changed: including the icon for the power charge, and also now to switch between power off and sleep, I need to hold down the shutdown icon until it changes to pause icon. 

Doesn't seem to be a problem to switch through terminal

crawfxrd, 2020-01-24:
Is the extension enabled?

joewav, 2020-01-24:
@crawfxrd all extensions were disabled 0_o 
thx

closed 2020-01-24 by brs17