pop-os/system76-power #153: Add ability to set an application to "Always use dedicated graphics"

Booteille, 2020-05-01:
Actually, this is awesome to just have to right click on a icon to use the dedicated graphics in hybrid mode.

Though, it's not useful for a keyboard experience to always have to use the mouse when wanting to launch a specific application with the dedicated GPU.

So, that would be awesome to be able to always start an application using the dedicated GPU.

crawfxrd, 2020-05-01:
There is a desktop entry key to control this, but that still requires launching by clicking the application icon so that it will read the file.

If you're running the binary from the command line you still need to specify the environment variables. Some possible solutions are, using Firefox as an example:

- Use `switcherooctl` from [switcheroo-control](https://gitlab.freedesktop.org/hadess/switcheroo-control) 2.3
```
switcherooctl launch firefox
```
- Create alias for the env vars and prefix commands as needed
```
alias prime="__GLX_VENDOR_LIBRARY_NAME=nvidia __NV_PRIME_RENDER_OFFLOAD=1"
prime firefox
```
- Alias commands to always run with the env vars set, and escape it if you don't want it set
```
alias firefox="__GLX_VENDOR_LIBRARY_NAME=nvidia __NV_PRIME_RENDER_OFFLOAD=1 firefox"
# Run on dGPU
firefox
# Run on iGPU
\firefox
```

Booteille, 2020-05-01:
> There is a desktop entry key to control this, but that still requires launching by clicking the application icon so that it will read the file.

Which one, exactly?

Thanks for tips about how to use it on the command line but I was mainly thinking about a simple checkbox, somewhere (why not when right-clicking on the desktop icon).
The idea is to make the option userfriendly. To don't have to find it by looking on the internet and modifying manually a file.

crawfxrd, 2020-05-01:
> Which one, exactly?

`PrefersNonDefaultGPU`, being [integrated into GNOME](https://gitlab.gnome.org/GNOME/gnome-shell/-/merge_requests/1226). We are currently using the non-standard `X-KDE-RunOnDiscreteGpu`, but this will change to the standardized key (pop-os/gnome-shell#24).

> thinking about a simple checkbox

This is the [next phase for integration](https://github.com/pop-os/system76-power/issues/125). I've mainly been waiting for the standardized key and upstream progress to start working on it again.

Booteille, 2020-05-01:
Great. Thanks a lot for your support!

crawfxrd, 2021-04-30:
Closing this since it's out of scope of this package.

g-c-c could potentially be integrated. I don't think launching through Nautilus or apps set to launch at startup work either.

And running from the command line will always require exporting the variables in some manner (left up to the user how to do that) since it's not controlled by the DE.

closed 2021-04-30 by crawfxrd