pop-os/system76-power #154: "Launch with Dedicated graphics" option not working with 'Dash-to-dock' extension

the1derer, 2020-05-04:
**Distribution**
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os

**Related Application and/or Package Version**
system76-power:
  Installed: 1.1.4~1587778238~20.04~08ef39d
  Candidate: 1.1.4~1587778238~20.04~08ef39d
  Version table:
 *** 1.1.4~1587778238~20.04~08ef39d 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
When clicking the option "Launch with Dedicated Graphics" while using "Hybrid Graphics" on the applications pinned on side dock installed via "Dash-to-Dock" extension of gnome-shell. The program is not started.

Although this option("Launch with Dedicated Graphics") works correctly when starting applications through the application menu.

**Steps to reproduce (if you know):**
1. Activate "Hybrid Graphics" from power option. 
2. Install [Dash-to-Dock extension](https://extensions.gnome.org/extension/307/dash-to-dock/)
3. Pin an application on this dock by using "Add to favourite"
4. Now right-click on the icon pinned on "Dash-to-Dock" and select "Launch using dedicated graphics"

**Expected behaviour:**
Should start the application using Nvidia graphics.

crawfxrd, 2020-05-04:
This is an issue with Dash-to-Dock (or any other launcher). [`_rebuildMenu()`](https://github.com/micheleg/dash-to-dock/blob/1788f31b049b622f78d0e65c56bef76169022ca9/appIcons.js#L820-L831) needs to be updated for GNOME 3.36.

crawfxrd, 2020-05-04:
Sorry, was looking at the wrong base. This is caused by us using a backport of gnome-shell.