pop-os/system76-power #158: Add "target frequency" slider to "performance mode"

ayoungethan, 2020-05-28:
I have been researching how Pop!_OS can be further optimized for creators who depend on low latency work. This is one of several optimizations that I think would have positive overall benefits for all Pop!_OS users (whereas at least one optimization won't really matter to anyone who doesn't need reliable low-jitter, low latency performance).

as per https://github.com/pop-os/system76-power/issues/108#issuecomment-635633315

The only times I can think of where you really need a different performance parameter than "Balanced" are three scenarios:

1.  For low latency applications (where you want a consistent frequency)
2.  For low latency throughput (where you want a consistent HIGH frequency)
3.  For situations where something in the computer has gone haywire and hijacked the CPU and you need to limit the max frequency to buy yourself some time before maxed-out CPU runs down your battery. That, IMO, is not a good reason to have something in the GUI, and if there was something there, it should be labeled "panic button" and ideally also help the user pause a runaway process so that they can evaluate how to proceed with a more level head (e.g., save work, kill the process via system monitor and/or restart the computer).

So, really, there are two main modes that mostly matter (ha!):

1. Balanced, for most things that most people do and
2. A "Performance" mode that lets you set a stable CPU frequency of your choice. This is really important for low latency audio work, for example, to minimize jitter and maximize throughput if you are doing a lot of DSP. A slider would let you adjust the target frequency. I say "target" because no guarantees with thermals, especially in a laptop. But it would let the user who needs low latency to choose an appropriate frequency sufficient for their DSP workload and find some balance with battery life if necessary, or help control fan noise (e.g., during a recording process).

This, in my opinion (2 settings with a target frequency slider in performance mode), would be much more simple and effective for a broader range of use cases!

ayoungethan, 2020-05-28:
The other half of this is https://github.com/pop-os/system76-power/issues/159

ayoungethan, 2020-06-02:
Here is another way to implement the same idea with a single slider adjustment: https://github.com/falkTX/Cadence/issues/250

I don't think it is better or worse. So the far left would be the same as selecting "balanced" power profile, and every other adjustment would equate to selecting the "performance" power profile at a specific target frequency.

curiousercreative, 2021-01-25:
@ayoungethan for use case #1 and 2, would a "perfomance" governor not suffice? While I agree that this governor setting could and should be bundled into the power profile, in the meantime you can accomplish this via cpupower-gui (found in pop store). I like the novelty of a frequency slider, but even that is quite complicated. Drop battery mode, but don't add a frequency slider IMO.

ayoungethan, 2021-01-25:
Good question!
Performance mode covers #1 definitely.

 For #2, I think any option that allows to set a stable custom CPU
frequency is needed. It doesn't need to be a slider. It could be a custom
performance profile where you can manually set the cpu frequency and
governor. This would also allow someone to recovery the obsoleted mode if
they decide they actually need it for some reason...

The reason a lower stable frequency is needed is for heat/noise/power
management during low latency use cases where overall throughput demand is
still relatively low.

This is most applicable for laptops as they run on battery, and often have
very little thermal regulation capacity that doesn't involve high and noisy
fan speeds and throttling.

A perfect use case is a laptop being used as a hard drive recorder with an
audio interface for recording an acoustic performance in an intimate
setting: it needs consistent and stable performance to help prevent buffer
under runs (no overhead and context switching of cpu frequency scaling),
but doesn't have large CPU i/o needs (just needs to handle bus throughput
with minimal processing), and requirements are skewed toward battery life
(so the recorder doesn't die and isn't tied to a power plug) and cool/quiet
operation (so it is not adding meaningfully to the background noise floor).

curiousercreative, 2021-01-26:
I'd love to see support for custom profiles

kylebakerio, 2021-02-27:
I also use cpupower-gui, and it's effective for this. It would be neat to see support for custom profile tuning somewhere but I'm not sure how helpful it would be for the vast majority of users.

ayoungethan, 2021-02-28:
I think it is more broadly socially and psychologically useful in that it
allows developers to design effective power management defaults and then
tell a vocal minority of users, eg, who complain about missing options they
think they need, that they can just create a custom profile vs tell
developers that they need to implement a default setting that doesn't make
sense in the vast majority of cases. it gives developers an "out"

and it is also very useful for niche use cases. this is why every other OS
has easily tweakable power management settings on one level or another.
Although the discussion of which settings to make easily tweakable and why
is a much more interesting and involved one...!