pop-os/system76-power #159: Eliminate "Battery Life" mode

ayoungethan, 2020-05-28:
I think Pop!_OS is better off without a "battery life" mode, as it doesn't do what it says it should do ("balanced" mode does it better and with better user experience!) and clutters up the interface.

as discussed here
https://github.com/pop-os/system76-power/issues/108#issuecomment-635633315

> as someone who LOVES long battery life (I have a Darter Pro that can give me 10-14 hrs of work time), I actually recommend getting rid of the "battery life" mode altogether. It doesn't actually save battery life, and in my (and I suspect, most) use cases it actually causes slightly worse battery life. It violates the "race to sleep" philosophy of integrated throughput performance and power saving. 

and here
https://github.com/pop-os/system76-power/issues/9#issuecomment-501518420

> the CPU tuning does not itself necessarily lead to more effective battery usage. For example, on my system, I notice in "battery life" mode via powertop that baseline computer power usage increases, which appears due to the computer entering into deeper c-states less frequently, and staying there for shorter periods of time, which means higher baseline power usage, as well as lower maximum power usage, but longer periods of time at the max power state.
> 
> This resembles older debates about "ondemand" vs "conservative" CPU frequency scaling governor settings, where people found that "ondemand" actually gave better performance AND battery life in MOST use cases. In the new Intel pstates governor, even the "performance" governor is supposed to allow for power saving equal to the old "ondemand" acpi governor.
> 
> This to me begs the question of whether "battery life" should include any CPU tuning at all. Simply downclocking the CPU does not necessarily lead to an intended outcome of reduced power consumption.
> 
> In my case, I am dealing with low latency audio processing. "battery life" only provides better battery life when doing dynamic open-ended (vs finite) CPU-intensive tasks (such as playing a virtual instrument), by limiting the overall CPU power available. It does NOT necessarily provide better battery life in standard use scenarios, including finite CPU-intensive tasks with well-defined completion parameters (such as transcoding an audio file), because it makes the processor work longer before it re-enters an idle sleep state. 

Thermal / fan noise management can be better managed with "target frequency" slider under "performance mode" that can be set to any native frequency range. High for throughput-intensive tasks, low for battery and thermal-conscious low latency and low jitter performance (eg for recording), as per https://github.com/pop-os/system76-power/issues/158

kylebakerio, 2021-02-27:
This is an interesting concept, but it would be great to actually get some more data on this before pulling the trigger on it. If nothing else, I think many users would find it frustrating that there was not a 'battery saver' mode and would see that as a shortcoming, and attempt to remedy it by replicating it themselves, leaving bug reports, or even moving to different distributions as a result. It might be tricky to communicate this successfully to users, and likely a 'battery saver' button needs to exist even if it effectively does almost nothing at all. (Another proposal talked about turning off animations, which is an interesting thought, and perhaps more features like that could be focused on instead that might make more of a difference?)

mkellyxp, 2021-03-19:
I'm just going to come in here and say that I use "Battery Life" mode by default on my Darp7 and my Lemp9 for thermal / fan noise reasons.  I don't know how much it saves battery.. but I do know my computer runs cooler and quieter in Battery Life mode, and I love it.  I realize i'm trading this for performance, and totally ok with that.

It would be a pity to remove an option that actually does something, that people like myself use.  To my knowledge it never automatically goes in Battery Life mode, so what's the harm in simply ignoring it?  I don't use tiling, but no need to remove it

kylebakerio, 2021-03-19:
That's valuable feedback, and interesting to note. Might make sense to rename it, but I'm not sure what that would be--"cool and quiet mode"? "low-performance mode"?

leviport, 2021-03-19:
@mkellyxp is right about there being huge differences on laptops with System76 open firmware, like the darp7 and lemp9. The different performance modes have completely different wattage limits and custom-tailored fan curves for each of them. It's a bummer more can't be done on machines that don't talk to the OS more, but we definitely want to keep the Battery Life mode in for users that have hardware that runs our firmware.

closed 2021-03-19 by leviport

ayoungethan, 2021-03-19:
fwiw i had a recent s76 laptop (darter pro) and battery life was always
uniformly worse on "battery life" mode by an hour or two. cpu spent more
time awake due to cpu throttling and performance suffered.

i also used that mode when i was concerned about fan noise or heat, which
was much more marginal than concern over cord-free work time.

there is confusion here in the discussion. my main concern is with cpu
throttling as an incongruent means of extending battery life, NOT with
other system settings that do not violate race to sleep. i don't have a
problem with a "battery life" mode as long as it actually does what it says
it does, which throttling does not. "cool and quiet" or "low performance"
is a much more accurate name for a throttled mode.

a throttled mode shouldn't be called "battery life" as that is a misnomer.
power saving modes should use race to sleep as a primary strategy, and
synchronized wake ups as a secondary strategy in order to allow as many
system components to sleep for longer.

unless someone disputes the race to sleep framework...then i suppose take
that up with chip manufacturers? As i understand it, that is their strategy
based on their research, and they are just depending on OEMs and software
developers to implement it correctly.

cheers
ethan