pop-os/system76-power #161: Higher battery usage on AMD APU compared to Nvidia GPU

Skrilltrax, 2020-06-03:
**Distribution (run `cat /etc/os-release`):** 
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.1.4\~1590091483\~20.04\~15264d3
  Candidate: 1.1.4\~1590091483\~20.04\~15264d3
  Version table:
 *** 1.1.4~1590091483\~20.04\~15264d3 100
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
Battery usage is almost twice when using the integrated only mode compared to the Nvidia mode or the Hybrid Prime offloading mode. My laptop configuration is Ryzen 7 3750H with an Nvidia GTX1660Ti MaxQ. I think the problem is that the Nvidia card is still running in the background. Attaching the gpu-manager.log while using Integrated mode.

**Steps to reproduce (if you know):**
Switch to Integrated mode check usage on Powertop and then switch back to Hybrid mode and check Powertop again.

**Expected behavior:**
The integrated mode should use less battery than Nvidia mode.

**Other Notes:**
gpu-manager.log (Integrated mode)

log_file: /var/log/gpu-manager.log
last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /run/u-d-c-nvidia-was-loaded file
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/5.6.15-050615-generic/updates/dkms
Looking for amdgpu modules in /lib/modules/5.6.15-050615-generic/updates/dkms
Is nvidia loaded? no
Was nvidia unloaded? no
Is nvidia blacklisted? yes
Is intel loaded? no
Is radeon loaded? no
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? no
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? yes
Is nvidia kernel module available? no
Is amdgpu kernel module available? no
Vendor/Device Id: 10de:2191
BusID "PCI:1@0:0:0"
Is boot vga? no
Error: can't access /sys/bus/pci/devices/0000:01:00.0/driver
The device is not bound to any driver.
Vendor/Device Id: 1002:15d8
BusID "PCI:5@0:0:0"
Is boot vga? yes
Found "/dev/dri/card0", driven by "amdgpu"
output 0:
	card0-eDP-1
Number of connected outputs for /dev/dri/card0: 1
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "amdgpu"
Does it require offloading? no
last cards number = 2
Has amd? yes
Has intel? no
Has nvidia? yes
How many cards? 2
Has the system changed? No
Unsupported discrete card vendor: 10de
Nothing to do

crawfxrd, 2020-06-03:
This will tell you if the NVIDIA card is being removed from the bus on startup:

```
journalctl --no-hostname -b -u system76-power
```

This will tell you if it's currently on the bus:

```
system76-power graphics power
```

In integrated mode, dynamic power management is not configured. So the device won't go into a low power state like in hybrid mode.

---

> Unsupported discrete card vendor: 10de

The version of ubuntu-drivers-common you are using does not support render offload on AMD+NVIDIA systems.

Skrilltrax, 2020-06-04:
Thanks for taking a look. I read the Nvidia docs where they explained that USB-C is keeping the card awake.

Are you sure about the ubuntu-drivers-common? Because nvidia-smi is working fine on my machine while using the hybrid mode and Nvidia drivers also show the card is in lowest power state.

Is there any way to power off nvidia card fully while in integrated mode?

crawfxrd, 2020-06-04:
According to the log, gpu-manager did not configure X for render offload (`/usr/share/X11/xorg.conf.d/11-nvidia-offload.conf` not written). `nvidia-smi` should report no processes used. Applications attempting to use the dGPU for rendering should fail.

If configured for render offload, `nvidia-smi` will report 2 instances of `/usr/lib/xorg/Xorg` running: one for GDM, one for current session.

ubuntu-drivers-common=1:0.8.1pop1 must be used for render offload support on AMD+NVIDIA.

> Is there any way to power off nvidia card fully while in integrated mode?

```
system76-power graphics power off
```

(It shouldn't be possible, but ensure the NVIDIA drivers are not loaded first.)

Skrilltrax, 2020-06-04:
`nvidia-smi` did show 2 instances  `/usr/lib/xorg/Xorg` when running in the hybrid mode. I'm currently on another distro, will test and report ASAP.

Skrilltrax, 2020-06-04:
**Integrated Mode**

```
log_file: /var/log/gpu-manager.log
last_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
new_boot_file: /var/lib/ubuntu-drivers-common/last_gfx_boot
can't access /run/u-d-c-nvidia-was-loaded file
can't access /opt/amdgpu-pro/bin/amdgpu-pro-px
Looking for nvidia modules in /lib/modules/5.4.0-7634-generic/updates/dkms
Looking for amdgpu modules in /lib/modules/5.4.0-7634-generic/updates/dkms
Is nvidia loaded? no
Was nvidia unloaded? no
Is nvidia blacklisted? yes
Is intel loaded? no
Is radeon loaded? no
Is radeon blacklisted? no
Is amdgpu loaded? yes
Is amdgpu blacklisted? no
Is amdgpu versioned? no
Is amdgpu pro stack? no
Is nouveau loaded? no
Is nouveau blacklisted? yes
Is nvidia kernel module available? no
Is amdgpu kernel module available? no
Vendor/Device Id: 10de:2191
BusID "PCI:1@0:0:0"
Is boot vga? no
Error: can't access /sys/bus/pci/devices/0000:01:00.0/driver
The device is not bound to any driver.
Vendor/Device Id: 1002:15d8
BusID "PCI:5@0:0:0"
Is boot vga? yes
Found "/dev/dri/card0", driven by "amdgpu"
output 0:
	card0-eDP-1
Number of connected outputs for /dev/dri/card0: 1
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "amdgpu"
Skipping "/dev/dri/card0", driven by "amdgpu"
Does it require offloading? yes
last cards number = 2
Has amd? yes
Has intel? no
Has nvidia? yes
How many cards? 2
Has the system changed? No
System with free drivers detected
Nothing to do
```
```
journalctl --no-hostname -b -u system76-power
-- Logs begin at Fri 2020-06-05 00:28:38 IST, end at Fri 2020-06-05 05:58:29 IST. --
Jun 05 00:46:25 systemd[1]: Starting System76 Power Daemon...
Jun 05 00:46:25 system76-power[767]: [INFO] Starting daemon
Jun 05 00:46:25 system76-power[767]: [INFO] Connecting to dbus system bus
Jun 05 00:46:25 system76-power[767]: [INFO] Rescanning PCI bus
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.0: NVIDIA graphics
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.2: Function for 0000:01:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.0: Function for 0000:01:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.3: Function for 0000:01:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.1: Function for 0000:01:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.0: AMD graphics
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.3: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.1: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.6: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.4: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.2: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.0: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:05:00.7: Function for 0000:05:00.0
Jun 05 00:46:25 system76-power[767]: [INFO] Disabling NMI Watchdog (for kernel debugging only)
Jun 05 00:46:25 system76-power[767]: [INFO] Setting automatic graphics power
Jun 05 00:46:25 system76-power[767]: [INFO] Disabling graphics power
Jun 05 00:46:25 system76-power[767]: [INFO] xhci_hcd: Unbinding 0000:01:00.2
Jun 05 00:46:25 system76-power[767]: [INFO] snd_hda_intel: Unbinding 0000:01:00.1
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.2: Removing
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.0: Removing
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.3: Removing
Jun 05 00:46:25 system76-power[767]: [INFO] 0000:01:00.1: Removing
Jun 05 00:46:25 system76-power[767]: [INFO] Initializing with the balanced profile
Jun 05 00:46:25 system76-power[767]: [WARN] Failed to set initial profile: Errors found when setting profile:
Jun 05 00:46:25 system76-power[767]:     - failed to set scsi host profiles: failed to set link time power management policy med_power_with_dipm on host0: No such file or >
Jun 05 00:46:25 system76-power[767]: [INFO] Registering dbus name com.system76.PowerDaemon
Jun 05 00:46:25 system76-power[767]: [INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Jun 05 00:46:25 systemd[1]: Started System76 Power Daemon.
Jun 05 00:46:25 system76-power[767]: [ERROR] fan daemon: platform hwmon not found
Jun 05 00:46:25 system76-power[767]: [INFO] Handling dbus requests
Jun 05 00:46:35 system76-power[767]: [INFO] DBUS Received GetSwitchable method
Jun 05 00:46:35 system76-power[767]: [INFO] DBUS Received GetGraphics method
Jun 05 00:46:35 system76-power[767]: [INFO] DBUS Received GetProfile method
```

Skrilltrax, 2020-06-04:
This is with the latest `ubuntu-common-drivers`