pop-os/system76-power #170: daemon "Rejected send message"

tleydxdy, 2020-06-09:
**Distribution (run `cat /etc/os-release`):**
 Arch Linux, newest updates
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
local/firmware-manager-git 0.1.1.r0.g5b04786-1
local/system76-acpi-dkms 1.0.1-1
local/system76-dkms-git 110-1
local/system76-driver-git 20.04.5.r0.gf7a8f58-1
local/system76-firmware-daemon-git 1.0.10.r0.g2f50212-1
local/system76-io-dkms-git r26.ea5f61a-2
local/system76-power-git r244.224fc96-1
```

**Issue/Bug Description:**
```
$ system76-power profile performance
setting power profile to Performance
daemon returned an error message: "Rejected send message, 1 matched rules; type=\"method_call\", sender=\":1.375\" (uid=1000 pid=7735 comm=\"system76-power profile performance \") interface=\"com.system76.PowerDaemon\" member=\"Performance\" error name=\"(unset)\" requested_reply=\"0\" destination=\"com.system76.PowerDaemon\" (uid=0 pid=7710 comm=\"/usr/bin/system76-power daemon \")"
```
just `profile` shows
```
$ system76-power profile
Power Profile: ?
CPU: 10% - 100%, Turbo
Backlight intel_backlight: 750/15000 = 5%
```

**Steps to reproduce (if you know):**
This is after the newest release. I have rebooted my computer just to make sure
It worked fine in the past

**Expected behavior:**

**Other Notes:**
I'm using galp3

tleydxdy, 2020-06-09:
just changed the packages to the release versions
```
local/firmware-manager 0.1.1-1
local/system76-acpi-dkms 1.0.1-1
local/system76-dkms 1.0.8-1
local/system76-driver 20.04.8-1
local/system76-firmware-daemon 1.0.12-1
local/system76-io-dkms 1.0.1-1
local/system76-power 1.1.5-1
```
Still the same behavior. I've also tried to downgrade the package, but seems like the issue persisted.

tleydxdy, 2020-06-25:
seems to be fixed, looks like it was a bug elsewhere.

elleven11, 2020-08-06:
I am having the same problem on arch, any workarrounds?

elleven11, 2020-08-06:
Ty for reopening, i will link my information:
**Distribution (run `cat /etc/os-release`):**
```
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://www.archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux
``` 
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
system76-acpi-dkms 1.0.1-1
system76-dkms 1.0.9-1
system76-driver 20.04.11-1
system76-firmware-daemon 1.0.13-1
system76-io-dkms 1.0.1-1
system76-power 1.1.6-1
```
Installed from the AUR

**Issue/Bug Description:**
When i run ```system76-power graphics``` i get prompt with this error:
```
daemon returned an error message: "Rejected send message, 1 matched rules; type=\"method_call\", sender=\":1.1554\" (uid=1000 pid=13124 comm=\"system76-power graphics \") interface=\"com.system76.PowerDaemon\" member=\"GetGraphics\" error name=\"(unset)\" requested_reply=\"0\" destination=\"com.system76.PowerDaemon\" (uid=0 pid=676 comm=\"/usr/bin/system76-power daemon \")"
```
and if i try to set a graphics mode:
```
setting graphics to nvidia
daemon returned an error message: "Rejected send message, 1 matched rules; type=\"method_call\", sender=\":1.1601\" (uid=1000 pid=13304 comm=\"system76-power graphics nvidia \") interface=\"com.system76.PowerDaemon\" member=\"SetGraphics\" error name=\"(unset)\" requested_reply=\"0\" destination=\"com.system76.PowerDaemon\" (uid=0 pid=676 comm=\"/usr/bin/system76-power daemon \")"
```
the logs when the deamon starts:
```
[INFO] Starting daemon
[INFO] Connecting to dbus system bus
[INFO] Rescanning PCI bus
[INFO] 0000:01:00.0: NVIDIA graphics
[INFO] 0000:01:00.2: Function for 0000:01:00.0
[INFO] 0000:01:00.0: Function for 0000:01:00.0
[INFO] 0000:01:00.3: Function for 0000:01:00.0
[INFO] 0000:01:00.1: Function for 0000:01:00.0
[INFO] 0000:00:02.0: Intel graphics
[INFO] 0000:00:02.0: Function for 0000:00:02.0
[INFO] Disabling NMI Watchdog (for kernel debugging only)
[INFO] Setting automatic graphics power
[INFO] Enabling graphics power
[INFO] Initializing with the balanced profile
[INFO] Registering dbus name com.system76.PowerDaemon
[INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
[ERROR] fan daemon: platform hwmon not found
 [INFO] Handling dbus requests
```
and the deamon does not throw errors when i send it commands.

rahulaggarwal965, 2020-08-07:
@elleven11 Hey, I am also on arch and I got the graphics switching to work by using sudo with the graphics command. The reason why this works is you can see when dbus rejects the message that it is receiving a uuid=1000, while uuid should be equal to 0 (root). Let me know if this works for you

mmstick, 2020-08-07:
@rahulaggarwal965 Your user has to be part of the adm group

elleven11, 2020-08-10:
okay, after adding the user to the adm group it worked.

closed 2020-08-29 by mmstick