pop-os/system76-power #176: Mode Selector Wrong after Suspend/Restore on lemp9

ntilley905, 2020-07-28:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
system76-power:
  Installed: 1.1.6~1593714362~20.04~6c354b7
  Candidate: 1.1.6~1593714362~20.04~6c354b7
  Version table:
 *** 1.1.6~1593714362~20.04~6c354b7 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
On my lemp9, when resuming from a suspend (suspend by closing laptop lid, resume by opening lid), the power mode seems to default to Balanced, as indicated by CPU temp. Normally in high performance I see max temps around 98 C under load and 88 C in balanced. After performing the steps below I will see 88 C, but the mode selector will have the previous mode, such as High Performance, still shown as selected. 

**Steps to reproduce (if you know):**
1. Select High Performance power mode from system settings drop down.
2. Close laptop lid to suspend.
3. Open laptop lid to resume.
Max CPU temps now indicate Balanced mode is running, but mode selector shows High Performance.

**Expected behavior:**
System will restore to previously selected power mode *or* mode selector will show Balanced.

**Other Notes:**

ntilley905, 2020-08-26:
I recognize that this is definitely an edge case but also just noticed that the same issue occurs during an update of system76-power. I updated via apt and before the update was in high performance with 98 C, after the update completed it still showed High Performance selected but the max temp was 88 C. After selecting Balanced and reselecting High Performance, it resumed the 98 C max temp.

jacobgkau, 2020-08-26:
Seems related to https://github.com/pop-os/system76-power/issues/17 (for the profile being reset to Balanced) and https://github.com/pop-os/system76-power/issues/172 (for the top-right indicator not getting updated after the profile is reset.)