pop-os/system76-power #214: DELL XPS15Z (L511Z) HDMI output is not working

asansli, 2021-01-29:
**Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os

 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
system76-power:
  Installed: 1.1.11~1609439340~20.10~7bdc841
  Candidate: 1.1.11~1609439340~20.10~7bdc841
  Version table:
 *** 1.1.11~1609439340~20.10~7bdc841 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main amd64 Packages
        100 /var/lib/dpkg/status
```

**Issue/Bug Description:**
system76-power graphics always outputs "integrated" 

Here is output:
```
da@xps15z:~$ system76-power graphics
integrated
```

Tried to switch to hybrid or nvidia but does not work. Although installed nvidia realted drivers and kernel packages can not switch to nvidia or hybrid etc. so HDMI output does not works.

Here is syslog output for command
```
Jan 30 00:02:02 xps15z system76-power[934]: [INFO] DBUS Received SetGraphics("nvidia",) method
Jan 30 00:02:02 xps15z system76-power[934]: [INFO] Setting /etc/prime-discrete to on
Jan 30 00:02:02 xps15z system76-power[934]: [INFO] Creating /etc/modprobe.d/system76-power.conf
Jan 30 00:02:02 xps15z system76-power[934]: [INFO] Enabling nvidia-fallback.service
Jan 30 00:02:02 xps15z system76-power[21373]: Failed to enable unit: Unit file nvidia-fallback.service does not exist.
Jan 30 00:02:02 xps15z system76-power[934]: [WARN] systemctl: failed with exit code: 1 (not an error if service does not exist!)
Jan 30 00:02:02 xps15z system76-power[934]: [INFO] Updating initramfs
Jan 30 00:02:02 xps15z system76-power[21374]: update-initramfs: Generating /boot/initrd.img-5.8.0-7630-generic
Jan 30 00:02:13 xps15z system76-power[26167]: cryptsetup: WARNING: Resume target cryptswap uses a key file
```

**Steps to reproduce (if you know):**

**Expected behavior:**

**Other Notes:**
Tried reinstalling nvidia-dkms-460 nvidia-driver-460 but nothing changed.

```
sudo apt install --reinstall nvidia-dkms-460 nvidia-driver-460
```

crawfxrd, 2021-01-29:
Do you have any files in `/etc/modprobe.d/` or `/usr/lib/modprobe.d/` blacklisting the NVIDIA drivers after switching to hybrid/NVIDIA mode?

asansli, 2021-01-30:
Here is output 
<pre><font color="#4E9A06"><b>da@xps15z</b></font>:<font color="#3465A4"><b>~</b></font>$ ll /etc/modprobe.d/
total 68K
drwxr-xr-x   2 root root 4.0K Dec 30 04:47 <font color="#3465A4"><b>.</b></font>/
drwxr-xr-x 151 root root  12K Jan 30 17:03 <font color="#3465A4"><b>..</b></font>/
-rw-r--r--   1 root root 2.5K Jan 29 18:21 alsa-base.conf
-rw-r--r--   1 root root  154 Feb 16  2020 amd64-microcode-blacklist.conf
-rw-r--r--   1 root root  325 May 21  2020 blacklist-ath_pci.conf
-rw-r--r--   1 root root 1.5K May 21  2020 blacklist.conf
-rw-r--r--   1 root root  210 May 21  2020 blacklist-firewire.conf
-rw-r--r--   1 root root  677 May 21  2020 blacklist-framebuffer.conf
-rw-r--r--   1 root root  156 Jul 31  2015 blacklist-modem.conf
lrwxrwxrwx   1 root root   41 Dec 30 04:26 <font color="#06989A"><b>blacklist-oss.conf</b></font> -&gt; /lib/linux-sound-base/noOSS.modprobe.conf
-rw-r--r--   1 root root  583 May 21  2020 blacklist-rare-network.conf
-rw-r--r--   1 root root  127 Jun 15  2020 dkms.conf
-rw-r--r--   1 root root  154 Nov 12 02:24 intel-microcode-blacklist.conf
-rw-r--r--   1 root root  347 May 21  2020 iwlwifi.conf
-rw-r--r--   1 root root  379 Sep 28 18:44 mdadm.conf
-rw-r--r--   1 root root   73 Jan 30 00:02 system76-power.conf
<font color="#4E9A06"><b>da@xps15z</b></font>:<font color="#3465A4"><b>~</b></font>$ ll /usr/lib/modprobe.d/
total 36K
drwxr-xr-x   2 root root 4.0K Jan 26 10:44 <font color="#3465A4"><b>.</b></font>/
drwxr-xr-x 142 root root  12K Jan 30 17:03 <font color="#3465A4"><b>..</b></font>/
-rw-r--r--   1 root root  655 May 21  2020 aliases.conf
-rw-r--r--   1 root root 1.6K Dec 29 01:15 blacklist_linux_5.8.0-7630-generic.conf
-rw-r--r--   1 root root  390 Oct 19 19:45 fbdev-blacklist.conf
-rw-r--r--   1 root root  164 Jan 25 22:06 nvidia-graphics-drivers.conf
-rw-r--r--   1 root root  765 Oct 19 19:45 systemd.conf
</pre>