pop-os/system76-power #215: Error activating hybrid graphic mode

andirsun, 2021-01-31:
**Distribution (run `cat /etc/os-release`):**
 
 NAME="elementary OS"
VERSION="5.1.7 Hera"
ID=elementary
ID_LIKE=ubuntu
PRETTY_NAME="elementary OS 5.1.7 Hera"
LOGO=distributor-logo
VERSION_ID="5.1.7"
HOME_URL="https://elementary.io/"
SUPPORT_URL="https://elementary.io/support"
BUG_REPORT_URL="https://github.com/elementary/os/issues/new"
PRIVACY_POLICY_URL="https://elementary.io/privacy-policy"
VERSION_CODENAME=hera
UBUNTU_CODENAME=bionic

 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

system76-power:
  Installed: 1.0.3~1580489003~18.04~21a72e5~dev
  Candidate: 1.0.3~1580489003~18.04~21a72e5~dev
  Version table:
 *** 1.0.3~1580489003~18.04~21a72e5~dev 500
        500 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

**Issue/Bug Description:**
I installed the system76-power daemon in my elementary os from system76 ppa
```
sudo apt install system76-power -y
```
I just run the command
```
>anderson@elementary-os:~/projects/slinqer/backend-bot/src$ system76-power graphics switchable
>switchable

```
I noticed Im able to set my graphic mode in hibryd but after run the command
```
system76-power graphics hybrid
```
Im facing this error
```
error: Found argument 'hybrid' which wasn't expected, or isn't valid in this context

USAGE:
    system76-power graphics [SUBCOMMAND]

For more information try --help

```
**Expected behavior:**
Set the hybrid mode to my laptop

bflanagin, 2021-02-01:
@andirsun Thanks for reaching out, they hybrid graphics mode isn't available on all systems and all setups so before anything else can be done could you give us more hardware info? Most importantly CPU,GPU, GPU driver (FOSS,Proprietary) and if your device has a model name or number that would be useful too. 

There may be a simple fix, or hybrid mode might not be available on your system. Until we have the info it will be hard to diagnose further.

andirsun, 2021-02-01:
Sure! these are the specs : 
- Quad-Core Intel® Core™ i5-8250U CPU @ 1.60GHz
- Nvidia MX250 (Propietary driver 460.32.03)
- My laptop is a Xiaomi notebook pro

bflanagin, 2021-02-02:
Thanks for the info @andirsun but unfortunately system76-power is primarily for our own devices and though it works with many other machines we can't support every configuration out there. If `system76-power graphics -h` doesn't list "hybrid" as one of its sub commands then it can't be done with our utility. 

Regardless you will get more help from our community mattermost server found at https:chat.pop-os.org as our github issue tracker is more for feature requests and bugs.  Hopefully,someone there has come across the issue on the same hardware and has a fix for you. 

Closing ticket

closed 2021-02-02 by bflanagin