pop-os/system76-power #216: Fans start spinning for 2 seconds when idle

pflanzenandi, 2021-02-02:
Normal behaviour: The fans (CPU & GPU) only start when the threshold of > 60 °C is reached. They then run until a CPU temperature of approx. 35 °C is reached. After that, they switch off completely.

Behaviour since today: Even at an idle temperature of 30 °C, the fans start briefly for 2 seconds. They are then immediately stopped by the fan management. After 10 seconds the fans start again and so on.

Pop_Os 20.10
Dell xps 9570
Power Settings: Integrated Graphics & Balanced

pflanzenandi, 2021-02-02:
When I use the following script to disable SMBIOS control, the fans run continuously. 

dell-bios-fan-control 0

https://github.com/TomFreudenberg/dell-bios-fan-control

pflanzenandi, 2021-02-02:
A gnome extension triggered this behaviour. Can be closed.

https://extensions.gnome.org/extension/1180/freon/

bflanagin, 2021-02-02:
Thanks for the report and self diagnosing the issue @pflanzenandi. We'll make a note of it for the next time someone has the issue!

closed 2021-02-02 by bflanagin