pop-os/system76-power #220: Compute graphics mode galp5

curiousercreative, 2021-02-09:
**Distribution (run `cat /etc/os-release`):**
 ```
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os
```
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
```
$ apt policy system76-power
system76-power:
  Installed: 1.1.15~1612538425~20.10~c5ce307
  Candidate: 1.1.15~1612538425~20.10~c5ce307
  Version table:
 *** 1.1.15~1612538425~20.10~c5ce307 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main amd64 Packages
        100 /var/lib/dpkg/status

$ apt policy system76-cuda-latest
system76-cuda-latest:
  Installed: 11.2~20.10
  Candidate: 11.2~20.10
  Version table:
 *** 11.2~20.10 500
        500 http://apt.pop-os.org/proprietary groovy/main all Packages
        100 /var/lib/dpkg/status

$ apt policy nvidia-driver-460
nvidia-driver-460:
  Installed: 460.39-1pop0~1611882295~20.10~5279175
  Candidate: 460.39-1pop0~1611882295~20.10~5279175
  Version table:
 *** 460.39-1pop0~1611882295~20.10~5279175 1001
       1001 http://ppa.launchpad.net/system76/pop/ubuntu groovy/main amd64 Packages
        100 /var/lib/dpkg/status
     460.32.03-0ubuntu0.20.10.1 500
        500 http://us.archive.ubuntu.com/ubuntu groovy-security/restricted amd64 Packages
        500 http://us.archive.ubuntu.com/ubuntu groovy-updates/restricted amd64 Packages
```

**Issue/Bug Description:**
Compute graphics mode does not make dGPU available it seems? Several applications that work well under Nvidia or Hybrid graphics mode infinitely hang when launch (gpu-burn, luxmark, DaVinci Resolve, Blackmagic RAW Speed Test, nvidia-smi, Geekbench 5) and from command line don't respond to ctrl+c or ctrl+d.

**Steps to reproduce (if you know):**
1. With galp5 1650Ti, switch to compute graphics mode
2. Reboot
3. In terminal run `nvidia-smi`

**Expected behavior:**
GPU info should be output

**Other Notes:**
Output from /var/log/kern.log:
```
kernel: [    3.303232] nvidia: module license 'NVIDIA' taints kernel.
kernel: [    3.303233] Disabling lock debugging due to kernel taint
kernel: [    3.307598] pstore: Using crash dump compression: deflate
kernel: [    3.307607] pstore: Registered efi as persistent store backend
kernel: [    3.309742] mc: Linux media interface: v0.10
[... 161 lines cut ...]
kernel: [   18.573298]  ? do_filp_open+0x8c/0x130
kernel: [   18.573300]  ? __check_object_size+0x1c/0x20
kernel: [   18.573303]  ? strncpy_from_user+0x4c/0x150
kernel: [   18.573305]  ? do_sys_openat2+0xa1/0x150
kernel: [   18.573307]  ? __x64_sys_openat+0x56/0x90
kernel: [   18.573309]  ? do_syscall_64+0x49/0xc0
kernel: [   18.573311]  ? entry_SYSCALL_64_after_hwframe+0x44/0xa9
kernel: [   18.573313] Modules linked in: rfcomm msr snd_seq_dummy snd_hrtimer nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype br_netfilter ccm cmac algif_hash algif_skcipher af_alg xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_counter nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink bridge wireguard curve25519_x86_64 stp llc libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libblake2s blake2s_x86_64 ip6_udp_tunnel udp_tunnel libcurve25519_generic libchacha libblake2s_generic aufs overlay bnep snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda joydev snd_sof_xtensa_dsp snd_sof_intel_hda snd_sof snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi ledtrig_audio snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi
kernel: [   18.573335]  snd_seq_midi_event snd_rawmidi iwlmvm snd_seq x86_pkg_temp_thermal mac80211 snd_seq_device intel_powerclamp nvidia_uvm(OE) intel_rapl_msr libarc4 snd_timer coretemp mei_hdcp btusb uvcvideo btrtl videobuf2_vmalloc btbcm videobuf2_memops btintel kvm_intel videobuf2_v4l2 bluetooth kvm nls_iso8859_1 videobuf2_common iwlwifi input_leds videodev nvidia(POE) mc intel_cstate snd serio_raw ecdh_generic efi_pstore mei_me ee1004 ecc cfg80211 soundcore mei 8250_dw intel_pmc_mux typec hid_multitouch processor_thermal_device intel_rapl_common int340x_thermal_zone intel_soc_dts_iosf roles intel_scu_pltdrv intel_hid sparse_keymap mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs blake2b_generic dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) wmi hid_generic i915 rtsx_pci_sdmmc crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel i2c_algo_bit
kernel: [   18.573363]  drm_kms_helper crypto_simd cryptd glue_helper syscopyarea sysfillrect psmouse sysimgblt fb_sys_fops r8169 cec nvme rtsx_pci rc_core realtek i2c_i801 i2c_smbus drm intel_lpss_pci intel_lpss thunderbolt idma64 xhci_pci nvme_core virt_dma xhci_pci_renesas i2c_hid hid pinctrl_tigerlake video pinctrl_intel system76_acpi(OE)
kernel: [   18.573375] CR2: 0000000000001f90
kernel: [   18.573378] ---[ end trace bc21c478191de744 ]---
kernel: [   18.606545] RIP: 0010:_nv036052rm+0x20/0xd0 [nvidia]
kernel: [   18.606550] Code: 2e 0f 1f 84 00 00 00 00 00 90 41 56 41 55 41 54 53 48 83 ec 08 4c 8b 27 49 8d 9c 24 90 1f 00 00 4d 8d b4 24 e8 1f 00 00 66 90 <4c> 8b 2b 4d 85 ed 74 16 49 8b 7d 00 48 85 ff 74 05 e8 da 2b 30 00
kernel: [   18.606553] RSP: 0018:ffffb78b8343f9c0 EFLAGS: 00010286
kernel: [   18.606555] RAX: 0000000000000000 RBX: 0000000000001f90 RCX: 0000000000000010
kernel: [   18.606556] RDX: 0000000000000000 RSI: ffff8eb295de8168 RDI: ffff8eb28f7f6b28
kernel: [   18.606558] RBP: ffff8eb28f732e80 R08: 0000000000000000 R09: ffff8eb28f732d98
kernel: [   18.606559] R10: 0000000000000098 R11: 000000000000001f R12: 0000000000000000
kernel: [   18.606561] R13: ffff8eb28f7f4008 R14: 0000000000001fe8 R15: ffff8eb28f7f60e8
kernel: [   18.606563] FS:  00007f44ccb47740(0000) GS:ffff8eb31fa40000(0000) knlGS:0000000000000000
kernel: [   18.606564] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: [   18.606565] CR2: 0000000000001f90 CR3: 00000007f9d72004 CR4: 0000000000760ee0
kernel: [   18.606567] PKRU: 55555554
```

crawfxrd, 2021-02-09:
Should be resolved by system76/coreboot#51.

curiousercreative, 2021-02-09:
@crawfxrd thanks! Would you guess an OTA firmware update in the coming week or two or should I explore building and flashing myself?

jackpot51, 2021-02-09:
@curiousercreative since you have experience building firmware, you can try updating from the current master of firmware-open. It includes the fix.

We will release the update in the next few days.

jacobgkau, 2021-02-09:
> thanks! Would you guess an OTA firmware update in the coming week or two or should I explore building and flashing myself?

We have an OTA firmware update prepared that's in the process of being released now.

jackpot51, 2021-02-09:
@curiousercreative the firmware is released now, please take a look

closed 2021-02-09 by jackpot51

curiousercreative, 2021-02-09:
@jackpot51 magical! I actually haven't flashed any of the firmware except for the EC yet. I can confirm the firmware update resolves the "crash" and the GPU is working in compute power mode. I'm new to these GPU power modes and I thought compute would be my preference, but according to `nvidia-smi`, the GPU's power floor is 20W in compute mode while in NVIDIA and Hybrid mode it's 4W. Is that to be expected?
```
$ nvidia-smi
Tue Feb  9 16:35:18 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 165...  Off  | 00000000:23:00.0 Off |                  N/A |
| N/A   54C    P0    20W /  N/A |      0MiB /  3911MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

```

curiousercreative, 2021-02-09:
Here's the output of `nvidia-smi` in Hybird mode:
```
$ nvidia-smi
Tue Feb  9 16:39:45 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 165...  Off  | 00000000:23:00.0 Off |                  N/A |
| N/A   50C    P8     4W /  N/A |      5MiB /  3911MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2338      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+
```

curiousercreative, 2021-02-09:
and here's in Hybrid mode with a compute app idling:
```
$ nvidia-smi
Tue Feb  9 16:41:43 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.39       Driver Version: 460.39       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 165...  Off  | 00000000:23:00.0 Off |                  N/A |
| N/A   50C    P8     4W /  N/A |     90MiB /  3911MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2338      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A      4747      C   /opt/resolve/bin/resolve           83MiB |
+-----------------------------------------------------------------------------+
```

jackpot51, 2021-02-09:
Do not use nvidia-smi to measure power when the GPU is off. It forces the GPU to turn back on and this makes the instantaneous power reading inaccurate, especially for compute mode where it turns on and back off instantly.

If you want to see when the GPU powers off and on, you can add acpi.aml_debug_output=1 to your kernel parameters. `powertop` or `sensors` can be used to get accurate power draw when on battery power.

curiousercreative, 2021-02-10:
Ooo, hooray! Without setting the kernel flags, `sensors` output below. Total system power consumption should be current * voltage, about 7W with these readings, correct? This is in compute mode, so that dGPU mustn't be consuming that 20W I was worried about :)
```
BAT0-acpi-0
Adapter: ACPI interface
in0:          14.90 V  
curr1:       536.00 mA
```