pop-os/system76-power #221: Cannot use Integrated Intel graphics

kylebakerio, 2021-02-21:
**Distribution (run `cat /etc/os-release`):**
 20.10
 
 
**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**

**Issue/Bug Description:**
xrandr only shows 144hz.

I seem only be able to boot when using my nvidia gpu, which I was assuming is a result of that high refresh rate, but later looking it up, honestly that should still be just fine off the intel gpu, so I don't know what the cause is. For now I've been focused on trying to add a 60hz refresh to the screen, but haven't been able to--but I'm not even sure that would solve the problem.

My xrandr output for this monitor:

    DP-0 connected primary 1920x1080+2160+0 (normal left inverted right x axis y axis) 344mm x 194mm
    1920x1080    144.00*+

My current stuck point, at trying to add any other refresh rate than 144 hz...

    gtf 1920 1080 60
    # using that output, I construct the next line:
    xrandr --newmode  "1920x1080_60_gtf"  172.80  1920 2040 2248 2576  1080 1081 1084 1118  -HSync +Vsync 
    xrandr --addmode DP-0 1920x1080_60_gtf
    # get error: 
    X Error of failed request:  BadMatch (invalid parameter attributes)

[Looking further](https://www.reddit.com/r/archlinux/comments/apoaud/cant_force_video_mode_xrandr_invalid_parameter/egggtv9?utm_source=share&utm_medium=web2x&context=3), it seems like perhaps the problem is I cannot change this while running the nvidia driver itself, but perhaps would run when on the intel card? Not sure what that means for me.

**Steps to reproduce (if you know):**
Fresh install on a Dell G5 5590 with an RTX 2070mq with a 144hz screen. Attempt to switch to 'integrated' or 'computer'. Boot fails.

**Expected behavior:**
Boot should work.

**Other Notes:**
I've documented my attempts to resolve this extensively on reddit:
https://www.reddit.com/r/pop_os/comments/lo8nuj/cant_use_igpu_only_nvidia_hybrid_work_compute_and/
https://www.reddit.com/r/pop_os/comments/lo7be3/cant_switch_to_intel_on_2010_dell_g5_5590_fresh/
https://www.reddit.com/r/linuxquestions/comments/lp4wi0/how_to_add_60hz_refresh_to_monitor_that_is_only/
https://www.reddit.com/r/pop_os/comments/lp0gos/how_to_switch_gpu_mode_from_recovery/

kylebakerio, 2021-02-21:
just tried downloading the 'intel' iso instead. While I did notice one difference, I think--the text looked 'cleaner' that scrolls past that reads `[OK]` for a bunch of startup tasks than it does on the other iso for the same screen--when I installed it to a USB stick (persistent) as a test, it still launched in `nvidia` mode, both in live mode and when booted into to the persistent install. I then tried switching that install to `integrated`, and it also failed to show any display.

kylebakerio, 2021-02-22:
I was unable to boot in to an mxlinux live boot, the fail looked similar. I was able to boot in to manjaro live boot, both with proprietary and open source. Both booted into nvidia mode, however, and also only showed 144hz option.

I do however see [here](https://wiki.archlinux.org/index.php/Dell_G5_5590-9340) on the arch wiki that the intel and nvidia gpus both work out of box for this user, virtually identical hardware.

bflanagin, 2021-02-22:
Its a strange issue @kylebakerio, we have a laptop that uses the same nvidia gpu that doesn't exhibit the issue. Given that others haven't had the issue and you have been able to replicate the issue on other distros it may be something specific with your hardware. We'll keep this issue open so that others can add reports, but there is little we can do without the specific hardware.

kylebakerio, 2021-02-22:
Thanks for the prompt response!

I'll keep updating here as I dig into it as well.

I do want to suggest, though, that there be some better built-in recovery for botched graphics mode switches. I love pop, but almost all of my bad experiences with pop (including on other hardware) revolve in some way around switching graphics modes. Don't get me wrong, it's also one of the _best_ features of pop, but bettery self-recovery from a bad switch would go a long way.

Windows (for which I have no love, don't get me wrong) will attempt when multiple boot attempts have failed to get into the OS in a row. Perhaps pop could have something similar, where if a switch is attempted, AND a failure to boot seems to occur, it falls back and/or attempts some kind of 'safe mode'?

There is of course already a 'recovery partition' option if you install to the full HD, and booting with a USB stick and using chroot is another option, but then one must find the right web page, and copy the right scripts over, and hope those work. Following those scripts the first time both did not work for me as I first dealt with this issue, and then as I attempted to troubleshoot, ended up killing my built-in windows install, which I ended up nuking and deciding I would just re-install later 'the pop way' at some point, but... I plan to edit that help page to make it clearer myself in the future, but I digress).

All of that should perhaps go into another issue, mind you, just figured I'd mention it.

kylebakerio, 2021-02-22:
@bflanagin what laptop model do you have? I'm wondering if it's possible somehow that the internal screen is directly connected to the nvidia gpu itself, and that that is why I can't switch? Is that possible / do you know how I would check that?

crawfxrd, 2021-02-22:
`lspci` will list the Intel device (00:02.0) if the eDP is connected to it.

kylebakerio, 2021-02-22:
Other people that seem to experience similar issues:
[switched to hybrid, can't boot, black screen](https://www.reddit.com/r/pop_os/comments/ejd0ry/refuses_to_start_after_i_tried_changing_the/?utm_content=title&utm_medium=post_embed&utm_name=a37216315d104c268274c8ba2cc9171b&utm_source=embedly&utm_term=ejd0ry)
^ this case is obviously not identical to mine, but it does speak to the need for bettery recovery handling

[switched to intel, can't boot](https://pop-planet.info/forums/threads/unable-to-boot-properly-after-switching-graphics-mode.832/)
^this is a similar issue to me, but they were able to get info when they removed the quiet/splash screen, while for me I was not. They did switch to nomodeset, and then some other stuff where they didn't indicate their source or reasoning, which I haven't tried yet, but could perhaps help me? When I was first reading this thread I didn't know what modeset was (that's the non-proprietary intel drivers, and he set to intel drivers, which usually work worse, right? But maybe in this case that's the source of the issue?).

[another user getting stuck with graphics boot problems](https://www.reddit.com/r/pop_os/comments/gcs6m5/possible_to_change_to_discrete_graphics_through/)

@crawfxrd 
lspci does indeed show my intel card:
00:02.0 Display controller: Intel Corporation UHD Graphics 630 (Mobile)

kylebakerio, 2021-02-22:
@bflanagin btw, is there any advice you'd have on confirming a hardware issue? I highly doubt it, it's basically a new laptop, but it is under warranty, so if there's a hardware problem I'd love to figure out what logs I should track down that might lead to me replacing it as needed.

kylebakerio, 2021-02-24:
need to look into it more, but this looks like a related issue / user experience for almost same laptop in manjaro, may be leveraging same code, (not sure exactly how manjaro and pop handle their nvidia support but I'm assuming it is similar based on my boot experience): https://www.reddit.com/r/ManjaroLinux/comments/d7a5kr/would_love_to_use_manjaro_but_im_having_trouble/

kylebakerio, 2021-02-24:
someone else who may have had a related issue on an identical laptop, perhaps in windows: https://www.dell.com/community/Inspiron/G5-5590-Appalling-Battery-Life-Intel-UHD-630-Driver-fail/td-p/7351595

kylebakerio, 2021-02-24:
[This review](https://www.notebookcheck.net/Dell-G5-15-5590-i7-8750H-RTX-2060-SSD-FHD-Laptop-Review.422281.0.html) mentions that,

> Alongside the RTX 2060, the laptop also comes with Intel's UHD Graphics 630 integrated GPU. Both cores provide a graphics-switching solution (Optimus). Intel's GPU is used for everyday tasks and when the device is on battery power. Nvidia's graphics unit takes care of demanding tasks such as computer games.

Which indicates (if correct) that the laptop should have hardware that is capable of this task, at least.

kylebakerio, 2021-02-24:
[This user](https://www.reddit.com/r/hackintosh/comments/ezhx8c/dell_g5_15_5590_catalina_complete/) claims to have installed it as a Hackintosh and says the UHD-630 is working.

kylebakerio, 2021-02-24:
@bflanagin, confirmed that it is not a hardware issue.

I reinstalled windows and confirmed that it has no problem running on the iGPU exclusively. First of all, the nvidia was no-op on a clean install. This is also confirmed by the inability to recognize external monitors, which was 'fixed' as soon as I installed the nvidia drivers for the laptop in windows.

Then, I went into the device manager and disabled the nvidia card. Immediately the external monitors went off, but I was able to use the built in display just fine.

Some possible helpful output, let me know what else I can dig into:
```
$ sudo lsmod | grep i915
i915                 2273280  4
i2c_algo_bit           16384  1 i915
drm_kms_helper        225280  2 nvidia_drm,i915
cec                    53248  2 drm_kms_helper,i915
drm                   565248  10 drm_kms_helper,nvidia_drm,i915
video                  49152  3 dell_wmi,dell_laptop,i915
```

```
$ lspci | grep -i display
00:02.0 Display controller: Intel Corporation UHD Graphics 630 (Mobile)
```

```
$ lspci | grep -i vga
01:00.0 VGA compatible controller: NVIDIA Corporation TU106BM [GeForce RTX 2070 Mobile / Max-Q] (rev a1)

```

kylebakerio, 2021-02-25:
I investigated a bit more in windows. I can confirm that the screen only reports 144hz as a possible refresh rate. However, when switch to the intel iGPU, it drops it to 64hz. This confuses windows, which reports 1hz in its dialogue, but can be seen elsewhere and confirmed with testufo.com . In the nvidia control app in windows, when the nvidia card is enabled, only 144hz is available, but you can set a custom refresh rate. I set it to 64 this way, and it worked, as I confirmed by toggling the custom refresh vs the 144hz refresh on and off while on testufo.com. The intel graphics command utility does not give me the ability to set, or adjust the refresh rate, even if it is the only video card enabled, so I don't see any way to attempt 144hz through the intel graphics.

**Is it possible that the intel card won't handle 144hz, but that pop boots up and sees that the screen only offers a 144hz mode, and that this is the source of my problem?**

I have been trying to manually set a 60 hz mode, but have not been able to figure out how to do that in pop, as I've detailed above. How could I test this theory? @bflanagin, @crawfxrd

kylebakerio, 2021-02-26:
Pretty sure [this guy](https://www.reddit.com/r/ManjaroLinux/comments/d7a5kr/would_love_to_use_manjaro_but_im_having_trouble/) was experiencing exactly what I was experiencing:

The solution link listed there is dead, but this seems to be the same content posted elsewhere: https://vfbsilva.medium.com/howto-set-up-prime-with-nvidia-proprietary-driver-c647e3597447

kylebakerio, 2021-02-27:
Here's someone with probably the same laptop who has a different distribution who triggered the intel video card on on accident and goes through the process of trying to get their nvidia card back on: https://forum.kde.org/viewtopic.php?f=309&t=166457

Seems like some relevant information in there.

I think there's enough evidence here to indicate this is possible. I just need to know how pop's graphics card selector works. I am hopeful that I just need to find the right config file and add a 64hz profile there. I'd essentially been trying this before, but perhaps was not able to add the value to the correct place?

kylebakerio, 2021-02-27:
TL;DR,  On a clean install, if I switch to `integrated`, I become unable to boot, and using `chroot` from recovery to call `sudo system76-power graphics nvidia` does not fix the problem--I continue to get a black screen with no backlight, even though from the logs it looks like the machine boots otherwise successfully?

@bflanagin, @crawfxrd ,

I have tried to do a clean install onto a usb drive to experiment with, because in my experience switching to 'integrated' breaks my install (have experienced multiple times on this machine). using `chroot` to go change back to `nvidia` from within recovery does not repair it.

This just happened again on my usb install. switching to `integrated` threw both the usb and my harddrive (nvme) install into `hybrid` (I want to assume that that's just me having bad memory, but I'm feeling like I've seen this before). 

This particular time I had also added an experimental xorg.conf file for the intel card. I added a file `/usr/share/X11/xorg.conf.d/20-intel.conf`with these contents:

```bash
Section "Monitor"
    Identifier      "DP-0"
    Modeline        "main60"  172.80  1920 2040 2248 2576  1080 1081 1084 1118  -HSync +Vsync

    # 1920x1080 @ 64.00 Hz (GTF) hsync: 71.68 kHz; pclk: 185.79 MHz
    Modeline        "main64"  185.79  1920 2048 2256 2592  1080 1081 1084 1120  -HSync +Vsync

    Option          "PreferredMode" "main64"

    Option          "UseEDIDFreqs" "Off"
    # this is an experiment, haven't tried it yet, may work^
   

EndSection
```

deleting that file, switching to nvidia, nothing makes it bootable again. I'm not sure what to look in the logs for, but one thing I have noticed when I compare my first boot's logs with my most recent boot's logs, is this difference:

first boot, where it eventually succeeded in giving me a screen:
```
root@pop-os:/# journalctl --boot=386ed264a6b84782af1037ff73ab52ff | grep vga
Feb 27 17:41:28 pop-os kernel: pci 0000:01:00.0: vgaarb: setting as boot VGA device
Feb 27 17:41:28 pop-os kernel: pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
Feb 27 17:41:28 pop-os kernel: pci 0000:01:00.0: vgaarb: bridge control possible
Feb 27 17:41:28 pop-os kernel: vgaarb: loaded
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: vgaarb: deactivate vga console
Feb 27 17:41:28 pop-os kernel: vga_switcheroo: enabled
Feb 27 17:44:14 pop-os kernel: snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
```

most recent boot, where it fails to give me a screen:
```
root@pop-os:/# journalctl --boot=f09b79531f864727a37f4461a4db9dfb | grep vga
Feb 27 19:09:27 pop-os kernel: pci 0000:01:00.0: vgaarb: setting as boot VGA device
Feb 27 19:09:27 pop-os kernel: pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
Feb 27 19:09:27 pop-os kernel: pci 0000:01:00.0: vgaarb: bridge control possible
Feb 27 19:09:27 pop-os kernel: vgaarb: loaded
Feb 27 19:09:29 pop-os kernel: snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
```

as you can see, the most recent boot lacks these lines:

```
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: vgaarb: deactivate vga console
Feb 27 17:41:28 pop-os kernel: vga_switcheroo: enabled
```

to be clear, it won't boot in spite of `root@pop-os:/# sudo system76-power graphics` returning `nvidia`, which is a working state on a fresh install.

Here is the tail of my logs from my most recent failed boot, which mostly looks quite normal and functional to me:

```
Feb 27 19:09:41 pop-os dbus-daemon[756]: [system] Activating via systemd: service name='org.freedesktop.PackageKit' unit='packagekit.service' requested by ':1.45' (uid=110 pid=1144 comm="/>
Feb 27 19:09:41 pop-os systemd[1]: Starting PackageKit Daemon...
Feb 27 19:09:41 pop-os PackageKit[1197]: daemon start
Feb 27 19:09:41 pop-os dbus-daemon[756]: [system] Successfully activated service 'org.freedesktop.GeoClue2'
Feb 27 19:09:41 pop-os systemd[1]: Started Location Lookup Service.
Feb 27 19:09:41 pop-os /usr/libexec/gdm-x-session[1104]: dbus-daemon[1104]: [session uid=110 pid=1104] Activating service name='org.gnome.Shell.Notifications' requested by ':1.5' (uid=110 >
Feb 27 19:09:41 pop-os dbus-daemon[756]: [system] Successfully activated service 'org.freedesktop.PackageKit'
[... 22 lines cut ...]
Feb 27 19:09:42 pop-os systemd[1]: Started Manage, Install and Generate Color Profiles.
Feb 27 19:09:42 pop-os /usr/libexec/gdm-x-session[1104]: dbus-daemon[1104]: [session uid=110 pid=1104] Successfully activated service 'org.gnome.Shell.Screencast'
Feb 27 19:09:42 pop-os systemd[1]: dmesg.service: Succeeded.
Feb 27 19:09:43 pop-os systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Feb 27 19:09:57 pop-os systemd[960]: tracker-extract.service: Succeeded.
Feb 27 19:10:08 pop-os tracker-store[1082]: OK
Feb 27 19:10:08 pop-os systemd[960]: tracker-store.service: Succeeded.
Feb 27 19:10:11 pop-os systemd[1]: systemd-localed.service: Succeeded.
Feb 27 19:10:11 pop-os systemd[1]: systemd-hostnamed.service: Succeeded.
Feb 27 19:10:41 pop-os geoclue[1196]: Service not used for 60 seconds. Shutting down..
Feb 27 19:10:41 pop-os systemd[1]: geoclue.service: Succeeded.
Feb 27 19:10:43 pop-os systemd-logind[782]: Power key pressed.
Feb 27 19:11:37 pop-os systemd-logind[782]: Power key pressed.
Feb 27 19:11:43 pop-os kernel: sysrq: This sysrq operation is disabled.
Feb 27 19:11:44 pop-os kernel: sysrq: This sysrq operation is disabled.
[... cut ...]
Feb 27 19:11:45 pop-os kernel: Emergency Sync complete
Feb 27 19:11:46 pop-os kernel: sysrq: Emergency Remount R/O

```

(you can see me pressing the power button followed by REISUB because I see nothing but a black screen, and no TTY works)

Because I'm unable to recover from a failed attempt to switch to integrated, I'm really afraid to test any changes on my primary install, which I'm using for work. Once I know I can recovery safely, I can start experimenting with my normal install on trying to get the intel to boot.

kylebakerio, 2021-02-28:
Just want to add that if I include that `20-intel.conf` file, that I at least get a backlight for a while--just a black screen, but at least the screen lights up, then, after a while, goes black again. When I remove that file, I don't get a backlight screen. In either case, I cannot switch to any TTY.

Please, guys, any guidance on logs to look for, basic troubleshooting steps here...? I'm not deeply familiar with linux, but I've been a casual user for 15 years and I'm a web dev. I can work through stuff, I just need a little guidance. I've googled the shit out of this issue. 

Just shooting blindly around to look at more logs, here's something I noticed when comparing my latest failed boot to my first (functioning, nvidia auto selected) boot:

first boot:
```bash
root@pop-os:/# journalctl --boot=386ed264a6b84782af1037ff73ab52ff | grep nouveau
Feb 27 17:41:28 pop-os kernel: nouveau: detected PR support, will not use DSM
Feb 27 17:41:28 pop-os kernel: fb0: switching to nouveaufb from EFI VGA
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: vgaarb: deactivate vga console
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: NVIDIA TU106 (166000a1)
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: bios: version 90.06.2d.00.bc
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: fb: 8192 MiB GDDR6
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: VRAM: 8192 MiB
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: GART: 536870912 MiB
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: BIT table 'A' not found
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: BIT table 'L' not found
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: TMDS table version 2.0
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB version 4.1
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB outp 00: 02800f66 04610020
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB outp 01: 02011f52 00020010
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB outp 02: 04022f76 04600010
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB outp 03: 04022f72 00020010
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB conn 00: 00020047
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB conn 01: 00010161
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: DCB conn 02: 01000248
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: MM: using COPY for buffer copies
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: unknown connector type 48
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: allocated 1920x1080 fb: 0x200000, bo 0000000054f654db
Feb 27 17:41:28 pop-os kernel: fbcon: nouveaudrmfb (fb0) is primary device
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: fb0: nouveaudrmfb frame buffer device
Feb 27 17:41:28 pop-os kernel: [drm] Initialized nouveau 1.3.1 20120801 for 0000:01:00.0 on minor 1
Feb 27 17:41:28 pop-os kernel: nouveau 0000:01:00.0: DRM: Disabling PCI power management to avoid bug
Feb 27 17:44:14 pop-os kernel: snd_hda_intel 0000:01:00.1: bound 0000:01:00.0 (ops nv50_audio_component_bind_ops [nouveau])
Feb 27 17:44:22 pop-os /usr/libexec/gdm-x-session[1039]: (II) modeset(0): [DRI2]   DRI driver: nouveau
Feb 27 17:44:22 pop-os /usr/libexec/gdm-x-session[1039]: (II) modeset(0): [DRI2]   VDPAU driver: nouveau
Feb 27 17:44:22 pop-os /usr/libexec/gdm-x-session[1039]: (II) AIGLX: Loaded and initialized nouveau
Feb 27 17:45:12 pop-os /usr/libexec/gdm-x-session[1719]: (II) modeset(0): [DRI2]   DRI driver: nouveau
Feb 27 17:45:12 pop-os /usr/libexec/gdm-x-session[1719]: (II) modeset(0): [DRI2]   VDPAU driver: nouveau
Feb 27 17:45:12 pop-os /usr/libexec/gdm-x-session[1719]: (II) AIGLX: Loaded and initialized nouveau
```

failed boot, even though I've selected `nvidia` via chroot:
```
root@pop-os:/# journalctl --boot=6e05a3b6e8234c93bd5320f89f0117ef | grep nouveau
**nothing**
```

For some reason, the switching back to nvidia fails this way. :/

I also have `/home/kb/.local/share/xorg/Xorg.1.log` and `/home/kb/.local/share/xorg/Xorg.1.log.old` files in my usb install that I can share from if needed. Some interesting info, but I don't want to spam anymore of my random logs in the dark, let me know what's helpful and I'll share it...

kylebakerio, 2021-02-28:
I can now see that `dmesg` run in the `chroot` terminal leaks through from the install I'm running it from, so I guess there are just limitations to `chroot` that aren't discussed in [this support article](https://support.system76.com/articles/pop-recovery/). If that's the case, I just don't know how to switch the graphics mode for an install that I cannot seem to boot in to.

kylebakerio, 2021-02-28:
I just tried `sudo apt install --reinstall system76-power` on the usb install, but it did not change anything. I also tried a shot in the dark change to my `20-intel.conf` to look like this: 

```
Section "Monitor"
    Identifier      "DP-0"
    Modeline        "main60"  172.80  1920 2040 2248 2576  1080 1081 1084 1118  -HSync +Vsync

    # 1920x1080 @ 64.00 Hz (GTF) hsync: 71.68 kHz; pclk: 185.79 MHz
    Modeline        "main64"  185.79  1920 2048 2256 2592  1080 1081 1084 1120  -HSync +Vsync

    Option          "PreferredMode" "main64"

    Option          "UseEDIDFreqs" "Off"
    # this is an experiment, haven't tried it yet, may work^
   

EndSection

Section "Device"
	Identifier  "Intel Graphics"
	Driver      "Intel"
	Option      "AccelMethod"  "sna"
	Option      "TearFree"	"True"
	Option      "Tiling"	"True"
	Option      "SwapbuffersWait" "True"
	#Option      "AccelMethod"  "uxa"
EndSection
```

But I assumed it wouldn't even be used after the re-install. 

But as I can see in chroot again, `journalctl --boot=083859d845074d79aa0fc87d0ee51357 | grep nouveau` still returns nothing for the most recent boot.... however, before the reinstall, the most recent boot's logs had no mention of `nvidia`, while the most recent boot _does_, so that's maybe worth noting?

```bash
root@pop-os:/# journalctl --list-boots
-10 386ed264a6b84782af1037ff73ab52ff Sat 2021-02-27 17:41:28 CET—Sat 2021-02-27 17:51:54 CET
 -9 3825abe2c5064164925dde6d61046588 Sat 2021-02-27 17:52:21 CET—Sat 2021-02-27 17:58:52 CET
 -8 f5dce2d731f54aceb0ac2541c63ebb9b Sat 2021-02-27 17:59:23 CET—Sat 2021-02-27 18:04:22 CET
 -7 084569ce6b2b4a98ba034111a5bf95d8 Sat 2021-02-27 18:14:39 CET—Sat 2021-02-27 18:15:59 CET
 -6 4b1efb30e29f4202b69d5b081f958088 Sat 2021-02-27 18:16:24 CET—Sat 2021-02-27 18:20:07 CET
 -5 9e3f96e781014246ad27af94f4a2360e Sat 2021-02-27 18:41:29 CET—Sat 2021-02-27 18:44:54 CET
 -4 248b3727b9594efb83398e52b922ea94 Sat 2021-02-27 18:58:16 CET—Sat 2021-02-27 18:59:33 CET
 -3 f09b79531f864727a37f4461a4db9dfb Sat 2021-02-27 19:09:27 CET—Sat 2021-02-27 19:11:46 CET
 -2 5a3c2b45114a4e96b77bec9fd2b88cfb Sun 2021-02-28 13:55:56 CET—Sun 2021-02-28 13:58:27 CET
 -1 6e05a3b6e8234c93bd5320f89f0117ef Sun 2021-02-28 14:07:40 CET—Sun 2021-02-28 14:07:40 CET
  0 083859d845074d79aa0fc87d0ee51357 Sun 2021-02-28 15:31:18 CET—Sun 2021-02-28 15:34:44 CET
root@pop-os:/# journalctl --boot=6e05a3b6e8234c93bd5320f89f0117ef | grep nvidia
root@pop-os:/# journalctl --boot=083859d845074d79aa0fc87d0ee51357 | grep nvidia
Feb 28 15:32:35 pop-os audit[680]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=680 comm="apparmor_parser"
Feb 28 15:32:35 pop-os audit[680]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=680 comm="apparmor_parser"
Feb 28 15:32:35 pop-os kernel: audit: type=1400 audit(1614522755.000:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=680 comm="apparmor_parser"
Feb 28 15:32:35 pop-os kernel: audit: type=1400 audit(1614522755.000:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=680 comm="apparmor_parser"
root@pop-os:/# 
```

I do note that even after the reinstall of `system76-power`, it kept its `hybrid` graphics state, so... whatever the necessary config file(s) is/are, I guess they are left in place. I would purge, but it wants to remove basically the whole pop desktop if I do so, so I am holding off on doing that for now. :/

(Once again, if I set to `nvidia` from the chroot instance, I see both installs switch... both were hybrid, now both are nvidia after typing the command into the chroot terminal)

kylebakerio, 2021-02-28:
I have found [this issue](https://github.com/pop-os/pop/issues/194), I'm not sure if related, and the discussion is just a bit above my head, but it is the same symptom description.

kylebakerio, 2021-02-28:
I noticed in my most recent logs this line:

`Feb 28 15:46:01 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes`

I have a pretty tenuous grasp of how all of this works, but that kind of looks to me like the intel driver reporting that it doesn't see any valid screen sizes? Googling that, I found [this old issue](https://bbs.archlinux.org/viewtopic.php?id=230541), which describes a similar scenario--trouble switching to intel, has a high refresh rate screen. They ended up submitting a bug to intel, and it was patched. In their case, [the error](https://bbs.archlinux.org/viewtopic.php?id=230541&p=2) was 

> The issue discussed in this thread was finally tracked down to eDP (Embedded DisplayPort) specs being incorrectly set, in particular DPCD (DisplayPort Configuration Data) registers with incorrect values, in violation to the eDP spec. I tried to find the eDP spec to download, but could not find it anywhere. Without it, I cannot tell what does those values of the DPCD spec table stands for.

> Anyway, eDP protocol implies in change of data between the sink (display) and the host (PC). The first thing that happens when you connect a display through the displayPort connector is a hot-plug detection. The sink requests an interrupt. The host replies by reading DPCD regs and EDID to find out the resolution and refresh rate (among other things) of the monitor before it can start driving it with video data. With that information, the host will calculate the eDP data rate of up to 4 lanes (4 x 1.62Gbps) and will configure the interface.

> My monitor is a 120Hz, full HD laptop-embedded LCD screen. Apparently, the chip driving this monitor is not informing correct values of DPCD to the intel driver (i915) and thus the black screen perceived at boot time. The patch from Jani Nikula, from intel, bypasses some of the eDP spec checks because of this buggy behavior.

kylebakerio, 2021-02-28:
grepping through all journalctl for nvidia shows these two lines showing up yesterday, and may be related to why pop fails to boot into nvidia?

```bash
root@pop-os:/# journalctl | grep nvidia
...
Feb 27 17:57:33 pop-os system76-power[841]: [INFO] Disabling nvidia-fallback.service
Feb 27 17:57:33 pop-os system76-power[2250]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
...
```

in context, showing before and two boots after those lines:

```bash
Feb 27 17:44:05 pop-os audit[489]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=489 comm="apparmor_parser"
Feb 27 17:44:05 pop-os kernel: audit: type=1400 audit(1614444245.508:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=489 comm="apparmor_parser"
Feb 27 17:44:05 pop-os audit[489]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=489 comm="apparmor_parser"
Feb 27 17:44:05 pop-os kernel: audit: type=1400 audit(1614444245.512:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=489 comm="apparmor_parser"
Feb 27 17:55:49 pop-os kernel: audit: type=1400 audit(1614444948.180:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=515 comm="apparmor_parser"
Feb 27 17:55:49 pop-os kernel: audit: type=1400 audit(1614444948.180:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=515 comm="apparmor_parser"
Feb 27 17:55:48 pop-os audit[515]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=515 comm="apparmor_parser"
Feb 27 17:55:48 pop-os audit[515]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=515 comm="apparmor_parser"
Feb 27 17:57:33 pop-os system76-power[841]: [INFO] Disabling nvidia-fallback.service
Feb 27 17:57:33 pop-os system76-power[2250]: Failed to disable unit: Unit file nvidia-fallback.service does not exist.
Feb 27 18:01:29 pop-os kernel: audit: type=1400 audit(1614445288.340:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=515 comm="apparmor_parser"
Feb 27 18:01:29 pop-os kernel: audit: type=1400 audit(1614445288.340:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=515 comm="apparmor_parser"
Feb 27 18:01:28 pop-os audit[515]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=515 comm="apparmor_parser"
Feb 27 18:01:28 pop-os audit[515]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=515 comm="apparmor_parser"
Feb 27 18:14:41 pop-os audit[683]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=683 comm="apparmor_parser"
Feb 27 18:14:41 pop-os audit[683]: AVC apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=683 comm="apparmor_parser"
Feb 27 18:14:41 pop-os kernel: audit: type=1400 audit(1614446081.496:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=683 comm="apparmor_parser"
Feb 27 18:14:41 pop-os kernel: audit: type=1400 audit(1614446081.496:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=68
```

I think the relevant logs to fix this and understand it are all here, I just don't know how to interpret them myself.

kylebakerio, 2021-02-28:
non-working usb install that is broken from being put into an 'integrated-only' mode (e.g., `system76-power graphics integrated`):
```bash
root@pop-os:/# journalctl | grep drm
Feb 28 15:46:01 pop-os kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Feb 28 15:46:01 pop-os kernel: i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
Feb 28 15:46:01 pop-os kernel: [drm] Initialized i915 1.6.0 20200515 for 0000:00:02.0 on minor 0
Feb 28 15:46:01 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Feb 28 15:46:01 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Feb 28 15:46:01 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```
(This is, again, in spite of using `chroot` to attempt to go back in and manually set `system76-power graphics nvidia`, and with  `system76-power graphics` returning `nvidia`)

Working install that I've never booted into an 'integrated-only' mode:
```
kyle@pop-os:~$ journalctl | grep drm
Feb 28 15:47:30 pop-os kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Feb 28 15:47:30 pop-os kernel: i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
Feb 28 15:47:30 pop-os kernel: [drm] Initialized i915 1.6.0 20200515 for 0000:00:02.0 on minor 0
Feb 28 15:47:30 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Feb 28 15:47:30 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Feb 28 15:47:30 pop-os kernel: i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Feb 28 15:47:30 pop-os systemd[1]: Condition check resulted in Load Kernel Module drm being skipped.
Feb 28 15:47:30 pop-os kernel: [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
Feb 28 15:47:31 pop-os kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Feb 28 15:47:33 pop-os kernel: [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 1
Feb 28 15:47:34 pop-os /usr/libexec/gdm-x-session[1038]: (II) xfree86: Adding drm device (/dev/dri/card1)
Feb 28 15:47:34 pop-os /usr/libexec/gdm-x-session[1038]: (II) xfree86: Adding drm device (/dev/dri/card0)
Feb 28 15:48:01 pop-os /usr/libexec/gdm-x-session[2170]: (II) xfree86: Adding drm device (/dev/dri/card1)
Feb 28 15:48:01 pop-os /usr/libexec/gdm-x-session[2170]: (II) xfree86: Adding drm device (/dev/dri/card0)
```

crawfxrd, 2021-03-03:
In integrated mode, both the open source and proprietary drivers are blacklisted. You should _not_ be seeing references to nouveau if it is configured for integrated graphics.

```
grep -r 'nouveau\|nvidia' /etc/modprove.d
grep -r 'nouveau\|nvidia' /usr/lib/modprove.d
```

By any chance, is i915 not in the initramfs?

```
lsinitramfs /boot/initrd.img-$(uname -r) | grep i915
```

And did you try booting without modeset? In the systemd-boot menu, edit the boot entry to have

```
i915.modeset=0
```

---

nvidia-fallback failing to start doesn't matter; It's not installed.

Graphics mode state is set in `/etc/prime-discrete` and can be `on` (nvidia), `on-demand` (hybrid), or `off` (compute/integrated). This file is also used by gpu-manager (ubuntu-drivers-common).

kylebakerio, 2021-03-03:
```bash
$ lsinitramfs /boot/initrd.img-$(uname -r) | grep i915
usr/lib/firmware/i915
usr/lib/firmware/i915/bxt_dmc_ver1_07.bin
usr/lib/firmware/i915/bxt_guc_33.0.0.bin
usr/lib/firmware/i915/bxt_huc_2.0.0.bin
usr/lib/firmware/i915/cml_guc_33.0.0.bin
usr/lib/firmware/i915/cml_huc_4.0.0.bin
usr/lib/firmware/i915/cnl_dmc_ver1_07.bin
usr/lib/firmware/i915/ehl_guc_33.0.4.bin
usr/lib/firmware/i915/ehl_huc_9.0.0.bin
usr/lib/firmware/i915/glk_dmc_ver1_04.bin
usr/lib/firmware/i915/glk_guc_33.0.0.bin
usr/lib/firmware/i915/glk_huc_4.0.0.bin
usr/lib/firmware/i915/icl_dmc_ver1_09.bin
usr/lib/firmware/i915/icl_guc_33.0.0.bin
usr/lib/firmware/i915/icl_huc_9.0.0.bin
usr/lib/firmware/i915/kbl_dmc_ver1_04.bin
usr/lib/firmware/i915/kbl_guc_33.0.0.bin
usr/lib/firmware/i915/kbl_huc_4.0.0.bin
usr/lib/firmware/i915/skl_dmc_ver1_27.bin
usr/lib/firmware/i915/skl_guc_33.0.0.bin
usr/lib/firmware/i915/skl_huc_2.0.0.bin
usr/lib/firmware/i915/tgl_dmc_ver2_06.bin
usr/lib/firmware/i915/tgl_guc_35.2.0.bin
usr/lib/firmware/i915/tgl_huc_7.0.12.bin
usr/lib/modules/5.8.0-7642-generic/kernel/drivers/gpu/drm/i915
usr/lib/modules/5.8.0-7642-generic/kernel/drivers/gpu/drm/i915/gvt
usr/lib/modules/5.8.0-7642-generic/kernel/drivers/gpu/drm/i915/gvt/kvmgt.ko
usr/lib/modules/5.8.0-7642-generic/kernel/drivers/gpu/drm/i915/i915.ko
```

```bash
kyle@pop-os:~$ grep -r 'nouveau\|nvidia' /usr/lib/modprobe.d
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:blacklist nouveau
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:blacklist lbm-nouveau
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:alias nouveau off
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:alias lbm-nouveau off
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:options nvidia-drm modeset=1
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:options nvidia NVreg_PreserveVideoMemoryAllocations=1
kyle@pop-os:~$ grep -r 'nouveau\|nvidia' /usr/lib/modprobe.d
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:blacklist nouveau
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:blacklist lbm-nouveau
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:alias nouveau off
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:alias lbm-nouveau off
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:options nvidia-drm modeset=1
/usr/lib/modprobe.d/nvidia-graphics-drivers.conf:options nvidia NVreg_PreserveVideoMemoryAllocations=1
```

kylebakerio, 2021-03-03:
I see that `/etc/prime-discrete` for the USB drive install is `off`, even though I had tun commands to try to turn it `on` via chroot. Can I just edit that text file to say `on` and try to boot, or is there something else that needs to be done?

kylebakerio, 2021-03-03:
`i915.modeset=0`

^I've been meaning to look into trying this, but haven't been able to figure out exactly where/how to do so yet. Can you tell me exactly where/how I add this line? Is it a file I modify, or is it a console at boot I drop into, or either/both, or...?

Thanks so much for the feedback!

kylebakerio, 2021-03-03:
Something else interesting, while I have a `/boot/efi/loader/loader.conf` file on my working install on the hard drive, `/boot/efi` is an empty folder on the usb install?

crawfxrd, 2021-03-04:
When you boot the machine repeatedly press space to get to the systemd-boot menu. Move the selection to the Pop!_OS boot entry and press 'E' to edit the entry. Add `i915.modeset=0` to the end of the string and press 'Enter' to boot.

crawfxrd, 2021-03-04:
You may need to report the issue on https://gitlab.freedesktop.org/drm/intel for more help.

kylebakerio, 2021-03-05:
excellent, thanks for the info. will try both of those.

kylebakerio, 2021-03-05:
to be clear, adding that bot entry modification only affects that one boot, right? and if I want to make it permanent, I should add it where?

crawfxrd, 2021-03-05:
Correct. Adding it through the systemd-boot menu only applies to _that_ boot. To make a boot arg persistent in Pop!\_OS, add it to `/etc/kernelstub/configuration` and then run `sudo kernelstub`.

kylebakerio, 2021-03-07:
Well, one time I was able to get into emergency mode--but then, in spite of blinking underscore, no keypress would register (except for the REISUB sequence). 

After that first time, though, no emergency mode again. 

Here's a video of me turning debug up to 10 and the splash and quiet off--perhaps the moment the screen goes black might be informative?

Here are the options I used:
https://i.imgur.com/xOnSCbu.jpg

Here's what happens after I press enter:
https://i.imgur.com/VGI3A4C.mp4

I had to trim it down to 59 seconds to upload to imgur, but the time from me hitting 'enter' on the boot option to the screen then suddenly going black is 1:03.

1) Any tips on filing this bug report to i915? How would you describe this problem?
2) What is the proper way, from the commandline, while within another install, to switch from nvidia -> integrated, and back?

kylebakerio, 2021-03-07:
btw, just to be clear--in those recent tests, I had this:

```bash
root@pop-os:/# cat /usr/share/X11/xorg.conf.d/20-intel.conf 
Section "Monitor"
    Identifier      "DP-0"
    Modeline        "main60"  172.80  1920 2040 2248 2576  1080 1081 1084 1118  -HSync +Vsync

    # 1920x1080 @ 64.00 Hz (GTF) hsync: 71.68 kHz; pclk: 185.79 MHz
    Modeline        "main64"  185.79  1920 2048 2256 2592  1080 1081 1084 1120  -HSync +Vsync

    Option          "PreferredMode" "main64"

    Option          "UseEDIDFreqs" "Off"
    # this is an experiment, haven't tried it yet, may work^
   

EndSection

Section "Device"
  Identifier "Intel Graphics"
  Driver "intel"
EndSection
```

kylebakerio, 2021-03-07:
I tried removing that file and switching it back to nvidia. I had an interesting boot sequence after that--for 1, it spent a long time and failed some boot tasks, I think that the failed switching between graphics modes has broken something about the install (I'll try a fresh one soon probably). For 2, it dropped me into a functioning, typable emergency mode! but then I tried pressing ctrl+D and enter, and then I quickly was dropped back into a black screen.

ashirviskas, 2021-03-09:
No idea if this page is correct, but it says that both GPUs are working properly with PRIME offload on this laptop. Most likely you saw this one, but in case you did not: https://wiki.archlinux.org/index.php/Dell_G5_5590-9340#Graphics

I'll try a few things on mine to see if I can switch to iGPU only. 

P.s. Sorry for not getting back to you on reddit

kylebakerio, 2021-03-09:
Indeed, I have seen that, but I'm a bit out of my field of expertise here, and feel a bit lost translating methods that have nothing to do with Pop to Pop, since Pop doesn't explain anywhere that I have found exactly what they leverage behind the scenes for their GPU switching and how it works--so, it being broken, I don't really understand how to fix it, as I barely understand what is being communicated elsewhere. I've used and set up Prime before on laptops years ago, but it's been a while, and I don't know how to translate what I'm reading there to my situation.

I've been meaning to actually install Manjaro and/or Ubuntu see if it allows me to switch, but haven't gotten to it yet, as I'm waiting until I finish up a client project in the next couple weeks.

Be sure to include whether you have the 144hz monitor with your experiment!

Thanks.

kylebakerio, 2021-03-12:
@ashirviskas, btw--I know that the HDMI and miniDP ports are hardwired to the nvidia card, but if you happen to have the ability to test it, I've been wondering if external monitors (especially two external monitors) are possible via the usb-c/thunderbolt port when in iGPU mode.

ashirviskas, 2021-03-13:
Yes, I do have a 144Hz display!

And I may also have means to test usb-c port for display output (have a USB-C monitor and a cable), but I've never got it to work and only used miniDP.

Though I don't think I've ever got it to work without Nvidia and I still haven't got to testing it.

kylebakerio, 2021-03-13:
I'm using triple monitor via the miniDP and HDMI on the back. I'm going to order a usb-c to dual-hdmi via MST adapter and trying getting triple monitors over that (both for a hardware project I'm doing, as well as to see if running triple monitors while on intel mode is a possibility). I'm also crossing my fingers that I might get surprised and see the external monitors "just work" when I attempt to boot in iGPU mode by chance, which would then make debugging the built-in monitor a lot easier.

Another experiment I want to try is actually installing manjaro and attempting to boot into  intel mode with it. I only attempted to live-boot with it, and it automatically chose nvidia. Perhaps it would handle a switch to the iGPU where pop fails.

kylebakerio, 2021-03-13:
This looks like it was the same issue:
https://github.com/pop-os/system76-power/issues/143

Might be worth digging into what happened there a bit more.

kylebakerio, 2021-03-28:
I've been busy with work so haven't had much time to dig into this issue on my own, but wanted to mention the latest big update:

I finally got a usb-c -> dual hdmi output. This is important, because the built-in video-out ports (HDMI and miniDP) are wired directly into the nvidia card (because this is a "gaming" laptop). That meant that until now, since I was having trouble with the built-in screen, I had no way to try using other screens when the nvidia card was turned off.

Well, as I expected, I **can** use external monitors in integrated mode through the usb-c port (!). The usb stick install I've been working with above booted, went to emergency mode, but then loaded just fine, in integrated mode, with output to my two external monitors.

That means I can finally do some proper debugging while in that environment and try xrandr commands while in that environment. I have some things I can try (I'll have to re-read through this thread myself to re-familiarize myself with what I've learned), but open to suggestions on how to approach this.

Looking at xrandr while booted into integrated, I got this:

[image]

Here's what xrandr looks like while I'm booted into the nvidia:

[image]

As you can see, the built-in screen doesn't show up while in `integrated`, but does in `nvidia`.

The natural first thought would be that perhaps the built-in monitor is connected directly to the nvidia as well, however, we can know this isn't true because when I boot into windows and turn off the nvidia through the hardware devices tab, I can watch the external screens stop working, display server restart, and internal screen work dialed down to 64hz. (I can also see from the nvidia control panel in windows that the built-in screen lacks any EDID settings other than 144hz, though I can manually override and set the screen to 64hz that way as well.) So the screen definitely is connected through the iGPU.

It really does seem that the problem is that there is just a failure to handle a screen with no EDID setting that the iGPU is capable of. (Which, btw, is actually pretty confusing--in theory, the iGPU _should_ be capable of 144hz at 1080p, at least from a processing and data bandwidth POV. However, given that it refuses to run at 144hz in windows, I assume there is some bottleneck here--whether hardware or software imposed--that exists.)

kylebakerio, 2021-03-28:
I have [filed an issue with i915](https://gitlab.freedesktop.org/drm/intel/-/issues/3307).

Included there is what I think is the EDID of the built-in screen (if I'm interpreting this right) that might be relevant:

```sh
Section "Monitor"
	Identifier "K3A8F HDMI"
	ModelName "K3A8F HDMI"
	VendorName "INN"
	# Monitor Manufactured week 32 of 2016
	# EDID version 1.3
	# Digital Display
	DisplaySize 700 390
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-83
	VertRefresh 49-72
	# Maximum pixel clock is 170MHz
	#Not giving standard mode: 1280x1024, 60Hz
	#Not giving standard mode: 1680x1050, 60Hz
	#Not giving standard mode: 1920x1080, 60Hz
	#Not giving standard mode: 1600x900, 60Hz
	#Not giving standard mode: 1440x900, 60Hz

	#Extension block found. Parsing...
	Modeline 	"Mode 18" 148.50 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync 
	Modeline 	"Mode 0" 148.50 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync 
	Modeline 	"Mode 1" 85.75 1366 1436 1579 1792 768 771 774 798 +hsync +vsync 
	Modeline 	"Mode 2" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 3" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 4" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 5" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 6" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 7" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 8" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 9" 25.200 640 656 752 800 480 490 492 525 -hsync -vsync
	Modeline 	"Mode 10" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 11" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 12" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 13" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 14" 74.250 1920 2558 2602 2750 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 15" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 16" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 17" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 19" 74.25 1920 2008 2052 2200 540 542 547 562 +hsync +vsync interlace
	Modeline 	"Mode 20" 74.25 1280 1390 1430 1650 720 725 730 750 +hsync +vsync 
	Modeline 	"Mode 21" 27.00 720 736 798 858 480 489 495 525 -hsync -vsync 
	Option "PreferredMode" "Mode 18"
EndSection
```

As I posted there: 
> That's a bit confusing to me; if this is the EDID for my built-in display, why do I see so many modes, when every GUI only shows a 144hz mode? And why is there no 64hz mode, which is what I see used on windows? But maybe I'm misinterpreting something.

kylebakerio, 2021-03-28:
This might be more meaningful EDID info. Here's what I get for the internal display when running `xrandr --props`:

```sh
DP-0 connected primary 1920x1080+2160+0 (normal left inverted right x axis y axis) 344mm x 194mm
	_MUTTER_PRESENTATION_OUTPUT: 0 
	CTM: 0 1 0 0 0 0 0 0 0 1 0 0 0 0 0 0 
		0 1 
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	Backlight: 100 
		range: (0, 100)
	EDID: 
		00ffffffffffff0009e54a0800000000
		011c0104a522137802de50a3544c9926
		0f505400000001010101010101010101
		0101010101019d8580a070383e406c30
		aa0058c21000001a0000000000000000
		0000000000000000001a000000fe0046
		35542050804e56313536464800000000
		0002412f9e021000000b010a2020004e
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: DisplayPort 
		supported: DisplayPort
	ConnectorType: Panel 
	ConnectorNumber: 0 
	_ConnectorLocation: 0 
	non-desktop: 0 
		supported: 0, 1
   1920x1080    144.00*+

```

kylebakerio, 2021-04-07:
Note: I recently tried installing manjaro as a dual boot to see how it would behave. It is less clear how to turn off the nvidia card in manjaro, but I think I installed optimus-manager. I didn't go through the proper steps to get gnome patched, etc., that seem necessary for that, but I noticed something interesting anyways--on rebooting, it seemed I had indeed switched to the intel gpu and turned off the nvidia gpu, as the built in screen wouldn't work and an external monitor connected to the hdmi port didn't work, but an external monitor connected through usb-c did.

However, what was interesting was that the DELL logo stayed on the built in screen. You can see this state shown here:

[image]

kylebakerio, 2021-04-07:
Have now asked here: https://unix.stackexchange.com/questions/643941/built-in-monitor-recognized-when-using-nvidia-driver-but-not-when-using-i915

kylebakerio, 2021-07-11:
Hello, just thought I'd update this issue, which is still not resolved.

I tried a live boot of 21.04 on this machine/setup. From what I can tell, it seems that it refuses to auto-boot to nvidia mode, and so it doesn't boot up at all, unlike 20.10 which would somehow realize only nvidia mode would work and auto-select nvidia for first boot up. 

I see some logs on text that doesn't match the screen size about an acpi error on the built-in screen, and I see the gray background color of the default login screen on the two external monitors that are connected via usb-c (MST over DP over USB-C).

appletechgeek, 2021-12-09:
got the same issue with a msi ge66 10ug. 10870h with 3070. 

when i installed the os it worked fine.

but ever since i disabled the igpu in the bios and then enabled it again (with a boot in between)

the system gets stuck on a grey screen. and only works if a external screen is connected during boot.

leviport, 2021-12-09:
@appletechgeek does anything happen if you press the Esc key when it's at the gray screen? What you describe _might_ just be Plymouth not getting all the way through its graphical initialization process. If that's the case, hitting Esc should drop down to a text-only version of the cryptdata unlock prompt. From there the drive can either be unlocked, or hitting Esc a second time will make Plymouth reload, which might make it display properly.