pop-os/system76-power #235: Arch on Lemur: Cannot set charge thresholds

phelucko, 2021-04-08:
**Distribution (run `cat /etc/os-release`):** 

```
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://www.archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux
```

**Issue/Bug Description:**

The Command `system76-power charge-thresholds 60 80` failed with error `daemon returned an error message: "Action com.system76.powerdaemon.set-charge-thresholds is not registered"`.

**Steps to reproduce (if you know):**

Run `system76-power charge-thresholds 60 80`

**Expected behavior:**

The charge thresholds should be changed.

**Other Notes**

My current system76-power package version is 1.1.16-2.
The firmware version is 2021-03-11_50eedc2.

MilesBHuff, 2021-04-24:
Same problem on Manjaro/oryp7.  If you set  the thresholds from Pop!_OS, though, they will persist in Arch/Manjaro.  So you could do that as a workaround.

Nox-404, 2021-04-28:
Same here (arch same versions as phelucko)

Nox-404, 2021-04-29:
```system76-power --version
system76-power 1.1.16```

```systemd[1]: Started System76 Power Daemon.
system76-power[7224]: [ERROR] fan daemon: platform hwmon not found
system76-power[7224]: [INFO] Handling dbus requests
system76-power[7224]: thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/hid_backlight.rs:77:55
system76-power[7224]: stack backtrace:
system76-power[7224]:    0:     0x55a37ef5e830 - <unknown>
system76-power[7224]:    1:     0x55a37ef8141c - <unknown>
system76-power[7224]:    2:     0x55a37ef5a565 - <unknown>
system76-power[7224]:    3:     0x55a37ef60aa5 - <unknown>
system76-power[7224]:    4:     0x55a37ef60603 - <unknown>
system76-power[7224]:    5:     0x55a37ef61241 - <unknown>
system76-power[7224]:    6:     0x55a37ef60d87 - <unknown>
system76-power[7224]:    7:     0x55a37ef5eccc - <unknown>
system76-power[7224]:    8:     0x55a37ef60ce9 - <unknown>
system76-power[7224]:    9:     0x55a37ef7fac1 - <unknown>
system76-power[7224]:   10:     0x55a37ef7f8e3 - <unknown>
system76-power[7224]:   11:     0x55a37ee2660b - <unknown>
system76-power[7224]:   12:     0x55a37ee300f7 - <unknown>
system76-power[7224]:   13:     0x55a37edef2a2 - <unknown>
system76-power[7224]:   14:     0x55a37ef67dea - <unknown>
system76-power[7224]:   15:     0x7f8a29a23299 - start_thread
system76-power[7224]:   16:     0x7f8a297fe053 - clone
system76-power[7224]:   17:                0x0 - <unknown>```

The error seems to be because the file `/sys/class/leds/system76_acpi::kbd_backlight/color` doesn't exist

Nox-404, 2021-04-29:
A simpler way to set the charge thresholds until there's a fix is to run the following:

```
echo 50 | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo 60 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

(use any values you want instead of 50/60)

Nox-404, 2021-04-29:
Okay the problem was a missing polkit action

Just add this to `/usr/share/polkit-1/actions/com.system76.powerdaemon.policy` :
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE policyconfig PUBLIC "-//freedesktop//DTD polkit Policy Configuration 1.0//EN"
    "http://www.freedesktop.org/software/polkit/policyconfig-1.dtd">
<policyconfig>
  <vendor>System76</vendor>
  <vendor_url>https://system76.com</vendor_url>
  <action id="com.system76.powerdaemon.set-charge-thresholds">
    <description>Set charge thresholds</description>
    <message>Setting charge thresholds requires authorization</message>
    <defaults>
      <allow_any>auth_admin</allow_any>
      <allow_inactive>auth_admin</allow_inactive>
      <allow_active>auth_admin</allow_active>
    </defaults>
  </action>
</policyconfig>
```
And then it'll work as expected

@phelucko @MilesBHuff

[image]

That file should be added to the arch package

MilesBHuff, 2021-04-29:
> The error seems to be because the file `/sys/class/leds/system76_acpi::kbd_backlight/color` doesn't exist
This error is unrelated -- it just means you don't have `system76-acpi-dkms` installed.  You can find it on the AUR.  It should be added as a dependency if its lack causes issues.

MilesBHuff, 2021-04-29:
> Okay the problem was a missing polkit action

Thanks @Nox-404 -- you rock!  I'll give this a try when I get off work.

Nox-404, 2021-04-30:
The polkit action is inside this repository (data folder)

You may now just update the package (aur) it has been updated and now install the file as intended !

(I think this issue can be closed now)

crawfxrd, 2021-04-30:
Fixed in 1.1.16-3 on AUR.

closed 2021-04-30 by crawfxrd